medusalix/xow #1: Does not pair with model 1708

NoXPhasma, 2019-12-21:
I have the model 1708 but it does not pair. The dongle goes into pairing mode, when I click the button on it and the controller does as well, but it never pairs.

It goes into pairing mode until it times out:
```
$ sudo ./xow 
2019-12-21 19:18:32 INFO  - xow v0.1 ©Severin v. W.
2019-12-21 19:18:32 INFO  - Dongle plugged in
2019-12-21 19:18:32 INFO  - Chip address: 62:45:b4:fe:81:16
2019-12-21 19:18:32 INFO  - Dongle initialized
2019-12-21 19:18:44 INFO  - Pairing initiated
2019-12-21 19:18:44 INFO  - Pairing initiated
2019-12-21 19:18:45 INFO  - Pairing initiated
2019-12-21 19:18:45 INFO  - Pairing initiated
2019-12-21 19:22:19 ERROR - Error in bulk write: LIBUSB_ERROR_TIMEOUT
2019-12-21 19:22:19 ERROR - Failed to write WLAN packet
2019-12-21 19:22:19 ERROR - Failed to send pairing packet
2019-12-21 19:22:19 ERROR - Failed to pair client
...
```

I've also tried to use a 2.4 channel, but it's the same. Please tell me if you need any more informations.

medusalix, 2019-12-21:
Please see if my latest commit fixes your issue. Pairing works with my controller now.

NoXPhasma, 2019-12-21:
Yes, pairing works now. Also thank you for doing this driver, great work!

NoXPhasma, 2019-12-21:
As your readme says, connection breaks on 5GHz but with 2.4GHz channels it won't pair.

medusalix, 2019-12-21:
Yeah, automatic channel selection is definitely needed. I'm taking a look at the pairing issue right now.

medusalix, 2019-12-21:
It pairs for me just fine even on 2.4 GHz (channel 1).

NoXPhasma, 2019-12-21:
I've tried 10 and 5 which don't work, but channel 1 works for me as well. Is this a limitation of the dongle?

medusalix, 2019-12-21:
I really don't know 😄. I'm going to make channel 1 the default because it seems like that's the most reliable one.

closed 2019-12-21 by medusalix

medusalix, 2019-12-28:
The reason why only certain channels work is because 802.11n WLAN uses **20 MHz bandwidth** by default. Channels 1, 6 and 11 are at least 20 MHz apart, unlike the [channels inbetween](https://en.wikipedia.org/wiki/List_of_WLAN_channels) that overlap their frequency ranges. With 5 GHz WLAN, the dongle additionaly only supports the channels that are available in all regions of the world. Hope that clears things up a bit.