medusalix/xow #2: Mapping of X and Y buttons is mirrored

NoXPhasma, 2019-12-21:
The X and Y buttons are mirrored, X is recognized as Y and vis versa. I guess there is something mixed up in the code :)

medusalix, 2019-12-21:
Buttons are mapped according to the Linux docs. So the DPAD is mapped using BTN_NORTH, BTN_EAST, BTN_SOUTH and BTN_WEST. Unfortunately, it looks like BNT_NORTH is [equivalent](https://github.com/torvalds/linux/blob/737214515d7965485c52bc5f869e57a6976cf216/include/uapi/linux/input-event-codes.h#L386) to BTN_X on Linux. I'm not sure which button scheme I should be following.

NoXPhasma, 2019-12-21:
The X and Y buttons definitely doesn't follow what the XBOX One S controller is mapped with xpad (wired), hid_microsoft (bluetooth) or xpadneo (blietooth). No matter what game/tool I use, it's wrong in all of them.

medusalix, 2019-12-21:
Should we just ignore the docs and map the buttons using BTN_A, BTN_B, BTN_X and BTN_Y?

NoXPhasma, 2019-12-21:
I think we should go the same route as xpadneo and xpad go. As this would make sure it's recognized the same way and we don't run into any issues.

I've also just found out that the DPAD won't be recognized in games as well. You've mapped it as buttons 11-14, but apparently with xpad it's mapped as axes 6 and 7.

NoXPhasma, 2019-12-21:
Your commit (https://github.com/medusalix/xow/commit/502abcf6698e0fc91129c6e9e2dfc4c82bf4188e) fixed the wrong Y and X mapping. Now only the DPAD is left and the controller works as intended :)

medusalix, 2019-12-21:
The latest commit should've now fixed the DPAD.

NoXPhasma, 2019-12-21:
It does, the mapping is now as expected.

closed 2019-12-21 by NoXPhasma