medusalix/xow #3: XBOX Elite Controller not working (only vibrate in full force)

NTMan, 2019-12-21:
Something went wrong. After launching the game, The Surge XBOX Elite Controllers amicably began to vibrate in full force.

Demonstration: https://youtu.be/_VevGQK8jAA

nadiadavis, 2019-12-21:
The readme states that Elite controllers are currently unsupported.

medusalix, 2019-12-21:
Yes, that's right. If I remember correctly, the elite controllers use a slightly different protocol than the normal controllers, which is not supported by this driver. I might add support in the future though.

Hielyr, 2019-12-22:
> Yes, that's right. If I remember correctly, the elite controllers use a slightly different protocol than the normal controllers, which is not supported by this driver. I might add support in the future though.

I would love to see support for the elite controllers. I have an Elite v2 controller, so if it would be at all helpful for me to provide any information, I'd be more than happy to do so!

NTMan, 2019-12-22:
> Yes, that's right. If I remember correctly, the elite controllers use a slightly different protocol than the normal controllers, which is not supported by this driver. I might add support in the future though.

Vibro of usual XBOX controller going mad too
Demonstration: https://youtu.be/Drdp2f5xARU

medusalix, 2019-12-22:
@NTMan It seems to be an issue with Proton games, I'll need to take a closer look at this.

John-Gee, 2019-12-22:
I have the same issue on a 1697 model with pure Wine, not using Proton or Steam.

NoXPhasma, 2019-12-22:
As I already stated on [Reddit](https://www.reddit.com/r/linux_gaming/comments/edusyx/linux_driver_for_the_new_xbox_one_wireless_dongle/fbouq2c/), for me the Gamepad rumbles very hard at launch on Proton games as well and then the rumble won't work in the game. This behaviour is not present in native or games running with Wine, in which the gamepad works perfectly fine.

This is with model 1708.

zypangpang, 2019-12-24:
Same here. With proton games, the Gamepad rumbles very hard at launch, but in my case the vibration still works in the game but still in full force.

NoXPhasma, 2019-12-24:
I've updated the firmware of my controller to version 4.x and now rumble works in Proton games as well. If anyone else experiences no rumble in Proton games, I suggest to update the controller firmware (a XBOX One console or Windows is required though).

medusalix, 2019-12-25:
I've removed the rumble of the triggers in the latest commit. Rumble in native games (like Broforce) feels very similar to their Windows counterparts now. This is the way `xpad` treats the rumble.

NoXPhasma, 2019-12-25:
This is much better indeed. Maybe you can take a look into the xpadneo code, how the trigger rumble is done in it.

John-Gee, 2019-12-26:
Just in case, I tried again, it still rumbles in full force from launch till closing a game in Wine.

NoXPhasma, 2019-12-26:
Have you tried to update the firmware of your controller?

John-Gee, 2019-12-26:
I have not, I don't have Windows to try.
It's likely the controller never got updated before it was given to me.
Is there a way to do it from Linux?

NoXPhasma, 2019-12-26:
You can download a Windows 10 VM image [1] legally for free, then passthrough the gamepad/dongle. Then install the Xbox accessories app with the Windows store and update the firmware. I have not done it this way, but it should work.

[1] https://developer.microsoft.com/windows/downloads/virtual-machines

NTMan, 2019-12-26:
[image]

In my case with firmware 4.8.1923.0 I still observed this issue.

John-Gee, 2019-12-26:
> You can download a Windows 10 VM image [1] legally for free, then passthrough the gamepad/dongle. Then install the Xbox accessories app with the Windows store and update the firmware. I have not done it this way, but it should work.
> 
> [1] [developer.microsoft.com/windows/downloads/virtual-machines](https://developer.microsoft.com/windows/downloads/virtual-machines)

I had a VM already setup so I tried, but I've spent a couple hours and so far only once did XBox accessories see the controller, and of course it had to fail the update, every other time it asks me to plug in a controller. Not sure what I'm missing, but based on above I don't know if it's worth looking more into that.

NoXPhasma, 2019-12-26:
The firmware Version looks like the latest. It might be actually the different model version which is causing this.

For me it only rumbles when WINE starts and ends for 3 seconds, no matter if it's a game or a WINE application. With Proton it only rumbles on launch. However, with WINE the rumble is stronger than with Proton.

medusalix, 2019-12-26:
I might have fixed this problem (while reading through the source code of [`ff-memless`](https://github.com/torvalds/linux/blob/master/drivers/input/ff-memless.c). Please see if the rumble behaves correctly when running the modified [`rumble`](https://github.com/medusalix/xow/tree/rumble) branch.

NTMan, 2019-12-26:
> I might have fixed this problem (while reading through the source code of [`ff-memless`](https://github.com/torvalds/linux/blob/master/drivers/input/ff-memless.c)). Please see if the rumble behaves correctly when running the modified [`rumble`](https://github.com/medusalix/xow/tree/rumble) branch.

xow now crashes: https://pastebin.com/gmXAPv8e

NTMan, 2019-12-26:
And yes, controller not rumble, but it does not rumble also at the moments when rumble should be (present when controller connected by wire)

medusalix, 2019-12-26:
@NTMan Oops, made a little logic error (should be fixed). Not sure about the crash though...

NoXPhasma, 2019-12-26:
It works for me in Wine, Proton and Native without any initial rumble. I need to do more testing, as I have currently the feeling that the rumble feels less intense as usual. Also for some reason the rumble won't work in Steam games (Native and Proton) once I exit the game and start a game without reconnect the gamepad. This was not the case with the default branch.

John-Gee, 2019-12-27:
The rumble branch works for me. I quickly compared it with wired, and with a Wii U Pro controller in Monster Boy and it seems about the same.

I've had no issues with restarting a game, it still works fine.

Thank you very much!

NoXPhasma, 2019-12-27:
> I quickly compared it with wired, and with a Wii U Pro controller in Monster Boy and it seems about the same.

You are right, I compared it wired and wireless as well and it is the same. Most probably I just got used to the intense rumble of the main branch.

NTMan, 2019-12-27:
> @NTMan Oops, I made a little logic error (should be fixed now). ~Not sure about the crash though...~
> 
> The crash might be caused by `select`ing on a file descriptor that's bigger than `FD_SETSIZE`. I'll have to switch to `poll/epoll` to avoid this issue.

Good job, the commit d56ce577943e6d0f70710892f1f861b49f592bbe fix problem with not working rumble.

But crash still here.
How reproduce:
1. Pairing gamepad with xow
2. Connect wired (here would disconnected wireless connection)
3. Unplug wired (here expected reconnect wireless connection, but instead we have crash)

NoXPhasma, 2019-12-27:
I followed your list and it did not crash for me with model 1708.

NTMan, 2019-12-27:
> I followed your list and it did not crash for me with model 1708.

Demonstration: https://youtu.be/6obtRp7PcLc

John-Gee, 2019-12-27:
No crash here following this list either.

medusalix, 2019-12-27:
@NTMan Please see if the newest commit fixes your issue.

NTMan, 2019-12-27:
> @NTMan Please see if the newest commit fixes your issue.

yes, no crashes observed on commit 28a51c4e53e37d41d4e39ddc71dd909818963c60

The last problem that I am observed when plug wire on the fly it infinitely reiteration last action.

For example:

1. Paring gamepad with xow.
2. Tilt stick to any direction.
3. Plug wired to the gamepad.

Demonstration: https://youtu.be/BpcJbzHW0UY

medusalix, 2019-12-27:
@NTMan Are you running `xpad` on your machine? I assume it reconnects over USB once you plug it in and `xpad` doesn't reset its input on connection.

NoXPhasma, 2019-12-28:
I'm not entirely sure, but I think [Ori and the Blind Forest Definitive Edition](https://store.steampowered.com/app/387290/Ori_and_the_Blind_Forest_Definitive_Edition/) makes use of `ff_effect.direction`. At least the rumble feels much stronger now with the latest commit, especially around the triggers, which now even make sound due to the rumble. And it would make sense, as it's a Microsoft Studio game and also was released first on XBOX One.

medusalix, 2019-12-29:
I've merged the changes into `master` in 1d44947ea3827f2a8bff36c58e35e3541f8ecead. Issue seems to be solved.

closed 2019-12-29 by medusalix