medusalix/xow #10: "double free or corruption (fasttop)double free or corruption (fasttop)"

PopeRigby, 2019-12-22:
I'm getting this output from my terminal when running `sudo xow`:

```
2019-12-22 12:36:47 INFO  - xow v0.2-2-g297efa4 ©Severin v. W.
2019-12-22 12:36:47 INFO  - Dongle plugged in
double free or corruption (fasttop)double free or corruption (fasttop)

zsh: abort      sudo xow
```

Model: 1708

medusalix, 2019-12-25:
Could you please run xow with [`valgrind`](https://valgrind.org/)? This should tell you exactly where the double free is occurring.

PopeRigby, 2019-12-26:
I tried turning on my controller, and it just kept blinking without pairing, so I pressed the pairing buttons on both the controller and dongle, and it still just kept blinking. Here's what the terminal outputted when I was doing all of this
```
~ $ sudo valgrind -s xow
==15252== Memcheck, a memory error detector
==15252== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==15252== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==15252== Command: xow
==15252== 
2019-12-26 11:21:14 INFO  - xow v0.2-2-g297efa4 ©Severin v. W.
2019-12-26 11:21:14 INFO  - Dongle plugged in
2019-12-26 11:21:15 INFO  - Chip address: 62:45:b5:04:3f:d4
2019-12-26 11:21:15 INFO  - Dongle initialized
2019-12-26 11:21:27 INFO  - Pairing initiated
2019-12-26 11:21:27 INFO  - Pairing initiated
2019-12-26 11:21:27 INFO  - Pairing initiated
^C==15252== 
==15252== Process terminating with default action of signal 2 (SIGINT)
==15252==    at 0x4BC79EF: poll (in /usr/lib/libc-2.30.so)
==15252==    by 0x4A85985: ??? (in /usr/lib/libusb-1.0.so.0.2.0)
==15252==    by 0x4A86C0C: libusb_handle_events_timeout_completed (in /usr/lib/libusb-1.0.so.0.2.0)
==15252==    by 0x4A86CB0: libusb_handle_events_completed (in /usr/lib/libusb-1.0.so.0.2.0)
==15252==    by 0x11D429: ??? (in /usr/bin/xow)
==15252==    by 0x10E8C0: ??? (in /usr/bin/xow)
==15252==    by 0x4AFA152: (below main) (in /usr/lib/libc-2.30.so)
==15252== 
==15252== HEAP SUMMARY:
==15252==     in use at exit: 8,937 bytes in 40 blocks
==15252==   total heap usage: 2,296 allocs, 2,256 frees, 963,852 bytes allocated
==15252== 
==15252== LEAK SUMMARY:
==15252==    definitely lost: 0 bytes in 0 blocks
==15252==    indirectly lost: 0 bytes in 0 blocks
==15252==      possibly lost: 4,845 bytes in 20 blocks
==15252==    still reachable: 4,092 bytes in 20 blocks
==15252==         suppressed: 0 bytes in 0 blocks
==15252== Rerun with --leak-check=full to see details of leaked memory
==15252== 
==15252== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

medusalix, 2019-12-26:
Some things that might be of importance:

- What version of the dongle are you using?
- Is the firmware of your controller up-to-date? If not, please update!
- Have you tried changing the WLAN channel?

PopeRigby, 2019-12-27:
> What version of the dongle are you using?

The slim version
[image]

> Is the firmware of your controller up-to-date?

Yep, it is.
> Have you tried changing the WLAN channel?

How would I do that?

medusalix, 2019-12-27:
> How would I do that?

It's defined in the Makefile. The list of channels is in mt76.h.

PopeRigby, 2019-12-27:
Which channels do you suggest I try?

medusalix, 2019-12-27:
Channel 165 is working for me (although it doesn't have a great range).

PopeRigby, 2019-12-28:
It seems to be working now. Maybe I wasn't pairing correctly, or maybe channel 1 doesn't work. I'm going to test that.

PopeRigby, 2019-12-28:
Nope, channel one seems to be working too. Maybe it was a problem with the AUR package.

medusalix, 2019-12-28:
I'm glad that it's working now 😃. Please reopen if that error occurs to you again.

closed 2019-12-28 by medusalix