medusalix/xow #6: Please add battery level measurements via "upower" mechanism

NTMan, 2019-12-22:
This allows seeing the power charge level of the gamepad in DE.

[image]

```
$ upower --dump
Device: /org/freedesktop/UPower/devices/mouse_hidpp_battery_0
  native-path:          hidpp_battery_0
  model:                Zone Touch Mouse T400
  serial:               4026-cf-15-61-0d
  power supply:         no
  updated:              Sun 22 Dec 2019 02:41:30 PM +05 (89 seconds ago)
  has history:          yes
  has statistics:       yes
  mouse
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    battery-level:       normal
    percentage:          55% (should be ignored)
    icon-name:          'battery-low-symbolic'

Device: /org/freedesktop/UPower/devices/gaming_input_sony_controller_battery_f4o93o9fo34o2eodd
  native-path:          sony_controller_battery_f4:93:9f:34:2e:dd
  model:                Sony Interactive Entertainment Wireless Controller
  serial:               f4:93:9f:34:2e:dd
  power supply:         no
  updated:              Sun 22 Dec 2019 02:42:56 PM +05 (3 seconds ago)
  has history:          yes
  has statistics:       yes
  gaming-input
    rechargeable:        yes
    warning-level:       none
    percentage:          50%
    icon-name:          'battery-good-charging-symbolic'
  History (charge):
    1577007776	0.000	unknown
  History (rate):
    1577007776	0.000	unknown

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         no
  updated:              Sun 22 Dec 2019 10:58:54 AM +05 (13445 seconds ago)
  has history:          no
  has statistics:       no
  unknown
    warning-level:       none
    icon-name:          'battery-missing-symbolic'

Daemon:
  daemon-version:  0.99.11
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  no
  critical-action: HybridSleep
```

medusalix, 2019-12-22:
I spent the last few hours figuring out how to add a device to UPower. Unfortunately, it looks like either the UPower daemon (`upowerd`) or xow can be connected to UPower at the same time. When xow connects, you lose the power status of all your other connected devices. Providing battery information from user space is not trivial at all...

medusalix, 2020-01-10:
Please head over to `upower` and upvote the corresponding [issue](https://gitlab.freedesktop.org/upower/upower/issues/105) on their GitLab. That missing feature is currently preventing me from adding upower integration.

cristianovitorino, 2020-04-05:
@medusalix Unfortunately the `upower` mantainer seems uninterested in helping. I was thiking in responding but maybe you can make a better argument there?

Also, the projects [xpadneo](https://atar-axis.github.io/xpadneo/) and specially [Solaar](https://pwr-solaar.github.io/Solaar/) which is a driver that deals with Logitech proprietary receivers for mice can offer some insight maybe, as both of them have access to battery information.

medusalix, 2020-04-07:
@cristianovitorino I think he's generally unconvinced of the necessity of proper upower integration for standard user mode software. Kernel drivers like `xpadneo` are already able to use `power_supply_register` to make their battery information available to upower and `Solaar` (being a rather popular project) apparently has its own [integration](https://gitlab.freedesktop.org/upower/upower/-/blob/master/src/linux/up-device-unifying.c) built into upower.

medusalix, 2020-06-06:
Unfortunately, it doesn't look like the `upower` project is currently being maintained by anyone. While looking through their open issues and merge requests, it seems like the authors reject even the simplest changes that would lead to an improvement of `upower`. I seriously doubt they would even accept a PR for an integration with xow.
Therefore, I've decided to add support for **desktop notifications** to xow. Whenever you connect a controller or the battery level of one of the connected controllers changes, you get an unintrusive notification informing you of the current battery status. When the level reaches a critical point, a *high-priority* notification is shown (which also works in fullscreen games).
I know notifications are only part of what you would expect from a full `upower` integration, but I think it's a step in the right direction while we're waiting for further updates to `upower`.
Any feedback would be greatly appreciated, so if you want to try things out, the `battery` branch is ready for you to check out. Do note that desktop notifications require you to run xow as your local user as explained in the *troubleshooting* section of the README.

cristianovitorino, 2020-06-08:
@medusalix Does it use `notify-send`? Can it use `notify-send`?

medusalix, 2020-06-08:
It does exactly the same things `notify-send` would do, but implemented internally by calling the notifications API directly.

cristianovitorino, 2020-06-08:
> It does the exact same things `notify-send` would do, but implemented internally by calling the notifications API directly.

I use [Dunst](https://dunst-project.org/) to get my notifications, usually from `notify-send`. Do you think it will work with your implementation? I can't test it right now.

medusalix, 2020-06-08:
> I use Dunst to get my notifications, usually from notify-send. Do you think it will work with your implementation? I can't test it right now. Will do later.

Yes, absolutely. Anything that adheres to Gnome's [Desktop Notifications Specification](https://developer.gnome.org/notification-spec/) should work (not limited to Gnome).

cristianovitorino, 2020-06-08:
> Yes, absolutely. Anything that adheres to Gnome's [Desktop Notifications Specification](https://developer.gnome.org/notification-spec/) should work (not limited to Gnome).

Awesome! Thanks!

x4mers, 2020-06-14:
Looks like this would allow xow to continue running the way it is, instead of moving to user service?

But then 5.8 would be req'd

https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.8-Notification-Queue

medusalix, 2020-06-14:
> phoronix.com/scan.php?page=news_item&px=Linux-5.8-Notification-Queue

How does that relate to desktop notifications? That's a notification queue for user space <-> kernel communication.

mibli, 2021-05-06:
Hey, instead of moving xow to userspace, userspace can have optional xow log parser or notificator, which could be separate (something perhaps a little bit more advanced than my sed-based utility service https://gist.github.com/mibli/89c02bf6fcd4283e6369fb0124ad7a1b#file-xow-notify-service)

kakra, 2021-05-06:
> userspace can have optional xow log parser or notificator

This still wouldn't properly interface with upower. There's no way for a user-space app to properly interface with upowerd other than by special hacks added to the upower dbus API (I think there are hacks for solaar and similar software). By default, upowerd only looks at power interfaces provided by kernel drivers.

But yes, with this solution, xow can run as a system-wide service because your idea decouples both. Instead of watching the log, the system-wide process could open a pipe/socket where it writes battery information. An (yet to be created) xow-user.service could then open the socket (or query the dbus endpoint) and watch for new battery notifications to directly write to the notification API of the DE.

medusalix, 2022-03-20:
I've released a new driver for the Xbox wireless dongle which supports battery reporting via `upower`. You can check it out [here](https://github.com/medusalix/xone).

NTMan, 2022-03-26:
Thanks, do plan working to include xone driver to the mainline linux kernel?
Also xone caused error message in kernel log `BUG: sleeping function called from invalid context at include/linux/sched/mm.h:257` can you look?

```
[62478.255424] usbcore: deregistering interface driver xpad
[62478.330752] usbcore: deregistering interface driver mt76x2u
[62590.824407] usb 3-1: USB disconnect, device number 2
[62591.270867] usb 3-1: new full-speed USB device number 6 using xhci_hcd
[62591.420542] usb 3-1: New USB device found, idVendor=2516, idProduct=0123, bcdDevice= 1.10
[62591.420551] usb 3-1: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[62591.420554] usb 3-1: Product: SK650 White
[62591.420557] usb 3-1: Manufacturer: Cooler Master Technology Inc.
[62591.446195] input: Cooler Master Technology Inc. SK650 White as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-1/3-1:1.0/0003:2516:0123.0011/input/input46
[62591.498222] hid-generic 0003:2516:0123.0011: input,hidraw0: USB HID v1.11 Keyboard [Cooler Master Technology Inc. SK650 White] on usb-0000:06:00.3-1/input0
[62591.502132] hid-generic 0003:2516:0123.0012: hiddev96,hidraw1: USB HID v1.11 Device [Cooler Master Technology Inc. SK650 White] on usb-0000:06:00.3-1/input1
[62591.506154] input: Cooler Master Technology Inc. SK650 White System Control as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-1/3-1:1.2/0003:2516:0123.0013/input/input47
[62591.559048] input: Cooler Master Technology Inc. SK650 White Consumer Control as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-1/3-1:1.2/0003:2516:0123.0013/input/input48
[62591.559178] input: Cooler Master Technology Inc. SK650 White Mouse as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-1/3-1:1.2/0003:2516:0123.0013/input/input49
[62591.559349] input: Cooler Master Technology Inc. SK650 White Keyboard as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-1/3-1:1.2/0003:2516:0123.0013/input/input50
[62591.559491] hid-generic 0003:2516:0123.0013: input,hidraw5: USB HID v1.11 Mouse [Cooler Master Technology Inc. SK650 White] on usb-0000:06:00.3-1/input2
[62592.901332] usb 7-1.1.4: USB disconnect, device number 7
[62595.975875] usb 7-1.1.4: new high-speed USB device number 14 using xhci_hcd
[62596.117088] usb 7-1.1.4: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[62596.117097] usb 7-1.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[62596.117101] usb 7-1.1.4: Product: XBOX ACC
[62596.117103] usb 7-1.1.4: Manufacturer: Microsoft Inc.
[62596.117105] usb 7-1.1.4: SerialNumber: 072144
[62596.135936] xone_gip_bus: loading out-of-tree module taints kernel.
[62596.135997] xone_gip_bus: module verification failed: signature and/or required key missing - tainting kernel
[62596.267396] usb 7-1.1.4: reset high-speed USB device number 14 using xhci_hcd
[62596.408914] xone-dongle 7-1.1.4:1.0: xone_mt76_send_firmware: build=201703281033____
[62596.561838] xone-dongle 7-1.1.4:1.0: xone_mt76_init_radio: id=0x7613
[62596.591092] xone-dongle 7-1.1.4:1.0: xone_mt76_init_address: address=62:45:b5:07:5f:b7
[62596.596247] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=1, power=30
[62596.598125] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=6, power=32
[62596.599998] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=11, power=33
[62596.602335] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=36, power=44
[62596.604135] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=40, power=44
[62596.606001] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=44, power=43
[62596.607885] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=48, power=43
[62596.609756] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=149, power=40
[62596.611621] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=153, power=40
[62596.613498] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=157, power=38
[62596.615373] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=161, power=38
[62596.617249] xone-dongle 7-1.1.4:1.0: xone_mt76_evaluate_channels: channel=165, power=38
[62596.617593] xone-dongle 7-1.1.4:1.0: xone_mt76_init_channels: channel=165
[62597.713383] usbcore: registered new interface driver xone-dongle
[62598.876884] xone-dongle 7-1.1.4:1.0: xone_dongle_toggle_pairing: enabled=1
[62605.790793] xone-dongle 7-1.1.4:1.0: xone_dongle_pair_client: address=7e:ed:82:53:f1:e7
[62605.791138] xone-dongle 7-1.1.4:1.0: xone_dongle_toggle_pairing: enabled=0
[62606.213108] xone-gip gip0: gip_create_adapter: registered
[62606.213272] xone-dongle 7-1.1.4:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:82:53:f1:e7
[62606.731518] BUG: sleeping function called from invalid context at include/linux/sched/mm.h:257
[62606.731524] in_atomic(): 1, irqs_disabled(): 1, non_block: 0, pid: 0, name: swapper/18
[62606.731528] preempt_count: 103, expected: 0
[62606.731530] RCU nest depth: 0, expected: 0
[62606.731531] INFO: lockdep is turned off.
[62606.731533] irq event stamp: 25074880
[62606.731535] hardirqs last  enabled at (25074879): [<ffffffffa7000cde>] asm_common_interrupt+0x1e/0x40
[62606.731543] hardirqs last disabled at (25074880): [<ffffffffa614b5ae>] do_idle+0xae/0x260
[... cut ...]
[62607.300385] xone-gip gip0.1: gip_add_client: added
[62607.305531] input: Microsoft X-Box One chatpad as /devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb7/7-1/7-1.1/7-1.1.4/7-1.1.4:1.0/gip0/gip0.1/input/input52
[62607.306279] input: Microsoft X-Box One chatpad as /devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb7/7-1/7-1.1/7-1.1.4/7-1.1.4:1.0/gip0/gip0.1/0003:045E:0B02.0014/input/input53
[62607.306799] hid-generic 0003:045E:0B02.0014: input,hidraw8: USB HID v1.01 Keyboard [Microsoft X-Box One chatpad] on gip0.1/input1
```

medusalix, 2022-03-31:
@NTMan

Yes, I plan on submitting `xone` for the mainline kernel in the future.
Thanks for the bug report, I'm working on a fix.
You can report issues on Discord if you want.