medusalix/xow #9: Package as a Flatpak?

PopeRigby, 2019-12-22:
Would it be feasible to package this as a Flatpak?

medusalix, 2019-12-22:
It would definitely be possible to package it as a Flatpak, the [app guidelines](https://github.com/flathub/flathub/wiki/App-Requirements#desktop-integration) for the Flathub however say that they intend to distribute desktop applications and not CLI programs.

PopeRigby, 2019-12-22:
That's annoying, although having it as a Flatpak would still make it easier to distribute. You would just have to distribute it from somewhere else if Flathub didn't except it.

medusalix, 2019-12-23:
I think it's *pretty easy* to build it yourself (literally 3 commands). Feel free to create and publish a Flatpak and I'll add the link to your package to the README 😃.

medusalix, 2020-02-09:
I just did some reasearch and it's apparently [not possible](https://github.com/flatpak/flatpak/issues/524) to package software that runs `systemd` services as Flatpaks. So I guess we'll have to rely on traditional software repositories for xow's distribution.

closed 2020-02-09 by medusalix

marcusmueller, 2024-12-27:
@medusalix request to reopen this; the statement that users have no services as cited in #524 is very much false. What do you think desktop portals are? User services are core functionality of systemd, enabling integration of flatpak.

kakra, 2024-12-27:
> is very much false

Only because context was missing: xow needs to run a root service. flatpak cannot do that on purpose. So that statement is still true in context of xow. User services are something different: flatpak can do that, no problem. But that hasn't been the point.

There are other obstacles: xow needs to deploy udev rules in root context. flatpak cannot do that either.

flatpak is about sandboxing apps on purpose, no system-level interaction possible. You could probably implement some portal to circumvent such restrictions but that might not be a good idea to start with. Additionally, portals always come with performance overhead and would add (comparatively) massive input latency if you try it that way.

Also, where's the point in making it into a flatpak? flatpak has been made to decouple complex dependencies between system and application from each other. xow is not really something that has complex dependencies. It mostly requires libusb - that's it. Why would you want to ship a whole distribution runtime just for such a little peace of software?

Finally, xow is probably EOL anyways because of limitations of that implementation. xone should be used instead.