medusalix/xow #11: SteamOS

andreashuetter, 2019-12-23:
How can I build this on SteamOS?

At my first try it complained that the command "git" was not found. So I did a

`sudo apt-get install git`

Now I am getting this:

```
desktop@steamos:~/Downloads/xow-master$ make BUILD=RELEASE
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
g++  -Wall -Wpedantic -std=c++17 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o xow.o xow.cpp
g++: error: unrecognized command line option ‘-std=c++17’
Makefile:23: recipe for target 'xow.o' failed
make: *** [xow.o] Error 1
```

offalynne, 2019-12-23:
you can ignore the git error but you need to update g++: `sudo apt install build-essential`

andreashuetter, 2019-12-23:
Hi,

thanks for you help, but unfortunately it still doesn't work:

```
desktop@steamos:~$ sudo apt install build-essential
[sudo] password for desktop: 
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden zusätzlichen Pakete werden installiert:
  dpkg-dev fakeroot libdpkg-perl libfakeroot libtimedate-perl
Vorgeschlagene Pakete:
  debian-keyring
Empfohlene Pakete:
  libalgorithm-merge-perl libfile-fcntllock-perl
Die folgenden NEUEN Pakete werden installiert:
  build-essential dpkg-dev fakeroot libdpkg-perl libfakeroot libtimedate-perl
0 aktualisiert, 6 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 2.802 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 4.717 kB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] j
Holen: 1 http://repo.steampowered.com/steamos/ brewmaster/main libtimedate-perl all 2.3000-2+bsos3 [42,4 kB]
Holen: 2 http://repo.steampowered.com/steamos/ brewmaster/main libdpkg-perl all 1.17.27+bsos1 [1.075 kB]
Holen: 3 http://repo.steampowered.com/steamos/ brewmaster/main dpkg-dev all 1.17.27+bsos1 [1.548 kB]
Holen: 4 http://repo.steampowered.com/steamos/ brewmaster/main build-essential amd64 11.7+bsos2 [7.268 B]
Holen: 5 http://repo.steampowered.com/steamos/ brewmaster/main libfakeroot amd64 1.20.2-1+bsos3 [44,9 kB]
Holen: 6 http://repo.steampowered.com/steamos/ brewmaster/main fakeroot amd64 1.20.2-1+bsos3 [84,8 kB]
Es wurden 2.802 kB in 6 s geholt (448 kB/s).                                   
Vormals nicht ausgewähltes Paket libtimedate-perl wird gewählt.
(Lese Datenbank ... 114229 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../libtimedate-perl_2.3000-2+bsos3_all.deb ...
Entpacken von libtimedate-perl (2.3000-2+bsos3) ...
Vormals nicht ausgewähltes Paket libdpkg-perl wird gewählt.
Vorbereitung zum Entpacken von .../libdpkg-perl_1.17.27+bsos1_all.deb ...
Entpacken von libdpkg-perl (1.17.27+bsos1) ...
Vormals nicht ausgewähltes Paket dpkg-dev wird gewählt.
Vorbereitung zum Entpacken von .../dpkg-dev_1.17.27+bsos1_all.deb ...
Entpacken von dpkg-dev (1.17.27+bsos1) ...
Vormals nicht ausgewähltes Paket build-essential wird gewählt.
Vorbereitung zum Entpacken von .../build-essential_11.7+bsos2_amd64.deb ...
Entpacken von build-essential (11.7+bsos2) ...
Vormals nicht ausgewähltes Paket libfakeroot:amd64 wird gewählt.
Vorbereitung zum Entpacken von .../libfakeroot_1.20.2-1+bsos3_amd64.deb ...
Entpacken von libfakeroot:amd64 (1.20.2-1+bsos3) ...
Vormals nicht ausgewähltes Paket fakeroot wird gewählt.
Vorbereitung zum Entpacken von .../fakeroot_1.20.2-1+bsos3_amd64.deb ...
Entpacken von fakeroot (1.20.2-1+bsos3) ...
Trigger für man-db (2.7.0.2-5+bsos3) werden verarbeitet ...
libtimedate-perl (2.3000-2+bsos3) wird eingerichtet ...
libdpkg-perl (1.17.27+bsos1) wird eingerichtet ...
dpkg-dev (1.17.27+bsos1) wird eingerichtet ...
build-essential (11.7+bsos2) wird eingerichtet ...
libfakeroot:amd64 (1.20.2-1+bsos3) wird eingerichtet ...
fakeroot (1.20.2-1+bsos3) wird eingerichtet ...
update-alternatives: /usr/bin/fakeroot-sysv wird verwendet, um /usr/bin/fakeroot (fakeroot) im automatischen Modus bereitzustellen

desktop@steamos:~/Downloads/xow-master$ make BUILD=RELEASE
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
g++  -Wall -Wpedantic -std=c++17 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o xow.o xow.cpp
g++: error: unrecognized command line option ‘-std=c++17’
Makefile:23: recipe for target 'xow.o' failed
make: *** [xow.o] Error 1
```

To me, it seems that GCC / G++ in the SteamOS Repository is still too outdated for the xow sources, even after sudo apt install build-essential:

```
desktop@steamos:~/Downloads/xow-master$ gcc --version
gcc (SteamOS 4.9.2-10+deb8u2+bsos1) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

desktop@steamos:~/Downloads/xow-master$ g++ --version
g++ (SteamOS 4.9.2-10+deb8u2+bsos1) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Is there anything I could do (assuming that I want to stay on SteamOS and not install any other OS on my Steam Machine)?

medusalix, 2019-12-23:
I've just made some changes to make xow compatible with C++11 (which should be supported even by g++ 4.8). You should be able to build it now.

andreashuetter, 2019-12-23:
Thank you, this is my output now (I am using g++ 4.9), the compilation process is going a little bit further now, but still spits out an error:

```
desktop@steamos:~/Downloads/xow-master$ gcc --version
gcc (SteamOS 4.9.2-10+deb8u2+bsos1) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
desktop@steamos:~/Downloads/xow-master$ make
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o xow.o xow.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/mt76.o dongle/mt76.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/dongle.o dongle/dongle.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/usb.o dongle/usb.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o controller/controller.o controller/controller.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o controller/input.o controller/input.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o utils/log.o utils/log.cpp
utils/log.cpp: In function ‘void Log::printHeader(FILE*, std::string, std::string)’:
utils/log.cpp:58:19: error: ‘put_time’ is not a member of ‘std’
         stream << std::put_time(&localTime, "%F %T");
                   ^
Makefile:23: recipe for target 'utils/log.o' failed
make: *** [utils/log.o] Error 1

```

medusalix, 2019-12-23:
It seems like `std::put_time` was not [implemented](https://stackoverflow.com/a/14137287) until g++ 5.0. You could just remove the line where it's failing (it's just for printing out the time when logging occurs).

andreashuetter, 2019-12-23:
OK, I just commented this line out in utils/log.cpp
`        //stream << std::put_time(&localTime, "%F %T");`
as you suggested, and now it compiles!

But when I plug in my dongle and launch the xow executable, it crashed immediately with a segmenation fault.

`sudo dmseg` gives the following output:
```
[ 1288.673388] usb 1-2: new high-speed USB device number 7 using xhci_hcd
[ 1288.826644] usb 1-2: New USB device found, idVendor=045e, idProduct=02e6, bcdDevice= 1.00
[ 1288.826649] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1288.826652] usb 1-2: Product: XBOX ACC
[ 1288.826654] usb 1-2: Manufacturer: Microsoft Inc.
[ 1288.826657] usb 1-2: SerialNumber: 010168
[ 1288.981804] usb 1-2: reset high-speed USB device number 7 using xhci_hcd
[ 1289.134577] mt76x2u 1-2:1.0: ASIC revision: 76320044
[ 1289.156950] mt76x2u 1-2:1.0: firmware: failed to load mediatek/mt7662u_rom_patch.bin (-2)
[ 1289.156952] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[ 1289.156953] mt76x2u 1-2:1.0: Direct firmware load for mediatek/mt7662u_rom_patch.bin failed with error -2
[ 1289.767074] mt76x2u: probe of 1-2:1.0 failed with error -2
[ 1289.767091] usbcore: registered new interface driver mt76x2u
[ 1411.988255] usb 1-2: reset high-speed USB device number 7 using xhci_hcd
[ 1412.147095] xow[2696]: segfault at 0 ip 00007fbae5a956ba sp 00007fbae418eb50 error 6 in libusb-1.0.so.0.1.0[7fbae5a8b000+17000]
[ 1412.147108] Code: 49 89 47 20 e8 27 9a ff ff 85 c0 0f 88 bf 00 00 00 4c 89 ff e8 e7 fe ff ff 45 8b 47 10 41 8b 47 18 48 8b 4c 24 08 41 83 f8 06 <89> 01 0f 87 b6 00 00 00 48 8d 05 17 77 00 00 4a 63 14 80 48 01 d0
```

I thought the firmware.bin that is included in the xow sources would contain the approbriate firmware and would be linked into the executable at compile time? Or does the dmesg output mean that I need to download the specific mt76x2u firmware vom mediatek and copy it to /lib/firmware?

medusalix, 2019-12-23:
> I thought the firmware.bin that is included in the xow sources would contain the appropriate firmware and that it would be linked into the executable automatically at compile time?

Yes, that's absolutely correct. The output in `dmesg` is actually related to Linux' built in `mt76` kernel driver. It conflicts with xow, leading to its crash. Please try adding this piece of code right **before** the [call](https://github.com/medusalix/xow/blob/059e72298933b5ace91ab353ac0ecf3bc29ff19d/dongle/usb.cpp#L66) to `libusb_claim_interface`:

```cpp
libusb_set_auto_detach_kernel_driver(handle, true);
```

andreashuetter, 2019-12-23:
Well, for some reason it still does not work.

I also added 

`libusb_set_auto_detach_kernel_driver(handle, true);`

`in dongle/usb.cpp, just before error = libusb_claim_interface(handle, 0);`

Then I did
```
desktop@steamos:~/Downloads/xow-master$ make clean
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
rm -f xow xow.o dongle/mt76.o dongle/dongle.o dongle/usb.o controller/controller.o controller/input.o utils/log.o firmware.o xow.d dongle/mt76.d dongle/dongle.d dongle/usb.d controller/controller.d controller/input.d utils/log.d firmware.d
desktop@steamos:~/Downloads/xow-master$ make BUILD=RELEASE
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
g++  -Wall -Wpedantic -std=c++11 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o xow.o xow.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/mt76.o dongle/mt76.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/dongle.o dongle/dongle.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/usb.o dongle/usb.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o controller/controller.o controller/controller.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o controller/input.o controller/input.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -O3 -DCHANNEL=1 -DVERSION=\"\" -c -o utils/log.o utils/log.cpp
utils/log.cpp: In function ‘void Log::printHeader(FILE*, std::string, std::string)’:
utils/log.cpp:55:17: warning: variable ‘localTime’ set but not used [-Wunused-but-set-variable]
         std::tm localTime = *std::localtime(&time);
                 ^
ld -r -b binary -o firmware.o firmware.bin
cc   xow.o dongle/mt76.o dongle/dongle.o dongle/usb.o controller/controller.o controller/input.o utils/log.o firmware.o  -lstdc++ -lusb-1.0 -lpthread -o xow
desktop@steamos:~/Downloads/xow-master$ sudo make install
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
cp xow /usr/local/bin
cp xow.service /etc/systemd/user
systemctl enable /etc/systemd/user/xow.service
systemctl start xow
```

again to be sure to have everything in a clean state.

But I get this as a result:
```
desktop@steamos:~/Downloads/xow-master$ systemctl status xow -l
● xow.service - Xbox One Wireless Dongle Driver
   Loaded: loaded (/etc/systemd/user/xow.service; enabled)
   Active: failed (Result: signal) since Mo 2019-12-23 15:05:57 CET; 1min 12s ago
  Process: 2825 ExecStart=/usr/local/bin/xow (code=killed, signal=SEGV)
 Main PID: 2825 (code=killed, signal=SEGV)
```

Note: I tested with both of my dongles (0x02e6 aka the old AND bulky one, and 0x02fe aka the new smaller one). I also tested them on two different Steam Machines, with no success.

medusalix, 2019-12-23:
Please run xow in the foreground (and make sure the service is stopped), build it with `BUILD=DEBUG` and post the logs.

andreashuetter, 2019-12-23:
OK, I did, but it doesn't produce any additonal logs:

```
desktop@steamos:~/Downloads/xow-master$ make clean
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
rm -f xow xow.o dongle/mt76.o dongle/dongle.o dongle/usb.o controller/controller.o controller/input.o utils/log.o firmware.o xow.d dongle/mt76.d dongle/dongle.d dongle/usb.d controller/controller.d controller/input.d utils/log.d firmware.d
desktop@steamos:~/Downloads/xow-master$ make BUILD=DEBUG
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o xow.o xow.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/mt76.o dongle/mt76.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/dongle.o dongle/dongle.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o dongle/usb.o dongle/usb.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o controller/controller.o controller/controller.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o controller/input.o controller/input.cpp
g++  -Wall -Wpedantic -std=c++11 -MMD -Og -g -DDEBUG -DCHANNEL=1 -DVERSION=\"\" -c -o utils/log.o utils/log.cpp
utils/log.cpp: In function ‘void Log::printHeader(FILE*, std::string, std::string)’:
utils/log.cpp:55:17: warning: variable ‘localTime’ set but not used [-Wunused-but-set-variable]
         std::tm localTime = *std::localtime(&time);
                 ^
ld -r -b binary -o firmware.o firmware.bin
cc   xow.o dongle/mt76.o dongle/dongle.o dongle/usb.o controller/controller.o controller/input.o utils/log.o firmware.o  -lstdc++ -lusb-1.0 -lpthread -o xow
desktop@steamos:~/Downloads/xow-master$ sudo make install
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
cp xow /usr/local/bin
cp xow.service /etc/systemd/user
systemctl enable /etc/systemd/user/xow.service
systemctl start xow
desktop@steamos:~/Downloads/xow-master$ sudo ./xow
 INFO  - xow  ©Severin v. W.
 INFO  - Dongle plugged in
desktop@steamos:~/Downloads/xow-master$ echo $?
139
```

It just crashes after printing "Dongle plugged in" and the return code is 139.

andreashuetter, 2019-12-23:
I made some additional tests on my Desktop PC (running Manjaro Linux 18) and there I got the new, smaller dongle to run (even without the above mentioned SteamOS specific workarounds):

```
[andreas@erazor xow-master]$ sudo ./xow
2019-12-23 15:17:48 INFO  - xow  ©Severin v. W.
2019-12-23 15:17:48 INFO  - Dongle plugged in
2019-12-23 15:17:48 INFO  - Chip address: 62:45:b4:fe:f1:d0
2019-12-23 15:17:48 INFO  - Dongle initialized
2019-12-23 15:17:50 INFO  - Controller '1' connected
2019-12-23 15:17:50 INFO  - Serial number: 02600238452701
```

The older, bulkier one crashes on Manjaro with
```
[andreas@erazor xow-master]$ sudo ./xow
2019-12-23 15:32:03 INFO  - xow  ©Severin v. W.
2019-12-23 15:32:05 INFO  - Dongle plugged in
xow: io.c:2146: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
```

On SteamOS still, with both types of dongles I don't get far:
```
desktop@steamos:~/Downloads/xow-master$ sudo ./xow
 INFO  - xow  ©Severin v. W.
 INFO  - Dongle plugged in
desktop@steamos:~/Downloads/xow-master$ echo $?
139
```

medusalix, 2019-12-23:
I think the version of `libusb` you have installed on SteamOS probably contains some kind of bug. You should probably try to get a newer version.

andreashuetter, 2019-12-23:
Which version of libusb would be appropriate for xow?

On SteamOS initially when I tried to compile I got the message "libusb.h" not found, so I installed

`sudo apt-get install libusb-1.0-0-dev`

Maybe that version is outdated.

On Manjaro 18 on the other hand I didn't need to install libusb, because it was already there. But when I perform a `find` command to determine which version is present there, it seems to be the same:

```
...
/usr/lib/libusb-1.0.so.0.2.0
/usr/lib/libusb-1.0.so.0
/usr/lib/libusbmuxd.so.6.0.0
/usr/lib/libusbmuxd.so
/usr/lib/wine/libusbd.def
/usr/lib/libusbmuxd.so.6
/usr/lib/libusb-1.0.so
/usr/lib/pkgconfig/libusb-1.0.pc
/usr/lib/pkgconfig/libusbmuxd.pc
/usr/lib32/libusb-1.0.so.0.2.0
/usr/lib32/libusb-1.0.so.0
/usr/lib32/wine/libusbd.def
/usr/lib32/libusb-1.0.so
/usr/lib32/pkgconfig/libusb-1.0.pc
/usr/include/libusb-1.0
/usr/include/libusb-1.0/libusb.h
```

medusalix, 2019-12-23:
That's not the actual version of the package though. You should run `dpkg -s libusb-1.0-0-dev` to see the installed version. I'm running libusb 1.0.23-1.

andreashuetter, 2019-12-23:
OK, seems that SteamOS has libusb 1.0.19-1.
```
desktop@steamos:~$ dpkg -s libusb-1.0-0-dev
Package: libusb-1.0-0-dev
Status: install ok installed
Priority: optional
Section: libdevel
Installed-Size: 293
Maintainer: Aurelien Jarno <aurel32@debian.org>
Architecture: amd64
Multi-Arch: same
Source: libusb-1.0
Version: 2:1.0.19-1+bsos2
Depends: libusb-1.0-0 (= 2:1.0.19-1+bsos2)
Recommends: libusb-1.0-doc
```

What's the best way to obtain a newer version if this is the only version in the SteamsOS repository? Download the sources from https://github.com/libusb/libusb and compile them manually?

medusalix, 2019-12-23:
As SteamOS is based on Debian, you could try to add the Debian package repository to your system (no idea how to do this). Oh, and by the way, libusb 1.0.19 is around [5 years](https://github.com/libusb/libusb/blob/b454b9a4d3f9b55ea26cd28c8bfe4dc547b6e1cb/ChangeLog#L78) old 😄.

andreashuetter, 2019-12-23:
I have good news!
After downloading the latest version of libusb from https://github.com/libusb/libusb and compiling it manually I was able to get xow to run on SteamOS and it also works with BOTH of my dongles (the older bulky one, and the new smaller one).

Strange thing, though, is that after rebooting the machine it does not work anymore. The LED is constantly glowing during the reboot process, but it won't work anymore. Only thing I can do is to completely shutdown the machine AND also cut the power cord off. When I then turn on again, the LED does not glow during the boot process and then after the boot process starts to glow and then it works.

Der-Eddy, 2019-12-24:
> As SteamOS is based on Debian, you could try to add the Debian package repositories to your system (no idea how to do this). Oh, and by the way, libusb 1.0.19 is around [5 years](https://github.com/libusb/libusb/blob/b454b9a4d3f9b55ea26cd28c8bfe4dc547b6e1cb/ChangeLog#L78) old smile.

Yea because SteamOS is based on Debian 8 Jessie (latest is Debian 10 Buster), which is already 5 years old  
unfortunately Valve doesn't really update it anymore

there are also some weird issues due to the really old Debian version like some newer games not working (e.g. Dead Cells, Ruiner) due to the old glibc version or you loose potential performance improvements because of the very old nvidia drivers or older Mesa/Kernel for AMD users  
not to mention the poor hardware support for anything released in the recent 5 years

Fortunately the community took care for alternatives:

- https://github.com/ShadowApex/steamos-ubuntu for a SteamOS like Ubuntu
- https://gamer-os.github.io/ based on a rolling release
- or install any newer linux distro and run Steam Big Picture at start, there is also a community updated [steamos-compositor package](https://github.com/gamer-os/steamos-compositor-plus) available with more options than the original one from Valve at the moment

I'm really sorry for going a bit off topic here but making SteamOS up2date or viable with anything recent is just not worth the time at the moment

medusalix, 2019-12-25:
@Der-Eddy Thanks for the writeup, I'm sure this will help people with similar issues in the future.

@andreashuetter Awesome 😃. I've pushed a commit (d0341e42d67344ffb4a944c130675ff366d27220) that should fix your reboot issue.

closed 2019-12-25 by medusalix

andreashuetter, 2019-12-25:
Thanks a lot, but there are 2 issues:

1. To compile the new version it was necessary again to edit utils/log.cpp and comment out the line
`stream << std::put_time(&localTime, "%F %T");`
It is quite annoying to manually comment out this particular line every time I want to test a new version of xow.
Would it be possible to use strftime instead of put_time (as suggested here https://lab.louiz.org/louiz/biboumi/commit/a195939cdaca343013d32ae902839ffdb28c33cb) to make it compilable even on SteamOS (gcc 4.9)?

2.The line
`libusb_set_auto_detach_kernel_driver(handle, true);`
is still missing in dongle/usb.cpp (right before the call to `libusb_claim_interface`).
Is this line suppossed to be added manually every time when a new version is pushed?

medusalix, 2019-12-25:
1. I don't want to go back to raw C strings just to make `strftime` work (and hardcode the string length). I think C++11 is a reasonably old standard to tailor the code to. You could always make yourself a git `patch` and apply that whenever you run xow.

2. I'm not sure if that's even needed anymore. I only assumed it would solve your problems, please try running xow without it.

andreashuetter, 2019-12-25:
Well, I made some addtional tests now.

With the new (smaller) dongle the reboot issue sems to be fixed now. Great work!
Even the line
`libusb_set_auto_detach_kernel_driver(handle, true);`
doesn't seem to be needed anymore in dongle/usb.cpp.

But the older, bulky dongle doesn't ssem to work reliably with the latest version of xow anymore. I will do more tests with the current version and the previous version to get more into detail.

As for the put_time / strftime issue in log.cpp - may I suggest to handle this with a preprocessor directive, something like this?
` #if __GNUC__ > 4.9 ...`