medusalix/xow #12: Elite Wireless Support

rwilliams, 2019-12-25:
I have the Original Elite Wireless and I'd be more than happy to test.

PopeRigby, 2019-12-25:
Why did you close this?

rwilliams, 2019-12-25:
It seems like it's already a pretty well known limitation after looking at the other threads. I'll re-open though.

PopeRigby, 2019-12-25:
Yeah. You should leave it open so we can track the issue.

Hielyr, 2019-12-27:
> Yeah. You should leave it open so we can track the issue.

Right. It's a feature request, not a bug.

rwilliams, 2020-01-13:
I'd be more than happy to do a packet capture for this with Wireshark if that would be helpful though I'll probably need a bit of guidance on getting the proper data.

medusalix, 2020-01-13:
Yes, that would definitely be helpful. What happens when you try to connect/pair your controller using xow on Linux?

rwilliams, 2020-01-13:
> xow on  master 
➜ sudo ./xow
2020-01-13 12:20:50 INFO  - xow v0.2-5-gcbe6761 ©Severin v. W.
2020-01-13 12:20:50 INFO  - Dongle plugged in
2020-01-13 12:20:50 INFO  - Chip address: 62:45:b5:0b:84:bb
2020-01-13 12:20:50 INFO  - Dongle initialized
2020-01-13 12:21:07 INFO  - Pairing initiated
2020-01-13 12:21:07 INFO  - Pairing initiated
2020-01-13 12:21:08 INFO  - Pairing initiated
2020-01-13 12:21:08 INFO  - Pairing initiated
2020-01-13 12:21:29 INFO  - Pairing initiated
2020-01-13 12:21:29 INFO  - Pairing initiated
2020-01-13 12:21:29 INFO  - Pairing initiated
2020-01-13 12:21:29 INFO  - Pairing initiated

I hit paring twice here. I'm a little behind on the build but i don't think you've added any related changes since.

rwilliams, 2020-01-13:
I just updated to master for science, same results.

rwilliams, 2020-01-13:
For Wireshark do I just select all the USBPcap options and go through a pairing cycle?

medusalix, 2020-01-13:
Yes, first you have to find the correct USBPcap device. If you plug in the dongle and you see lots of messages appearing, then it's probably the correct one. Then you pair your controller, generate some inputs (buttons, triggers, bumpers) and disconnect the controller.

rwilliams, 2020-01-13:
Ok. I think i got the filters right to only capture the receiver and controller. The first capture is turning the controller on, already paired. Doing a pairing cycle and then hitting a bunch of buttons. The second capture is specifically for the paddles on the underside. Starting with Top right(B)
, Bottom right(B), Top left(X), Bottom Left(X). `I only actually use one paddle on each side for rocket league so that's why the bindings are the same.` I've attached the config from xbox accessories app. I'm more than happy to redo this with different bindings if needed.

[elitewireless1.pcapng.gz](https://github.com/medusalix/xow/files/4056238/elitewireless1.pcapng.gz)
[paddles.pcapng.gz](https://github.com/medusalix/xow/files/4056240/paddles.pcapng.gz)
[image]

medusalix, 2020-01-14:
I've looked through your captures and didn't spot any obvious differences that might prevent it from pairing. Could you do another capture on Linux please (with xow open)? The debug logs might also be helpful (`make` with `BUILD=DEBUG`). Please make sure to wait for around 20 seconds after you press the pairing button (the pairing can be quite slow).

rwilliams, 2020-01-14:
Will do

rwilliams, 2020-01-14:
https://wiki.wireshark.org/CaptureSetup/USB#Linux

I have no idea what to choose here to filter.
[image]

I think when I chose any it was getting data from the dongle but i'd like to get a capture that's more specific. Where are the logs for xow located?

medusalix, 2020-01-14:
You have to run `sudo modprobe usbmon` for USB devices to appear in Wireshark.

rwilliams, 2020-01-14:
Thanks! Will post soon.

rwilliams, 2020-01-15:
```
xow on  master took 31s 
➜ sudo ./xow
2020-01-15 04:18:56 INFO  - xow v0.2-21-g6de45b3 ©Severin v. W.
2020-01-15 04:18:57 INFO  - Dongle plugged in
2020-01-15 04:18:57 INFO  - Chip address: 62:45:b5:0b:84:bb
2020-01-15 04:18:57 INFO  - Dongle initialized
2020-01-15 04:19:02 INFO  - Pairing initiated
2020-01-15 04:19:02 INFO  - Pairing initiated
2020-01-15 04:19:02 INFO  - Pairing initiated
2020-01-15 04:19:02 INFO  - Pairing initiated
2020-01-15 04:19:21 INFO  - Controller '1' connected
2020-01-15 04:19:21 INFO  - Controller '1' disconnected
2020-01-15 04:19:42 INFO  - Controller '1' connected
2020-01-15 04:22:44 ERROR - Error in bulk read: 1
2020-01-15 04:22:44 ERROR - Error in bulk read: 1
2020-01-15 04:22:44 INFO  - Dongle removed
^C

xow on  master took 3m 52s 
➜ sudo ./xow
2020-01-15 04:23:01 INFO  - xow v0.2-21-g6de45b3 ©Severin v. W.
2020-01-15 04:23:01 INFO  - Dongle plugged in
2020-01-15 04:23:01 INFO  - Chip address: 62:45:b5:0b:84:bb
2020-01-15 04:23:01 INFO  - Dongle initialized
2020-01-15 04:23:01 INFO  - Controller '1' connected
```

Well it looks like it pairs. It shows up properly in steam. None of the inputs seem to work though. I got usbmon going but i didn't see any action on any of the usbmon filters that seemed like they were from the dongle when I looked at them.

medusalix, 2020-01-15:
Looks like the pairing works just fine (sometimes it takes a while for controllers to connect, even on Windows). The input doesn't work because the packet the elite controllers sends is a little bit bigger than the normal input (due to the additional paddles) and xow ignores packets with an incorrect size. That shouldn't be too hard to fix though...

medusalix, 2020-01-17:
@rwilliams What inputs do the paddles generate when you plug in the controller over USB? If you run `evtest` you should be able to see all the events generated.

rwilliams, 2020-01-17:
This is what I expected. There is no difference between hitting b or a paddle mapped to b. I tested for x too. Can test more bind combinations to look for edge cases if you need. In this case my b and x were east and north respectively.

> Supported events:
>   Event type 0 (EV_SYN)
>   Event type 1 (EV_KEY)
>     Event code 304 (BTN_SOUTH)
>     Event code 305 (BTN_EAST)
>     Event code 307 (BTN_NORTH)
>     Event code 308 (BTN_WEST)
>     Event code 310 (BTN_TL)
>     Event code 311 (BTN_TR)
>     Event code 314 (BTN_SELECT)
>     Event code 315 (BTN_START)
>     Event code 316 (BTN_MODE)
>     Event code 317 (BTN_THUMBL)
>     Event code 318 (BTN_THUMBR)
>   Event type 3 (EV_ABS)
>     Event code 0 (ABS_X)
>       Value     93
>       Min   -32768
>       Max    32767
>       Fuzz      16
>       Flat     128
>     Event code 1 (ABS_Y)
>       Value   -208
>       Min   -32768
>       Max    32767
>       Fuzz      16
>       Flat     128
>     Event code 2 (ABS_Z)
>       Value      0
>       Min        0
>       Max     1023
>     Event code 3 (ABS_RX)
>       Value    637
>       Min   -32768
>       Max    32767
>       Fuzz      16
>       Flat     128
>     Event code 4 (ABS_RY)
>       Value   -192
>       Min   -32768
>       Max    32767
>       Fuzz      16
>       Flat     128
>     Event code 5 (ABS_RZ)
>       Value      0
>       Min        0
>       Max     1023
>     Event code 16 (ABS_HAT0X)
>       Value      0
>       Min       -1
>       Max        1
>     Event code 17 (ABS_HAT0Y)
>       Value      0
>       Min       -1
>       Max        1
>   Event type 21 (EV_FF)
>     Event code 80 (FF_RUMBLE)
>     Event code 81 (FF_PERIODIC)
>     Event code 88 (FF_SQUARE)
>     Event code 89 (FF_TRIANGLE)
>     Event code 90 (FF_SINE)
>     Event code 96 (FF_GAIN)
> Properties:
> Testing ... (interrupt to exit)
> Event: time 1579266373.053129, type 3 (EV_ABS), code 1 (ABS_Y), value -140
> Event: time 1579266373.053129, -------------- SYN_REPORT ------------
> Event: time 1579266373.381130, type 3 (EV_ABS), code 4 (ABS_RY), value -125
> Event: time 1579266373.381130, -------------- SYN_REPORT ------------
[... 159 lines cut ...]
> Event: time 1579266407.485433, type 3 (EV_ABS), code 4 (ABS_RY), value -309
> Event: time 1579266407.485433, -------------- SYN_REPORT ------------
> Event: time 1579266407.493446, type 3 (EV_ABS), code 1 (ABS_Y), value -204
> Event: time 1579266407.493446, type 3 (EV_ABS), code 3 (ABS_RX), value 1202
> Event: time 1579266407.493446, -------------- SYN_REPORT ------------
> Event: time 1579266407.501446, type 3 (EV_ABS), code 3 (ABS_RX), value 1310
> Event: time 1579266407.501446, -------------- SYN_REPORT ------------
> Event: time 1579266407.509447, type 3 (EV_ABS), code 1 (ABS_Y), value -117
> Event: time 1579266407.509447, -------------- SYN_REPORT ------------
> Event: time 1579266407.517446, type 3 (EV_ABS), code 3 (ABS_RX), value 1452
> Event: time 1579266407.517446, -------------- SYN_REPORT ------------
> Event: time 1579266407.537447, type 3 (EV_ABS), code 1 (ABS_Y), value -42
> Event: time 1579266407.537447, type 3 (EV_ABS), code 3 (ABS_RX), value 1556
> Event: time 1579266407.537447, type 3 (EV_ABS), code 4 (ABS_RY), value -242
> Event: time 1579266407.537447, -------------- SYN_REPORT ------------
> Event: time 1579266407.561447, type 3 (EV_ABS), code 3 (ABS_RX), value 1657
> Event: time 1579266407.561447, -------------- SYN_REPORT ------------
> Event: time 1579266407.601447, type 3 (EV_ABS), code 1 (ABS_Y), value 24
> Event: time 1579266407.601447, -------------- SYN_REPORT ------------
> Event: time 1579266407.917450, type 3 (EV_ABS), code 3 (ABS_RX), value 1749
> Event: time 1579266407.917450, -------------- SYN_REPORT ------------
> Event: time 1579266408.225439, type 3 (EV_ABS), code 4 (ABS_RY), value -175
> Event: time 1579266408.225439, -------------- SYN_REPORT ------------
> Event: time 1579266410.461458, type 3 (EV_ABS), code 0 (ABS_X), value 30
> Event: time 1579266410.461458, -------------- SYN_REPORT ------------

medusalix, 2020-01-17:
> There is no difference between hitting b or a paddle mapped to b.

That looks like Microsoft actually implemented the button remapping in hardware. Based on your paddle captures it seems like we can just treat the bigger input packets as those of a normal Xbox One controller in the code. Please checkout the [`elite`](https://github.com/medusalix/xow/tree/elite) and let me know if it works 😃.

rwilliams, 2020-01-17:
Will do. Yeah sorry for not mentioning earlier that's it's all handled via firmware/hardware internally. I have to map the buttons in Windows but the mappings work without software support just fine in linux.

rwilliams, 2020-01-17:
Looks good. Pairing still takes some effort though. Maybe the elite wireless 2 will just work now.

medusalix, 2020-01-17:
Yeah, both elite controller models should work now. Have you got an elite v2 that you can test this with?

rwilliams, 2020-01-17:
I don't unfortunately. I've been pretty happy with my original elite so it's hard to justify dropping the money on the updated one.

medusalix, 2020-01-17:
Yes, it's **very** expensive and the built-in battery would be a dealbreaker for me. I've merged the changes into master in b58edb4ce50910fb7401c3d31919b2cc448aaaba, so I'll close this now.

closed 2020-01-17 by medusalix

Hielyr, 2020-01-31:
I have a v2 I can test with if necessary. I'll work on seeing if it works with the default build with the v1 changes added, and if not we can go from there.

Hielyr, 2020-02-01:
> I have a v2 I can test with if necessary. I'll work on seeing if it works with the default build with the v1 changes added, and if not we can go from there.

I have confirmed that the v2 Elite controller works as well! 
The model of the controller I'm using is 1797.

rwilliams, 2020-02-01:
Off topic but if I can resell or gift my V1 is the V2 worth getting?

Hielyr, 2020-02-01:
> Off topic but if I can resell or gift my V1 is the V2 worth getting?

I've never owned a v1, so I'm unable to make a direct comparison. 
What convinced me to get one for myself was this video review by Tim Rogers: https://www.youtube.com/watch?v=PhrNA07kzjc

He goes into a pretty decent comparison between the the two controllers.