medusalix/xow #15: License and source of firmware.bin

samueloph, 2019-12-27:
Hello,

I'm working in the packaging of xow for Debian and would like some clarification about the file `firmware.bin`.

Could you explain where/how did you get `firmware.bin`?

I'm mostly interested in its source code and license.

Beware that if you don't own the file and the licensing, you need to clarify it, otherwise it's as if you were saying that the file is owned by you and licensed under GPL-2.

medusalix, 2019-12-28:
Thanks for bringing that question up. The `firmware.bin` was actually copied from my Windows machine (`C:\Windows\System32\DriverStore\FileRepository\mt7612us_rl.inf_amd64_75ab4556628c9849\FW_ACC_00U.bin`). I've found two projects on GitHub that share firmware for similar Mediatek chipsets ([`openwrt`](https://github.com/openwrt/mt76/blob/master/firmware/LICENSE) and [`linux-firmware`](https://github.com/wkennington/linux-firmware/blob/master/LICENCE.mediatek)) but they use different license disclaimers. Am I allowed to include my `firmware.bin` if I put one of those disclaimers in the repository or do I need written consent from Mediatek to do so?

samueloph, 2019-12-30:
@medusalix 

I found this discussion at https://github.com/paroj/xpad/issues/25 which you've probably already seem it, it doesn't help much but I guess it' s worth linking it here.

Dealing with this blob seems like a PITA (no sources/no clear license), it seems weird that you found it with two different licenses and that might be an indication that something's wrong.

I'm not a lawyer, and so this is not a legal advice, but it looks like you could get the blob from any of these projects and just use the same license, as their license seems to allows it. And in this case you should point to where you've got the blob from.
 
Now, that being said, considering they have different licenses, it's probably more safe to contact Mediatek (like you did) to clarify it.

Having the license for the blob would be enough for the package to go into the non-free section of Debian. It would be ideal if they could introduce the firmware to linux directly (as linux has other blobs already), or even better, open source it, but I don' t have high hopes here.

Thanks for your work on this, looking forward to use my controller wirelessly on my system!

medusalix, 2020-01-05:
As MediaTek apparently doesn't bother answering my mails (sent another one through their contact form a few days ago), I'm including my message here for reference:

[image]

If they don't reply within the next week, I'll assume they're fine with me distributing their firmware under the included disclaimer ([LICENSE-FIRMWARE](https://github.com/medusalix/xow/blob/master/LICENSE-FIRMWARE)).

medusalix, 2020-01-12:
@samueloph I think it's safe to say that MediaTek doesn't care about my project as of now (got no answer at all). Unless you find a way to reach out to one of their developers/managers/employees, I would just go ahead and publish the Debian package. I added disclaimers to the README and the releases page and I guess there's nothing more I could possibly do to be on the safe side (legally).

samueloph, 2020-01-18:
@medusalix ack, I'm gonna go ahead and work on a package based on the git snapshot of the project (since there' s lots of important changes coming in as I see) for the non-free section.

Thanks for sorting this out :)

Feel free to close this issue if you'd like, I reckon there was a chance you'd like to leave it open so I didn't do it.

medusalix, 2020-01-18:
@samueloph Awesome 😃. Please let me know if the `Makefile` needs any modifications in order to make the Debian packaging easier for you. Ideally, I'd like to have version 0.3 available on the Debian repositories.

samueloph, 2020-01-25:
@medusalix just saw 0.3 released, that' s great, I will finish the packaging between now and the MiniDebcamp that we're going to have just before FOSDEM.

samueloph, 2020-02-20:
Update: I ended up not doing it in FOSDEM, but now I finished it and just uploaded to the Debian's NEW queue, now I need to wait for the ftp-master approval of the package, this usually take between 2 days and a month to happen.

medusalix, 2020-02-21:
Thank you so much for your work, I'm really looking forward to trying it out 😃.

samueloph, 2020-03-23:
@medusalix I was discussing this with a person working on the non-free side of Debian and he called out something that I accidentally missed when discussing this issue.

Again, regular disclaimer that I'm not a lawyer and this is not legal advice, as usual.

I had pointed out that:
"...it looks like you could get the blob from any of these projects and just use the same license, as their license seems to allow it. And in this case you should point to where you've got the blob from."

And then I thought the issue was solved when you described the license of the file, but I failed to notice when you said:
"It seems to be a firmware file from 2017 for the MT7612U which you can't find on the internet. I've found two projects on GitHub that share firmware for similar MediaTek chipsets"

Which means those licenses are for different files and thus not what I thought had happened (to get the blob from another project, keeping the same license),

I recommend trying to find this specific blob in some other project and find out what's its license, or to get a confirmation from Mediatek itself, maybe contacting people who signed off this: https://github.com/wkennington/linux-firmware/commit/3fcf221a7fbc317052802776613b8f021195179b to get a license for the blob you use. The point is to get evidence/confirmation that this is the license the copyright holders intend the file to be.

Regards,

medusalix, 2020-03-23:
@samueloph I've just noticed that the whole driver package is actually available from [Microsoft's update catalog](https://www.catalog.update.microsoft.com). Would it be feasible to download the driver from Microsoft's servers at installation time? Extracting the firmware from the driver's `cab` is quite easy:

```sh
# Xbox - Net - 7/11/2017 12:00:00 AM - 1.0.46.1 from Update Catalog
wget -O driver.cab http://download.windowsupdate.com/c/msdownload/update/driver/drvs/2017/07/1cd6a87c-623f-4407-a52d-c31be49e925c_e19f60808bdcbfbd3c3df6be3e71ffc52e43261e.cab
7z x driver.cab FW_ACC_00U.bin
echo 48084d9fa53b9bb04358f3bb127b7495dc8f7bb0b3ca1437bd24ef2b6eabdf66 FW_ACC_00U.bin | sha256sum -c
```

If this is not possible I'll try to contact them again.

samueloph, 2020-03-24:
@medusalix your suggestion of downloading it at install time would not solve the issue as the license of the file might prohibit even that (to use the blob in a different way).

You should maybe contact people from openwrt or the linux kernel who maintains the mediatek chip's support, as it could be that you can use the same file as them or perhaps they can contact mediatek and provide support directly from linux.

I can see that the package https://packages.debian.org/sid/firmware-misc-nonfree (which comes from linux) mentions support for ` * MediaTek MT76x2u wifi firmware (mediatek/mt7662u.bin)` which would be the file `/lib/firmware/mediatek/mt7662u.bin` but the checksum doesn't match so you'd have to check if that works.

Maybe the best scenario here is to find out that you can use some blob from linux or that the maintainers can help with adding support there, that way xow would only need the linux module.

Also, I found this pages which mentions that linux already supports the chipset which leads me to believe that the same blob could be used instead but you will know better than me.

https://cateee.net/lkddb/web-lkddb/MT76x2U.html
https://wireless.wiki.kernel.org/en/users/drivers/mediatek

medusalix, 2020-03-24:
@samueloph The firmware included in `linux-firmware` does unfortunately lack some of the functionality that xow needs. I don't know what modifications MediaTek made to make it compatible with Xbox controllers.

@nbd168 Do you know how to contact a MediaTek employee in charge of MT76 development? I tried reaching out to Sean Wang and multiple other people via mail but received no response. I assume you also had to get permission from MediaTek for your firmware files and I'd be thankful if you could help me in this regard.

medusalix, 2020-05-08:
@LorenzoBianconi Could you comment on this issue?

x4mers, 2020-05-09:
I take it you've already looked at this, so I don't know if it's of any use, but the Linux package available on their Chinese site has source code in it, with firmware content. It may just be for the ones you said are incompatible though.

https://www.mediatek.cn/products/broadbandWifi/mt7612u

medusalix, 2020-05-13:
> It may just be for the ones you said are incompatible though.

Yeah, unfortunately.

---

A quick update on the current situation: I wrote another mail to Sean Wang and his colleague Ryder Lee from a different email address and got a response:

[image]

Let's see whether I'll be allowed to keep the firmware in this repository!

Neverest, 2020-05-24:
Don't know if its relevant, but they are adding support for MT76x2 on linux kernel 5.8.

https://git.kernel.org/pub/scm/linux/kernel/git/kvalo/wireless-drivers-next.git/commit/?id=eda31200e68d38fbb974e7ad02bcc2de2cfe6863

medusalix, 2020-06-09:
> Don't know if its relevant, but they are adding support for MT76x2 on linux kernel 5.8.

Support for `mt76` wireless radios has been in the kernel much longer. We only require Microsoft's proprietary firmware for this project, so we'll have to wait for MediaTek's response.