medusalix/xow #16: Possible to sync in both windows and linux?

Hedronmx, 2019-12-31:
I used the xbox 360 adapter and with that adapter i could sync once and my controllers would work in both windows and ubuntu.

Could it be possible to do that? One sync to rule them all.

medusalix, 2019-12-31:
I'm paired on both Windows and Linux just fine. I turn on the controller and it connects automatically. So apparently it works with the Xbox One adapter too.

Hedronmx, 2019-12-31:
Do you sync fist in windows? Because i have to pair my controllers if I switch OS

andreashuetter, 2019-12-31:
When I pair my controllers on windows then they also work on Linux. I don't have to pair them again in Linux.

Hedronmx, 2019-12-31:
I just realized the service was having problem starting. I just restarted the service and now they pair! Awesome!

closed 2019-12-31 by Hedronmx

andreashuetter, 2019-12-31:
If your service has sometimes troubles starting (and sometimes not) maybe you should take a look on this issue:
https://github.com/medusalix/xow/issues/14

Hedronmx, 2020-01-01:
I am using the new dongle tho... I should not need to blacklist the module... right?

andreashuetter, 2020-01-01:
Try
`lsmod|grep mt76`
and
`sudo dmesg|grep mt76`

What does it say? If the outputs are empty then you don't need to blacklist the module.