medusalix/xow #18: Make receiver's LED turn off when no controller is connected

Hedronmx, 2020-01-04:
Is this possible?

andreashuetter, 2020-01-04:
> Is this possible?

Isn't it better the way it is now: The LED shows whether the xow service is running (and connected to the dongle) or not?

NoXPhasma, 2020-01-04:
> Isn't it better the way it is now: The LED shows whether the xow service is running (and connected to the dongle) or not?

That's not how it works now, the LED keeps off even if the driver runs until a controller is connected the first time. And then the LED stays on.

If possible I would like to have the LED off as well.

medusalix, 2020-01-04:
This driver should ideally behave identically to the Windows driver in terms of functionality. On Windows, the LED turns off when the last controller disconnects. I've implemented that feature in the `disconnect` branch. Let me know if it works for you and whether you've encountered any bugs.

Hedronmx, 2020-01-04:
Ill give it a go. Thank you!

NoXPhasma, 2020-01-04:
I can confirm that it works with one controller, connected and disconnected it several times without any issue. However, I only have one controller here at the moment.

medusalix, 2020-01-04:
@Hedronmx Could you do a packet capture on Windows please (with multiple controllers and disconnecting them one after the other)? It's pretty hard for me to actually test new features as I only got one controller.

Hedronmx, 2020-01-04:
How can I do that?

Hedronmx, 2020-01-05:
I just realized that turning off the controllers in the reverse order you turned them on, then the led does turn off and you can reconnect the controller with no issue

medusalix, 2020-01-05:
@Hedronmx Instructions for making the capture:

- Install [Wireshark](https://www.wireshark.org/). Make sure to tick the option to install **USBPcap** in the installer.
- At the bottom of the Wireshark window you should see multiple USBPcap entries.
- You have to identify which one is the correct one by clicking on one of them to start the capture.
- Now plug in the dongle. If you see lots of messages appearing, then it's probably the correct one. Otherwise you have to repeat the procedure with one of the other USBPcap entries.
- Turn on the controllers and wait for them to pair. Try disconnecting them in a different order (try out some combinations).
- Stop the capture and save the captured data. Upload the created *.pcapng file to any file hoster you like and post the link.

Thank you.

Hedronmx, 2020-01-07:
https://send.firefox.com/download/1c459ca95569def7/#lG192cgRlGtTH98JPtZ_Qg

There you go! hopefully i did it right.

First time I disconnected out of order, second time I disconnected reversed order.

How I can get the debug logs of xow?

medusalix, 2020-01-07:
Thanks for the capture, I'm going to take a look at it.
You can build a debug version of xow by running `make clean` followed by `make BUILD=DEBUG`.

medusalix, 2020-01-12:
@Hedronmx Service logs can be accessed using `journalctl -b -u xow`. Could you please also do a capture on your Linux machine running xow?

Hedronmx, 2020-01-18:
I updated to latest commit of master and now the leds turn off everytime.

NoXPhasma, 2020-01-18:
The latest master doesn't even have the disconnect functionality.

Hedronmx, 2020-01-18:
The I have no idea!!! Its working just fine....

NoXPhasma, 2020-01-18:
Are you sure you are actually in the `master` branch and not still in `disconnect`? You can check that by running this command in the xow folder: `cat .git/HEAD`.

Hedronmx, 2020-01-19:
I guess i was running the old service then, I restarted and now the led wont turn off. Ill go back to disconnect and get you the logs

Hedronmx, 2020-01-19:
```
-- Logs begin at Sat 2020-01-18 17:05:10 PST, end at Sat 2020-01-18 17:18:57 PST. --
ene 18 17:05:11 carlos-pc systemd[1]: Started Xbox One Wireless Dongle Driver.
ene 18 17:05:11 carlos-pc xow[6376]: 2020-01-18 17:05:11 INFO  - xow v0.2-25-gb58edb4 ©Severin v. W.
ene 18 17:05:12 carlos-pc xow[6376]: 2020-01-18 17:05:12 INFO  - Dongle plugged in
ene 18 17:05:12 carlos-pc xow[6376]: 2020-01-18 17:05:12 DEBUG - Firmware already loaded
ene 18 17:05:12 carlos-pc xow[6376]: 2020-01-18 17:05:12 DEBUG - Chip version: 7612
ene 18 17:05:12 carlos-pc xow[6376]: 2020-01-18 17:05:12 INFO  - Chip address: 62:45:b5:03:8b:c7
ene 18 17:05:12 carlos-pc xow[6376]: 2020-01-18 17:05:12 INFO  - Dongle initialized
ene 18 17:11:02 carlos-pc xow[6376]: 2020-01-18 17:11:02 DEBUG - Client associating: 7e:ed:82:03:b9:a7
ene 18 17:11:02 carlos-pc xow[6376]: 2020-01-18 17:11:02 DEBUG - Client identifier: 1
ene 18 17:11:02 carlos-pc xow[6376]: 2020-01-18 17:11:02 INFO  - Controller '1' connected
ene 18 17:11:02 carlos-pc xow[6376]: 2020-01-18 17:11:02 INFO  - Serial number: 02HU0097864920
ene 18 17:11:06 carlos-pc xow[6376]: 2020-01-18 17:11:06 DEBUG - Battery type: 1, level: 3
ene 18 17:11:08 carlos-pc xow[6376]: 2020-01-18 17:11:08 DEBUG - Client associating: 7e:ed:81:fc:5e:f4
ene 18 17:11:08 carlos-pc xow[6376]: 2020-01-18 17:11:08 DEBUG - Client identifier: 2
ene 18 17:11:08 carlos-pc xow[6376]: 2020-01-18 17:11:08 INFO  - Controller '2' connected
ene 18 17:11:08 carlos-pc xow[6376]: 2020-01-18 17:11:08 INFO  - Serial number: 02OW0069672917
ene 18 17:11:12 carlos-pc xow[6376]: 2020-01-18 17:11:12 DEBUG - Battery type: 1, level: 3
ene 18 17:11:22 carlos-pc xow[6376]: 2020-01-18 17:11:22 DEBUG - Battery type: 1, level: 3
ene 18 17:11:23 carlos-pc xow[6376]: 2020-01-18 17:11:23 DEBUG - Client disassociating: 7e:ed:82:03:b9:a7
ene 18 17:11:23 carlos-pc xow[6376]: 2020-01-18 17:11:23 DEBUG - Client identifier: 1
ene 18 17:11:23 carlos-pc xow[6376]: 2020-01-18 17:11:23 INFO  - Controller '1' disconnected
ene 18 17:11:32 carlos-pc xow[6376]: 2020-01-18 17:11:32 DEBUG - Battery type: 1, level: 3
ene 18 17:11:40 carlos-pc xow[6376]: 2020-01-18 17:11:40 DEBUG - Battery type: 1, level: 3
ene 18 17:11:40 carlos-pc xow[6376]: 2020-01-18 17:11:40 DEBUG - Client disassociating: 7e:ed:81:fc:5e:f4
ene 18 17:11:40 carlos-pc xow[6376]: 2020-01-18 17:11:40 DEBUG - Client identifier: 2
ene 18 17:11:40 carlos-pc xow[6376]: 2020-01-18 17:11:40 INFO  - Controller '2' disconnected
ene 18 17:12:08 carlos-pc xow[6376]: 2020-01-18 17:12:08 DEBUG - Client associating: 7e:ed:81:fc:5e:f4
ene 18 17:12:08 carlos-pc xow[6376]: 2020-01-18 17:12:08 DEBUG - Client identifier: 2
ene 18 17:12:08 carlos-pc xow[6376]: 2020-01-18 17:12:08 INFO  - Controller '2' connected
ene 18 17:12:09 carlos-pc xow[6376]: 2020-01-18 17:12:09 INFO  - Serial number: 02OW0069672917
ene 18 17:12:12 carlos-pc xow[6376]: 2020-01-18 17:12:12 DEBUG - Battery type: 1, level: 3
ene 18 17:12:16 carlos-pc xow[6376]: 2020-01-18 17:12:16 DEBUG - Client disassociating: 7e:ed:81:fc:5e:f4
ene 18 17:12:16 carlos-pc xow[6376]: 2020-01-18 17:12:16 DEBUG - Client identifier: 2

```

medusalix, 2020-01-19:
@Hedronmx Thanks for the logs, but you're still on master 😄. The version info should be `v0.2-17-g357c8cb` (first line of the logs).

Hedronmx, 2020-01-19:
Ok so it happened again!! i download the zip version of the disconnect branch. Installed it from there and the led turns off just fine. The logs i sent where from the disconnected branch install before a reboot.

Why is that happening??

medusalix, 2020-01-21:
I'm going to buy a second controller and do the testing myself.

Hedronmx, 2020-01-21:
Ok, sorry

Neverest, 2020-01-23:
I gave it a try and it does turn off the led from the adapter when I disconnect the controller. Here's my log:

```
-- Logs begin at Thu 2018-07-26 15:24:31 AST, end at Thu 2020-01-23 09:09:34 AST. --
Jan 23 09:07:22 neverest-ubuntu systemd[1]: Started Xbox One Wireless Dongle Driver.
Jan 23 09:07:22 neverest-ubuntu xow[1192]: 2020-01-23 09:07:22 INFO  - xow v0.2-17-g357c8cb ©S
Jan 23 09:07:22 neverest-ubuntu xow[1192]: 2020-01-23 09:07:22 INFO  - Dongle plugged in
Jan 23 09:07:22 neverest-ubuntu xow[1192]: 2020-01-23 09:07:22 DEBUG - Firmware already loaded
Jan 23 09:07:22 neverest-ubuntu xow[1192]: 2020-01-23 09:07:22 DEBUG - Chip version: 7632
Jan 23 09:07:22 neverest-ubuntu xow[1192]: 2020-01-23 09:07:22 INFO  - Chip address: 62:45:b4:
Jan 23 09:07:22 neverest-ubuntu xow[1192]: 2020-01-23 09:07:22 INFO  - Dongle initialized
Jan 23 09:07:23 neverest-ubuntu xow[1192]: 2020-01-23 09:07:23 DEBUG - Client associating: 7e:
Jan 23 09:07:23 neverest-ubuntu xow[1192]: 2020-01-23 09:07:23 INFO  - Controller '1' connecte
Jan 23 09:07:23 neverest-ubuntu xow[1192]: 2020-01-23 09:07:23 INFO  - Serial number: 02640261
Jan 23 09:07:44 neverest-ubuntu xow[1192]: 2020-01-23 09:07:44 DEBUG - Battery type: 1, level:
Jan 23 09:07:50 neverest-ubuntu xow[1192]: 2020-01-23 09:07:50 DEBUG - Client lost: 1
Jan 23 09:07:50 neverest-ubuntu xow[1192]: 2020-01-23 09:07:50 INFO  - Controller '1' disconne
Jan 23 09:08:01 neverest-ubuntu xow[1192]: 2020-01-23 09:08:01 DEBUG - Client associating: 7e:
Jan 23 09:08:01 neverest-ubuntu xow[1192]: 2020-01-23 09:08:01 INFO  - Controller '1' connecte
Jan 23 09:08:01 neverest-ubuntu xow[1192]: 2020-01-23 09:08:01 INFO  - Serial number: 02640261
Jan 23 09:08:05 neverest-ubuntu xow[1192]: 2020-01-23 09:08:05 DEBUG - Battery type: 1, level:
Jan 23 09:08:10 neverest-ubuntu xow[1192]: 2020-01-23 09:08:10 DEBUG - Battery type: 1, level:
Jan 23 09:08:11 neverest-ubuntu xow[1192]: 2020-01-23 09:08:11 DEBUG - Client lost: 1
Jan 23 09:08:12 neverest-ubuntu xow[1192]: 2020-01-23 09:08:12 INFO  - Controller '1' disconne
Jan 23 09:08:15 neverest-ubuntu xow[1192]: 2020-01-23 09:08:15 DEBUG - Client associating: 7e:
Jan 23 09:08:15 neverest-ubuntu xow[1192]: 2020-01-23 09:08:15 INFO  - Controller '1' connecte
Jan 23 09:08:16 neverest-ubuntu xow[1192]: 2020-01-23 09:08:16 INFO  - Serial number: 02640261
Jan 23 09:08:20 neverest-ubuntu xow[1192]: 2020-01-23 09:08:20 DEBUG - Battery type: 1, level:
Jan 23 09:08:26 neverest-ubuntu xow[1192]: 2020-01-23 09:08:26 DEBUG - Battery type: 1, level:
Jan 23 09:08:27 neverest-ubuntu xow[1192]: 2020-01-23 09:08:27 DEBUG - Client lost: 1
Jan 23 09:08:27 neverest-ubuntu xow[1192]: 2020-01-23 09:08:27 INFO  - Controller '1' disconne
Jan 23 09:08:30 neverest-ubuntu xow[1192]: 2020-01-23 09:08:30 DEBUG - Client associating: 7e:
Jan 23 09:08:30 neverest-ubuntu xow[1192]: 2020-01-23 09:08:30 INFO  - Controller '1' connecte
Jan 23 09:08:31 neverest-ubuntu xow[1192]: 2020-01-23 09:08:31 INFO  - Serial number: 02640261
Jan 23 09:08:34 neverest-ubuntu xow[1192]: 2020-01-23 09:08:34 DEBUG - Battery type: 1, level:
Jan 23 09:08:42 neverest-ubuntu xow[1192]: 2020-01-23 09:08:42 DEBUG - Battery type: 1, level:
Jan 23 09:08:43 neverest-ubuntu xow[1192]: 2020-01-23 09:08:43 DEBUG - Client lost: 1
Jan 23 09:08:44 neverest-ubuntu xow[1192]: 2020-01-23 09:08:44 INFO  - Controller '1' disconne
Jan 23 09:08:47 neverest-ubuntu xow[1192]: 2020-01-23 09:08:47 DEBUG - Client associating: 7e:
Jan 23 09:08:47 neverest-ubuntu xow[1192]: 2020-01-23 09:08:47 INFO  - Controller '1' connecte
Jan 23 09:08:48 neverest-ubuntu xow[1192]: 2020-01-23 09:08:48 INFO  - Serial number: 02640261
Jan 23 09:08:52 neverest-ubuntu xow[1192]: 2020-01-23 09:08:52 DEBUG - Battery type: 1, level:
Jan 23 09:09:04 neverest-ubuntu xow[1192]: 2020-01-23 09:09:04 DEBUG - Client lost: 1
Jan 23 09:09:05 neverest-ubuntu xow[1192]: 2020-01-23 09:09:05 INFO  - Controller '1' disconne
Jan 23 09:09:07 neverest-ubuntu xow[1192]: 2020-01-23 09:09:07 DEBUG - Client associating: 7e:
Jan 23 09:09:07 neverest-ubuntu xow[1192]: 2020-01-23 09:09:07 INFO  - Controller '1' connecte
Jan 23 09:09:07 neverest-ubuntu xow[1192]: 2020-01-23 09:09:07 INFO  - Serial number: 02640261
Jan 23 09:09:11 neverest-ubuntu xow[1192]: 2020-01-23 09:09:11 DEBUG - Battery type: 1, level:
Jan 23 09:09:17 neverest-ubuntu xow[1192]: 2020-01-23 09:09:17 DEBUG - Battery type: 1, level:
Jan 23 09:09:18 neverest-ubuntu xow[1192]: 2020-01-23 09:09:18 DEBUG - Client lost: 1
Jan 23 09:09:18 neverest-ubuntu xow[1192]: 2020-01-23 09:09:18 INFO  - Controller '1' disconne
Jan 23 09:09:20 neverest-ubuntu xow[1192]: 2020-01-23 09:09:20 DEBUG - Client associating: 7e:
Jan 23 09:09:20 neverest-ubuntu xow[1192]: 2020-01-23 09:09:20 INFO  - Controller '1' connecte
Jan 23 09:09:21 neverest-ubuntu xow[1192]: 2020-01-23 09:09:21 INFO  - Serial number: 02640261
Jan 23 09:09:25 neverest-ubuntu xow[1192]: 2020-01-23 09:09:24 DEBUG - Battery type: 1, level:
Jan 23 09:09:30 neverest-ubuntu xow[1192]: 2020-01-23 09:09:30 DEBUG - Client lost: 1
Jan 23 09:09:30 neverest-ubuntu xow[1192]: 2020-01-23 09:09:30 INFO  - Controller '1' disconne
lines 32-54/54 (END)
Jan 23 09:08:34 neverest-ubuntu xow[1192]: 2020-01-23 09:08:34 DEBUG - Battery type: 1, level: 2
Jan 23 09:08:42 neverest-ubuntu xow[1192]: 2020-01-23 09:08:42 DEBUG - Battery type: 1, level: 2
Jan 23 09:08:43 neverest-ubuntu xow[1192]: 2020-01-23 09:08:43 DEBUG - Client lost: 1
[... cut ...]
Jan 23 09:09:20 neverest-ubuntu xow[1192]: 2020-01-23 09:09:20 DEBUG - Client associating: 7e:ed:81:0c:08:0a
Jan 23 09:09:20 neverest-ubuntu xow[1192]: 2020-01-23 09:09:20 INFO  - Controller '1' connected
Jan 23 09:09:21 neverest-ubuntu xow[1192]: 2020-01-23 09:09:21 INFO  - Serial number: 02640261083736
Jan 23 09:09:25 neverest-ubuntu xow[1192]: 2020-01-23 09:09:24 DEBUG - Battery type: 1, level: 2
Jan 23 09:09:30 neverest-ubuntu xow[1192]: 2020-01-23 09:09:30 DEBUG - Client lost: 1
Jan 23 09:09:30 neverest-ubuntu xow[1192]: 2020-01-23 09:09:30 INFO  - Controller '1' disconnected
```

medusalix, 2020-01-23:
@Neverest It appears to be working fine with only one controller connected. People reported problems when using multiple controllers, so I'll have to get a second controller first.

Neverest, 2020-01-23:
> @Neverest It appears to be working fine with only one controller connected. People reported problems when using multiple controllers, so I'll have to get a second controller first.

Oh, seems that I skipped those post talking about that. Sorry.

@Hedronmx I got the disconnect branch by typing `git clone -b disconnect --single-branch  https://github.com/medusalix/xow.git`

Hedronmx, 2020-01-23:
@Neverest  Yeah I did have the disconnect branch, the weird thing happens when on the disconnect branch I install the master. You should give it a try. Install the master and then connect and disconnect the controllers. Let me know if it did turn off the led after installing master.

Dont reboot or anything also.

Neverest, 2020-01-24:
@Hedronmx  It does as you say, but isn't that because systemd changes reflect on reboot?

Hedronmx, 2020-01-24:
Do you have multiple controllers? @Neverest if you do then try this

> tested it and it works just fine with one controller. Connecting multiple controllers keep the led on even when no connection is active
> 
> edit 2-
> when disconnecting multiple controllers, the dongle doesnt work anymore. I turn on the controller again and the led of the controllers indicates that is connected but the gamepad doesnt send any input

medusalix, 2020-02-04:
I did some testing with my 1537 and 1708 controller and it worked just fine. All the inputs were recognized properly (tested using `evtest`) no matter how often and in which order I connected or disconnected the controllers. The LED turned off every single time I disconnected both controllers.

Hedronmx, 2020-02-06:
All my controllers are 1708, could that be the difference? Am I doing something wrong?

medusalix, 2020-02-07:
I've just pushed the changes form the `disconnect` branch to `master`.

@Hedronmx Please try to run `make uninstall` followed by a reboot before installing the newest version. Maybe that's why it's not working for you.

Hedronmx, 2020-02-07:
I see, ill give it a shot! Thanks for your help and patience!

Hedronmx, 2020-02-08:
Works perfectly! Just tested it. Sorry If I made some mistakes

closed 2020-02-08 by Hedronmx