medusalix/xow #19: Segfault on disconnecting USB adapter when pairing

StopMotionCuber, 2020-01-06:
# Issue
When setting the USB adapter to pairing mode and disconnecting the device, the driver crashes with a segfault.

# Steps to reproduce
1. Start xow with USB adapter attached
2. Press the physical pairing button on the USB adapter
3. Wait for "Pairing Initiated" log message
4. Remove the USB adapter physically

# Expected results
The driver should detect that the adapter got removed, wait for reconnection and continue to run

# Actual results
The driver crashes with a segmentation fault.

# Additional infos
I ran the process with valgrind and will post the relevant infos for the segfault below.
I'm not sure where in this stack the issue should have been catched, so I do not feel confident enough to submit a Pull request on my own.
```
==18572== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==18572==  Access not within mapped region at address 0x40
==18572==    at 0x4BCB7D9: libusb_control_transfer (in /usr/lib/libusb-1.0.so.0.2.0)
==18572==    by 0x1176A3: UsbDevice::controlTransfer(UsbDevice::ControlPacket) (usb.cpp:108)
==18572==    by 0x10D3A3: MT76::controlRead(unsigned short, VendorRequest) (mt76.cpp:975)
==18572==    by 0x1102E4: MT76::writeBeacon(bool) (mt76.cpp:749)
==18572==    by 0x110E7B: MT76::handleButtonPress() (mt76.cpp:175)
==18572==    by 0x1144D6: MT76::handleBulkPacket(Bytes const&) (mt76.cpp:216)
==18572==    by 0x1146EC: operator() (mt76.cpp:47)
==18572==    by 0x1146EC: __invoke_impl<void, MT76::added()::<lambda()> > (invoke.h:60)
==18572==    by 0x1146EC: __invoke<MT76::added()::<lambda()> > (invoke.h:95)
==18572==    by 0x1146EC: _M_invoke<0> (thread:244)
==18572==    by 0x1146EC: operator() (thread:251)
==18572==    by 0x1146EC: std::thread::_State_impl<std::thread::_Invoker<std::tuple<MT76::added()::{lambda()#1}> > >::_M_run() (thread:195)
==18572==    by 0x495EEE3: execute_native_thread_routine (thread.cc:80)
==18572==    by 0x4BE44CE: start_thread (in /usr/lib/libpthread-2.30.so)
==18572==    by 0x4D182D2: clone (in /usr/lib/libc-2.30.so)
==18572==  If you believe this happened as a result of a stack
==18572==  overflow in your program's main thread (unlikely but
==18572==  possible), you can try to increase the size of the
==18572==  main thread stack using the --main-stacksize= flag.
==18572==  The main thread stack size used in this run was 8388608.
```

medusalix, 2020-01-06:
Thanks for the detailed bug report and the stack trace 😃. It could be some kind of bug in libusb's implementation of `libusb_control_transfer`. I'll take a closer look at it.

StopMotionCuber, 2020-01-06:
Well, the problem is that `handle` is a nullpointer and `libusb_control_transfer` does not like nullpointer as handles.
I'm not sure how to handle that correctly (pun intended).

Also there is a use after free in `Log::error("Error in bulk read: %d", transfer->status);` as transfer is freed already. I was too lazy to make a PR for that, please fix in one of your commits.

medusalix, 2020-01-06:
Ah, now I see. So the problem is that `UsbDevice::close` gets called and sets the handle to `nullptr` while we're doing the control transfer in the dongle thread. We could *handle* this by acquiring the `readMutex` before closing the device. Could you try if adding a simple `std::unique_lock<std::mutex> lock(readMutex);` in the `UsbDevice::close` function works?

> Also there is a use after free

Good catch 👍. Going to push that together with the other fix.

medusalix, 2020-01-07:
Ah, now I see. So `UsbDevice::close` sets the handle to `nullptr` while the dongle thread is still doing its control transfers. The only reason this is being done is to prevent `UsbDevice::open` from opening the device multiple times. I'm going to replace this with a boolean variable to indicate if the device is currently open. `libusb_control_transfer` will probably return `LIBUSB_ERROR_NO_DEVICE` but thats at least better than a crash.

> Also there is a use after free

Good catch 👍. Will push that together with the other fix.

medusalix, 2020-01-10:
@StopMotionCuber I've put checks into place that *handle* the edge case of removing the device while the pairing is in progress. Please see if commit 365c2bc311fe76f9c6056ce672167d0153fc8ad7 fixes your issues.

closed 2020-01-14 by medusalix