medusalix/xow #20: After connecting controller on windows, I have to unplug and plug again in ubuntu for the service to start

Hedronmx, 2020-01-08:
If I connect my controller in windows and then restart to ubuntu, if I just turn on the controller it wont work, I need to replug the adapter for the controller to connect again.

medusalix, 2020-01-12:
Are you using the older dongle? You might have to blacklist the `mt76` kernel module as described in issue #14. If that doesn't work then I'll have to see your debug logs.

Hedronmx, 2020-01-12:
I am using this dongle.
https://img-prod-cms-rt-microsoft-com.akamaized.net/cms/api/am/imageFileData/RE1B9yp?ver=f8bf&q=90&m=6&h=200&w=200&b=%23FFFFFFFF&f=jpg&o=f&aim=true

Hedronmx, 2020-02-12:
I realized its everytime the service starts, since after installing XOW I need to replug the adapter. Also I ran sudo ./xow and it was stuck on initializing adapter

medusalix, 2020-02-20:
@Hedronmx I've pushed a (potential) fix to the [`restart`](https://github.com/medusalix/xow/tree/restart) branch. It works for me, though I'm only switching between my Linux VM and Windows. Please make sure you've completely removed the old version before checking out the fix.

@andreashuetter As you've had issues with reboots [in the past](https://github.com/medusalix/xow/issues/11#issuecomment-568541593), I'd like to make sure we're not bringing them back. Could you please double check if this fix doesn't break anything for you?

andreashuetter, 2020-02-21:
OK, so I did some additonal testing today, but so far only with the old dongle. As soon as I find the time, I can also repeat those tests with the new dongle.
I tested both branches (master & restart) and didn't find any differences between them so far. No matter if I use commit **c3aeee0** (restart branch) or commit **e8a6d75** (master branch), the behavior always the same:
When cold-booting my machine, it always seems to work.
When warm-booting my machine, then sometimes it happens that
- the LED of the controllers is brighter than usual
- no input from the controllers is detected
In this case I can either replug the adapter to make it work again, or I can turn the controllers off and on (which isn't really a problem as for me).

I am using 1708 controllers.

medusalix, 2020-02-23:
@andreashuetter Thanks for the extensive test. It's always hard for me to reproduce issues that happen only sometimes, especially considering that I run Linux in a virtual machine.

@Hedronmx I've pushed the changes to master, please comment if the issue isn't fixed.

closed 2020-02-23 by medusalix

Hedronmx, 2020-02-24:
Sorry I couldn't test it! Was really busy. I will give it go on master. Thank you!