medusalix/xow #22: [META] Project Support

rwilliams, 2020-01-13:
medusalix have you considered setting up a paypal or something like a brave creators account so folks can tip you for your great work on this project.

rwilliams, 2020-01-13:
I'm an idiot :P

rwilliams, 2020-01-13:
Ok. Would it be possible to get an English paypal link too and you might be interested in creators.brave.com you can link your github to it to accept BAT tips.

medusalix, 2020-01-16:
I assumed PayPal would allow donations in all currencies and convert those to the corresponding value in EUR (minus the fees). Is that not the case?

I took a look at Brave Creators, but they require you to share all your personal information with their payment processor, which I don't really want to do. I'm fine with PayPal at the moment.

rwilliams, 2020-01-16:
Euros is not the problem. Trying to beat the recapcha boss in a foreign
language is.

rwilliams, 2020-01-16:
Ok, it looks like your email is at the top of the current link provided. Right now that link is in German and google translate doesn't seem to work on it. The recaptcha boss in particular is hard, though i guess i could manually paste the phrase they are looking for in google translate.

medusalix, 2020-01-16:
> Right now that link is in German

Ah, so the problem is the German page and not the actual payment process.
From a quick google search it seems like other projects fixed this issue by append `&lc=US` to the donation link. Could you please try if that works? Thank you very much for your donation 😃.

rwilliams, 2020-01-16:
That didn't work. This did.
https://www.paypal.com/donate/?token=P-T6Gs5p3CPBmtOx2B1tpTm15mASHgLFUXw5JEKPghrua1vELEWJ3-coks3NDtdQEXcjB0&country.x=US&locale.x=US

medusalix, 2020-01-17:
That's the link PayPal redirects me to when I click https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=PLN6F3UGS37DE&lc=US. It works just fine on my system, maybe you need to clear the cache/cookies.

rwilliams, 2020-01-17:
I didn't append the `&lc=US` to your link, I guess i misunderstood and appended it to the generated link like an idiot. Good to go now.

closed 2020-01-17 by medusalix