medusalix/xow #23: Need to start command line first to make dongle link when started as service later.

zyll71, 2020-01-17:
I set up xow to run as a service. However, the service dies when I try to link the pad up to the dongle.
When I make sure the service is shut down again, unplug and plug the dongle in again, and I then start xom from the command line with the dongle plugged in already, all I need to do is wake the pad and the connection is established.
What is really strange: I can kill the command line instance of xom and restart the service again. The pad and the dongle then reconnect as expected and work.

Is there something wrong with the way the service initializes the dongle vs the command line init?
And why is the effect of the command line xom persistent?

Im running this on openSuSE Leap 15.1, Kernel 4.12.14-lp151.28.36-default x86_64
The Pad and dongle are brand new XBox One Controller and Wireless Adapter.

Details:

I boot PC without dongle plugged in, insert the dongle, and switch on the pad.
The pad seems to lock in (slow flashing stops) and after about 1s seems to loose connection (slow flashing starts again), dongle light is on (continuous).

> sudo journalctl -b -u xow
-- Logs begin at Fri 2020-01-17 09:31:44 CET, end at Fri 2020-01-17 09:43:01 CET. --  
Jan 17 09:31:48 [#] systemd[1]: Started Xbox One Wireless Dongle Driver.  
Jan 17 09:31:48 [#] xow[1062]: 2020-01-17 09:31:48 INFO  - xow v0.2-22-g1341b1e ©Severin v. W.  
Jan 17 09:38:50 [#.#.#] xow[1062]: 2020-01-17 09:38:50 INFO  - Dongle plugged in  
Jan 17 09:38:50 [#.#.#] xow[1062]: 2020-01-17 09:38:50 DEBUG - Firmware loaded  
Jan 17 09:38:50 [#.#.#] xow[1062]: 2020-01-17 09:38:50 DEBUG - Chip version: 7632  
Jan 17 09:38:50 [#.#.#] xow[1062]: 2020-01-17 09:38:50 INFO  - Chip address: 62:45:b5:10:43:39  
Jan 17 09:38:50 [#.#.#] xow[1062]: 2020-01-17 09:38:50 INFO  - Dongle initialized  
Jan 17 09:39:21 [#.#.#] xow[1062]: 2020-01-17 09:39:21 DEBUG - Client associating: 7e:ed:82:3c:05:f4  
Jan 17 09:39:21 [#.#.#] xow[1062]: 2020-01-17 09:39:21 DEBUG - Client identifier: 1  
Jan 17 09:39:21 [#.#.#] xow[1062]: terminate called after throwing an instance of 'InputException'  
Jan 17 09:39:21 [#.#.#] xow[1062]:   what():  Error opening device: Permission denied  
Jan 17 09:39:21 [#.#.#] systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT  
Jan 17 09:39:21 [#.#.#] systemd[1]: xow.service: Unit entered failed state.  
Jan 17 09:39:21 [#.#.#] systemd[1]: xow.service: Failed with result 'signal'.  

Note: "Error opening device: Permission denied". 

> lsusb  
[..]
Bus 003 Device 006: ID 045e:02fe Microsoft Corp.
[..]

> ls -la /dev/bus/usb/003/006
crw-rw-rw- 1 root root 189, 261 Jan 17 09:39 /dev/bus/usb/003/006

Here all users have rw permission.
I restart the service after the pad has gone back to sleep:  

> sudo systemctl start xow  

Check the status:  
> sudo journalctl -b -u xow  
[..]
Jan 17 09:54:15 [#.#.#] systemd[1]: Started Xbox One Wireless Dongle Driver.
Jan 17 09:54:15 [#.#.#] xow[2343]: 2020-01-17 09:54:15 INFO  - xow v0.2-22-g1341b1e ©Severin v. W.
Jan 17 09:54:15 [#.#.#] xow[2343]: 2020-01-17 09:54:15 INFO  - Dongle plugged in
Jan 17 09:54:15 [#.#.#] xow[2343]: 2020-01-17 09:54:15 DEBUG - Firmware already loaded
Jan 17 09:54:15 [#.#.#] xow[2343]: 2020-01-17 09:54:15 DEBUG - Chip version: 7632
Jan 17 09:54:15 [#.#.#] xow[2343]: 2020-01-17 09:54:15 INFO  - Chip address: 62:45:b5:10:43:39
Jan 17 09:54:15 [#.#.#] xow[2343]: 2020-01-17 09:54:15 INFO  - Dongle initialized

So the service is up and running again, dongle initialized.  
htop shows me, that xow is running as user 64499 (the DynamicUsers=true in the xow.service.in)
I unplug the dongle.
per htop, xow is still  running as user 64499

> sudo journalctl -b -u xow  
[..]
Jan 17 09:58:37 [#.#.#] xow[2343]: 2020-01-17 09:58:37 ERROR - Error in bulk read: 1
Jan 17 09:58:37 [#.#.#] xow[2343]: 2020-01-17 09:58:37 ERROR - Error in bulk read: 1
Jan 17 09:58:37 [#.#.#] xow[2343]: 2020-01-17 09:58:37 INFO  - Dongle removed

I guess this is as expected. I shut down the service and check:

> sudo systemctl stop xow  

> sudo journalctl -b -u xow  
[..]
Jan 17 10:00:43 [#.#.#] systemd[1]: Stopping Xbox One Wireless Dongle Driver...
Jan 17 10:00:43 [#.#.#] systemd[1]: Stopped Xbox One Wireless Dongle Driver.

Now I start xow from the command line and shut it down again brute force with Ctrl-C (just get the drive running, but don't touch the dongle):

> sudo ./xow
2020-01-17 10:02:16 INFO  - xow v0.2-22-g1341b1e ©Severin v. W.
^C

Next I plug in the dongle and restart the service:

> sudo systemctl start xow

> sudo journalctl -b -u xow  
[..]
Jan 17 10:03:54 [#.#.#] systemd[1]: Started Xbox One Wireless Dongle Driver.
Jan 17 10:03:54 [#.#.#] xow[2444]: 2020-01-17 10:03:54 INFO  - xow v0.2-22-g1341b1e ©Severin v. W.
Jan 17 10:03:55 [#.#.#] xow[2444]: 2020-01-17 10:03:55 INFO  - Dongle plugged in
Jan 17 10:03:55 [#.#.#] xow[2444]: 2020-01-17 10:03:55 DEBUG - Firmware loaded
Jan 17 10:03:55 [#.#.#] xow[2444]: 2020-01-17 10:03:55 DEBUG - Chip version: 7632
Jan 17 10:03:55 [#.#.#] xow[2444]: 2020-01-17 10:03:55 INFO  - Chip address: 62:45:b5:10:43:39
Jan 17 10:03:55 [#.#.#] xow[2444]: 2020-01-17 10:03:55 INFO  - Dongle initialized

I start the pad again, and the same problem arises:

Jan 17 10:05:07 [#.#.#] xow[2444]: 2020-01-17 10:05:07 DEBUG - Client associating: 7e:ed:82:3c:05:f4
Jan 17 10:05:07 [#.#.#] xow[2444]: 2020-01-17 10:05:07 DEBUG - Client identifier: 1
Jan 17 10:05:07 [#.#.#] xow[2444]: terminate called after throwing an instance of 'InputException'
Jan 17 10:05:07 [#.#.#] xow[2444]:   what():  Error opening device: Permission denied
Jan 17 10:05:07 [#.#.#] systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Jan 17 10:05:07 [#.#.#] systemd[1]: xow.service: Unit entered failed state.
Jan 17 10:05:07 [#.#.#] systemd[1]: xow.service: Failed with result 'signal'.

Again, I unplug the dongle
Next I plug the dongle back in again, game pad is sleeping.

> sudo journalctl -b -u xow  

shows nothing has happened (Is there an udev issue?).

I run once more:
> sudo ./xow
2020-01-17 10:11:55 INFO  - xow v0.2-22-g1341b1e ©Severin v. W.
2020-01-17 10:11:55 INFO  - Dongle plugged in
2020-01-17 10:11:55 DEBUG - Firmware loaded
2020-01-17 10:11:55 DEBUG - Chip version: 7632
2020-01-17 10:11:55 INFO  - Chip address: 62:45:b5:10:43:39
2020-01-17 10:11:55 INFO  - Dongle initialized

and press the link button on the dongle and wake the pad up again.

2020-01-17 10:12:38 INFO  - Pairing initiated
2020-01-17 10:12:38 INFO  - Pairing initiated
2020-01-17 10:12:38 INFO  - Pairing initiated
2020-01-17 10:12:38 INFO  - Pairing initiated
2020-01-17 10:12:49 DEBUG - Client associating: 7e:ed:82:3c:05:f4
2020-01-17 10:12:49 DEBUG - Client identifier: 1
2020-01-17 10:12:49 INFO  - Controller '1' connected
2020-01-17 10:12:49 INFO  - Serial number: 02600087216940
2020-01-17 10:12:53 DEBUG - Battery type: 1, level: 3

The controller is connected and working.
Now for the really strange part:
I shut down the xow instance again.

^C

And then start the service once more:
> sudo systemctl start xow

> sudo journalctl -b -u xow  
[..]
Jan 17 10:15:11 [#.#.#] systemd[1]: Started Xbox One Wireless Dongle Driver.
Jan 17 10:15:11 [#.#.#] xow[2566]: 2020-01-17 10:15:11 INFO  - xow v0.2-22-g1341b1e ©Severin v. W.
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 INFO  - Dongle plugged in
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 DEBUG - Firmware already loaded
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 DEBUG - Chip version: 7632
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 INFO  - Chip address: 62:45:b5:10:43:39
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 INFO  - Dongle initialized
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 DEBUG - Client associating: 7e:ed:82:3c:05:f4
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 DEBUG - Client identifier: 1
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 INFO  - Controller '1' connected
Jan 17 10:15:12 [#.#.#] xow[2566]: 2020-01-17 10:15:12 INFO  - Serial number: 02600087216940
Jan 17 10:15:34 [#.#.#] xow[2566]: 2020-01-17 10:15:34 DEBUG - Battery type: 1, level: 3
Jan 17 10:15:54 [#.#.#] xow[2566]: 2020-01-17 10:15:54 DEBUG - Battery type: 1, level: 3
Jan 17 10:16:14 [#.#.#] xow[2566]: 2020-01-17 10:16:14 DEBUG - Battery type: 1, level: 3
Jan 17 10:16:34 [#.#.#] xow[2566]: 2020-01-17 10:16:34 DEBUG - Battery type: 1, level: 3
Jan 17 10:16:54 [#.#.#] xow[2566]: 2020-01-17 10:16:54 DEBUG - Battery type: 1, level: 3
Jan 17 10:17:14 [#.#.#] xow[2566]: 2020-01-17 10:17:14 DEBUG - Battery type: 1, level: 3

per htop, xow is running as user 64499 again.
All seems well.

medusalix, 2020-01-17:
The `Error opening device: Permission denied` is thrown when xow doesn't have the required permissions to access the `/dev/uinput` device. When you run `make install` a `udev` rule is copied to `/lib/udev/rules.d` to make `/dev/input` read-write for all users.

zyll71, 2020-01-17:
(Since I had fiddled with the udev rule in the mean time and to be 100% sure, I de-installed, rebuilt and re-installed xow. Then a reboot.)

For some reason, the udev rules are not being applied when the service is started or they are not being applied.

> ls -la /dev/uinput
crw------- 1 root root 10, 223 Jan 17 12:44 /dev/uinput

> ls -la /lib/udev/rules.d/99-xow.rules
-rw-r--r-- 1 root root 261 Jan 17 12:17 /lib/udev/rules.d/99-xow.rules

> cat /lib/udev/rules.d/99-xow.rules
# Make dongles and uinput accessible to all users
SUBSYSTEM=="usb", ATTRS{idVendor}=="045e", ATTRS{idProduct}=="02e6", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="045e", ATTRS{idProduct}=="02fe", MODE="0666"
KERNEL=="uinput", SUBSYSTEM=="misc", MODE="0666"

> sudo lsusb
[..]
Bus 003 Device 006: ID 045e:02fe Microsoft Corp.
[..]

> sudo udevadm info -a -n /dev/bus/usb/003/006

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.4':
    KERNEL=="3-10.4"
    SUBSYSTEM=="usb"
    DRIVER=="usb"
    ATTR{authorized}=="1"
    ATTR{avoid_reset_quirk}=="0"
    ATTR{bConfigurationValue}=="1"
    ATTR{bDeviceClass}=="00"
    ATTR{bDeviceProtocol}=="00"
    ATTR{bDeviceSubClass}=="00"
    ATTR{bMaxPacketSize0}=="64"
    ATTR{bMaxPower}=="500mA"
    ATTR{bNumConfigurations}=="1"
    ATTR{bNumInterfaces}==" 1"
    ATTR{bcdDevice}=="0100"
    ATTR{bmAttributes}=="a0"
    ATTR{busnum}=="3"
    ATTR{configuration}=="XBOX ACC"
    ATTR{devnum}=="6"
    ATTR{devpath}=="10.4"
    ATTR{idProduct}=="02fe"
    ATTR{idVendor}=="045e"
    ATTR{ltm_capable}=="no"
    ATTR{manufacturer}=="Microsoft Inc."
    ATTR{maxchild}=="0"
    ATTR{product}=="XBOX ACC"
    ATTR{quirks}=="0x0"
    ATTR{removable}=="removable"
    ATTR{serial}=="413878"
    ATTR{speed}=="480"
    ATTR{urbnum}=="133"
    ATTR{version}==" 2.01"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb3/3-10':
    KERNELS=="3-10"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="09"
[... 82 lines cut ...]
    KERNELS=="pci0000:00"
    SUBSYSTEMS==""
    DRIVERS==""

> sudo journalctl -b -u xow
[sudo] password for root:
-- Logs begin at Fri 2020-01-17 12:44:50 CET, end at Fri 2020-01-17 12:46:04 CET. --
Jan 17 12:44:54 [x] systemd[1]: Started Xbox One Wireless Dongle Driver.
Jan 17 12:44:54 [x]  xow[1105]: 2020-01-17 12:44:54 INFO  - xow v0.2-22-g1341b1e ©Severin v. W.
Jan 17 12:44:54 [x]  xow[1105]: 2020-01-17 12:44:54 INFO  - Dongle plugged in
Jan 17 12:44:54 [x]  xow[1105]: 2020-01-17 12:44:54 DEBUG - Firmware already loaded
Jan 17 12:44:54 [x]  xow[1105]: 2020-01-17 12:44:54 DEBUG - Chip version: 7632
Jan 17 12:44:54 [x]  xow[1105]: 2020-01-17 12:44:54 INFO  - Chip address: 62:45:b5:10:43:39
Jan 17 12:44:54 [x]  xow[1105]: 2020-01-17 12:44:54 INFO  - Dongle initialized
Jan 17 12:44:54 [x]  xow[1105]: 2020-01-17 12:44:54 DEBUG - Client associating: 7e:ed:82:3c:05:f4
Jan 17 12:44:54 [x]  xow[1105]: 2020-01-17 12:44:54 DEBUG - Client identifier: 1
Jan 17 12:44:54 [x]  xow[1105]: terminate called after throwing an instance of 'InputException'
Jan 17 12:44:54 [x]  xow[1105]:   what():  Error opening device: Permission denied
Jan 17 12:44:54 [x]  systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Jan 17 12:44:54 [x]  systemd[1]: xow.service: Unit entered failed state.
Jan 17 12:44:54 [x]  systemd[1]: xow.service: Failed with result 'signal'.

Jan 17 12:44:54 Tulkas systemd[1]: xow.service: Main process exited, code=kille>
Jan 17 12:44:54 Tulkas systemd[1]: xow.service: Unit entered failed state.
Jan 17 12:44:54 Tulkas systemd[1]: xow.service: Failed with result 'signal'.

medusalix, 2020-01-17:
I've found [this issue](https://github.com/chrippa/ds4drv/issues/93#issuecomment-265300511) which seems to deal with the exact same problem. Please try if the mentioned workaround solves the udev issues.

zyll71, 2020-01-17:
> Creating` /etc/modules-load.d/uinput.conf `with the contents of `uinput`
alone did not suffice. The permissions do seem better:
```
> ls -la /dev/uinput
crw-rw-rw- 1 root root 10, 223 Jan 17 13:52 /dev/uinput
```
However, the service does not run beyond
```
> sudo journalctl -b -u xow
-- Logs begin at Fri 2020-01-17 13:52:44 CET, end at Fri 2020-01-17 13:55:37 CET. --
Jan 17 13:52:48 [x] systemd[1]: Started Xbox One Wireless Dongle Driver.
Jan 17 13:52:48 [x] xow[1091]: 2020-01-17 13:52:48 INFO  - xow v0.2-22-g1341b1e ©Severin v. W.
Jan 17 13:52:48 [x] xow[1091]: 2020-01-17 13:52:48 INFO  - Dongle plugged in
Jan 17 13:52:48 [x] xow[1091]: 2020-01-17 13:52:48 DEBUG - Firmware already loaded
Jan 17 13:52:48 [x] xow[1091]: 2020-01-17 13:52:48 DEBUG - Chip version: 7632
Jan 17 13:52:48 [x] xow[1091]: 2020-01-17 13:52:48 INFO  - Chip address: 62:45:b5:10:43:39
Jan 17 13:52:48 [x] xow[1091]: 2020-01-17 13:52:48 INFO  - Dongle initialized
```

Adding `GROUP="uinput"` to `/lib/udev/rules.d/99-xow.rules` does not help either (is this the right pace?). So I have removed it again.

zyll71, 2020-01-17:
Hmm, I can't seem to find a group `uinput` in my `/etc/group`. No wonder that did not work. A openSuSE thing calling it something different perhaps?

zyll71, 2020-01-17:
Please disregard the above.
No idea what I borked previously. For some reason, after a reboot, the pad and the dongle now connect using the workaround you mentioned. 

> Creating/etc/modules-load.d/uinput.confwith the contents of uinput

Thank you very much!

medusalix, 2020-01-17:
I've added the workaround to the `Makefile` in b0cdf7dc5be2acbb40337b52f253b835ec72753b. Problem should be solved.

closed 2020-01-17 by medusalix

jerky33, 2021-11-13:
I was receiving the same error when trying to pair my controller running under debian 11
`xow[PID]:   what():  Error opening device: Permission denied`
I was able to resolve it by changing the permission on the service

First I disabled the service
```
sudo systemctl disable xow
```
Next I changed the permissions to the service file to allow writing by all users
```
sudo chmod 666 /etc/systemd/system/xow.service
```
Then I re-enabled the service
```
sudo systemctl enable xow
```
Finally I started the service and was able to pair my controller as expected.
```
sudo systemctl start xow
```

I hope this help someone

GLHF

kakra, 2021-11-13:
> Next I changed the permissions to the service file to allow writing by all users

This makes no sense - and it creates a security problem. This file is never written to, and it should not be possible. The permission problem is more likely with the device node, and xow should probably log which one.

jerky33, 2021-11-14:
Now that the controller is paired I reverted the permissions back to the default 644 and was able to connect the controller, now that it's paired