medusalix/xow #24: Restarting the computer doesn' t turn off the adapter(old one) or the controller.

Neverest, 2020-01-23:
On windows if I restart my computer the adapter led and the controller turn off. When I try this on linux (Ubuntu Mate 18.04.3) the adapter led stays on and the controller led is blinking, as if looking to connect again until I boot up to linux or windows.

Using the old bulky adapter and 1708 controller.

medusalix, 2020-01-23:
Yes, that feature works by sending a special message to all controllers when the system is shut down (on Windows). I've tried adding a signal handler for Linux' `SIGTERM` signal (which is sent on program termination), but there are some [limitations](https://stackoverflow.com/a/48262055) that make the actual implementation pretty difficult.

medusalix, 2020-02-18:
@Neverest Please checkout the `poweroff` branch. Any kind of driver termination/shutdown should now power the dongle's LED and all connected controllers off.

I'd really appreciate if @NoXPhasma, @andreashuetter or anyone else that still sticks around could test this. I don't want this new feature to cause any problems to the existing functionality of the driver.

Neverest, 2020-02-18:
> @Neverest Please checkout the `poweroff` branch. Any kind of driver termination/shutdown should now power the dongle's LED and all connected controllers off.
>
> I'd really appreciate if @NoXPhasma, [...]

Tested it just now by restarting it 6 times either directly to linux or changing to windows. The controller and the adapter LED shutoff instantly. Shutting down also powered it down as it should.

Cant test with more than one controller though. Friend that had one isn't close by anymore.

NoXPhasma, 2020-02-18:
Can confirm it works perfectly by stopping the systemd service with one Controller:
```
Feb 18 20:01:50 NoXP-PC systemd[1]: Started Xbox One Wireless Dongle Driver.
Feb 18 20:01:50 NoXP-PC xow[27061]: 2020-02-18 20:01:50 INFO  - xow v0.3-18-ge0fff7f ©Severin v. W.
Feb 18 20:01:51 NoXP-PC xow[27061]: 2020-02-18 20:01:51 INFO  - Dongle plugged in
Feb 18 20:01:51 NoXP-PC xow[27061]: 2020-02-18 20:01:51 INFO  - Chip address: 62:45:b4:fe:81:16
Feb 18 20:01:51 NoXP-PC xow[27061]: 2020-02-18 20:01:51 INFO  - Dongle initialized
Feb 18 20:01:54 NoXP-PC xow[27061]: 2020-02-18 20:01:54 INFO  - Controller '1' connected
Feb 18 20:01:54 NoXP-PC xow[27061]: 2020-02-18 20:01:54 INFO  - Serial number: 02600108383810
Feb 18 20:02:01 NoXP-PC systemd[1]: Stopping Xbox One Wireless Dongle Driver...
Feb 18 20:02:01 NoXP-PC xow[27061]: 2020-02-18 20:02:01 ERROR - Packet for unconnected controller '1'
Feb 18 20:02:01 NoXP-PC xow[27061]: 2020-02-18 20:02:01 INFO  - Dongle disconnected
Feb 18 20:02:01 NoXP-PC xow[27061]: 2020-02-18 20:02:01 ERROR - Packet for unconnected controller '1'
Feb 18 20:02:01 NoXP-PC systemd[1]: xow.service: Succeeded.
Feb 18 20:02:01 NoXP-PC systemd[1]: Stopped Xbox One Wireless Dongle Driver.
Feb 18 20:02:05 NoXP-PC systemd[1]: Started Xbox One Wireless Dongle Driver.
Feb 18 20:02:05 NoXP-PC xow[27151]: 2020-02-18 20:02:05 INFO  - xow v0.3-18-ge0fff7f ©Severin v. W.
Feb 18 20:02:05 NoXP-PC xow[27151]: 2020-02-18 20:02:05 INFO  - Dongle plugged in
Feb 18 20:02:05 NoXP-PC xow[27151]: 2020-02-18 20:02:05 INFO  - Chip address: 62:45:b4:fe:81:16
Feb 18 20:02:05 NoXP-PC xow[27151]: 2020-02-18 20:02:05 INFO  - Dongle initialized
Feb 18 20:02:09 NoXP-PC xow[27151]: 2020-02-18 20:02:09 INFO  - Controller '1' connected
Feb 18 20:02:09 NoXP-PC xow[27151]: 2020-02-18 20:02:09 INFO  - Serial number: 02600108383810
Feb 18 20:02:12 NoXP-PC systemd[1]: Stopping Xbox One Wireless Dongle Driver...
Feb 18 20:02:12 NoXP-PC xow[27151]: 2020-02-18 20:02:12 ERROR - Packet for unconnected controller '1'
Feb 18 20:02:12 NoXP-PC xow[27151]: 2020-02-18 20:02:12 INFO  - Dongle disconnected
Feb 18 20:02:12 NoXP-PC xow[27151]: 2020-02-18 20:02:12 ERROR - Packet for unconnected controller '1'
Feb 18 20:02:12 NoXP-PC systemd[1]: xow.service: Succeeded.
Feb 18 20:02:12 NoXP-PC systemd[1]: Stopped Xbox One Wireless Dongle Driver.
```

medusalix, 2020-02-21:
Thanks for your feedback, I've merged the changes into master. Please reopen in case you encounter any bugs.

closed 2020-02-21 by medusalix