medusalix/xow #25: Elite controller does not pair

Waxyen, 2020-01-23:
I have an Elite controller with model number 1698, but it won't pair.

I have built the project with the `make BUILD=DEBUG` command, and this is the result:

```
2020-01-23 19:43:42 INFO  - xow v0.2-25-gb58edb4 ©Severin v. W.
2020-01-23 19:43:43 INFO  - Dongle plugged in
2020-01-23 19:43:43 DEBUG - Firmware already loaded
2020-01-23 19:43:43 DEBUG - Chip version: 7632
2020-01-23 19:43:43 INFO  - Chip address: 62:45:b4:f5:76:96
2020-01-23 19:43:43 INFO  - Dongle initialized
2020-01-23 19:43:48 INFO  - Pairing initiated
2020-01-23 19:43:48 INFO  - Pairing initiated
2020-01-23 19:43:48 INFO  - Pairing initiated
2020-01-23 19:43:48 INFO  - Pairing initiated
2020-01-23 19:44:15 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:44:15 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:44:15 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:44:15 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:44:46 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:45:00 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:45:00 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:45:00 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:45:26 INFO  - Pairing initiated
2020-01-23 19:45:26 INFO  - Pairing initiated
2020-01-23 19:45:27 INFO  - Pairing initiated
2020-01-23 19:45:27 INFO  - Pairing initiated
2020-01-23 19:45:28 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:45:45 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:45:45 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 19:45:45 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
```
I have tried to get some packages captures via Wireshark, but I don't see any traffic on the usbmon{x} interfaces in Wireshark. (I do see usbmon0...6 in Wireshark, but no traffic.

Is there anything else you'd like me to try/test?

medusalix, 2020-01-23:
Thanks for the detailed issue description 👍. Could you please make sure that your elite controller is running the latest firmware? According to the logs the controller responds to the pairing requests (which is a good sign) but doesn't finish the pairing process. Other people have successfully used their elite controller with xow, but in your case I'll definitely need a packet capture.

Waxyen, 2020-01-23:
Updating the Elite controller is a good idea, but the Xbox Accessories app did not see any update. The software version of the Elite controller is currently 2.3.2388.0. Can you check if you have this software version too?

If the software versions match, I'd be more than happy to do some packet captures, but as I mentioned, I'm not sure how. Do the issues with Wireshark that I described seem familiar?

Waxyen, 2020-01-23:
I did try to connect another time, since the Xbox Accessories app did reboot the Elite controller. The output is a little different this time. It might give some clues as well. But the packet capture is probably still needed.

```
2020-01-23 22:40:19 INFO  - xow v0.2-25-gb58edb4 ©Severin v. W.
2020-01-23 22:40:20 INFO  - Dongle plugged in
2020-01-23 22:40:20 DEBUG - Firmware already loaded
2020-01-23 22:40:20 DEBUG - Chip version: 7632
2020-01-23 22:40:20 INFO  - Chip address: 62:45:b4:f5:76:96
2020-01-23 22:40:20 INFO  - Dongle initialized
2020-01-23 22:40:39 INFO  - Pairing initiated
2020-01-23 22:40:51 DEBUG - Client associating: 7e:ed:89:8b:50:0e
2020-01-23 22:40:51 DEBUG - Client identifier: 1
2020-01-23 22:40:51 INFO  - Controller '1' connected
2020-01-23 22:41:15 INFO  - Pairing initiated
2020-01-23 22:41:31 INFO  - Pairing initiated
2020-01-23 22:42:30 DEBUG - Client associating: 7e:ed:89:8b:50:0e
2020-01-23 22:42:30 DEBUG - Client identifier: 1
2020-01-23 22:42:30 DEBUG - Controller '1' reconnecting
2020-01-23 22:42:30 INFO  - Controller '1' connected
2020-01-23 22:42:30 DEBUG - Client disassociating: 7e:ed:89:8b:50:0e
2020-01-23 22:42:30 DEBUG - Client identifier: 1
2020-01-23 22:42:30 INFO  - Controller '1' disconnected
2020-01-23 22:42:55 INFO  - Pairing initiated
2020-01-23 22:42:56 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 22:42:59 DEBUG - Client associating: 7e:ed:89:8b:50:0e
2020-01-23 22:42:59 DEBUG - Client identifier: 1
2020-01-23 22:42:59 INFO  - Controller '1' connected
2020-01-23 22:42:59 DEBUG - Client disassociating: 7e:ed:89:8b:50:0e
2020-01-23 22:42:59 DEBUG - Client identifier: 1
2020-01-23 22:42:59 INFO  - Controller '1' disconnected
2020-01-23 22:43:26 INFO  - Pairing initiated
2020-01-23 22:43:59 ERROR - Error in bulk read: 1
2020-01-23 22:43:59 ERROR - Error in bulk read: 1
2020-01-23 22:43:59 INFO  - Dongle removed
2020-01-23 22:43:59 ERROR - Failed to write command
2020-01-23 22:43:59 ERROR - Failed to burst write register
2020-01-23 22:43:59 ERROR - Failed to write beacon
2020-01-23 22:43:59 ERROR - Failed to write pairing beacon
2020-01-23 22:44:04 INFO  - Dongle plugged in
2020-01-23 22:44:04 DEBUG - Firmware loaded
2020-01-23 22:44:04 DEBUG - Chip version: 7632
2020-01-23 22:44:04 INFO  - Chip address: 62:45:b4:f5:76:96
2020-01-23 22:44:04 INFO  - Dongle initialized
2020-01-23 22:44:07 INFO  - Pairing initiated
2020-01-23 22:44:07 INFO  - Pairing initiated
2020-01-23 22:44:07 INFO  - Pairing initiated
2020-01-23 22:44:07 INFO  - Pairing initiated
2020-01-23 22:44:29 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 22:44:29 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 22:44:29 DEBUG - Client pairing: 7e:ed:89:8b:50:0e
2020-01-23 22:44:32 DEBUG - Client associating: 7e:ed:89:8b:50:0e
2020-01-23 22:44:32 DEBUG - Client identifier: 1
2020-01-23 22:44:32 INFO  - Controller '1' connected
2020-01-23 22:44:32 DEBUG - Client disassociating: 7e:ed:89:8b:50:0e
2020-01-23 22:44:32 DEBUG - Client identifier: 1
2020-01-23 22:44:32 INFO  - Controller '1' disconnected
2020-01-23 22:44:45 ERROR - Error in bulk read: 1
2020-01-23 22:44:45 ERROR - Error in bulk read: 1
2020-01-23 22:44:45 INFO  - Dongle removed
2020-01-23 22:44:45 DEBUG - Client disassociating: 
2020-01-23 22:44:45 DEBUG - Client identifier: 1
2020-01-23 22:44:45 ERROR - Failed to write command
2020-01-23 22:44:45 ERROR - Failed to burst write register
2020-01-23 22:44:45 ERROR - Failed to write WCID
2020-01-23 22:44:45 ERROR - Failed to disassociate client
```

medusalix, 2020-01-23:
I'm currently not at home, so I'll check the firmware version tomorrow. Although I'm not even sure if my normal Xbox One controller gets the same firmware as the elite controllers.
I've never had any problems with the `usbmon` captures, I really don't know why they're failing for you.
From the logs you posted above it looks like the controller is connecting just fine (at `22:40:51`). Did you press the pairing button a second time (at `22:41:15`)?

Waxyen, 2020-01-24:
Funny business going on here. Well, maybe it's easy to explain.

What I've done:
1. Removed the xow project from my PC
2. Cloned the xow project from Github and `cd`-ed in there
3. Did a build with `make BUILD=DEBUG`
4. Executed `sudo ./xow`
5. Hit the pair button on my dongle
6. Powered ON the Xbox Elite controller and hit the pairing button there
7. It's connected.

I did these exact steps. Here is the output of the xow program:

```
2020-01-25 00:34:35 INFO  - xow v0.3-2-gb6b635a ©Severin v. W.
2020-01-25 00:34:35 INFO  - Dongle plugged in
2020-01-25 00:34:35 DEBUG - Firmware loaded
2020-01-25 00:34:35 DEBUG - Chip version: 7632
2020-01-25 00:34:35 INFO  - Chip address: 62:45:b4:f5:76:96
2020-01-25 00:34:35 INFO  - Dongle initialized
2020-01-25 00:34:39 INFO  - Pairing initiated
2020-01-25 00:34:39 INFO  - Pairing initiated
2020-01-25 00:34:39 INFO  - Pairing initiated
2020-01-25 00:34:39 INFO  - Pairing initiated
2020-01-25 00:34:45 DEBUG - Client associating: 7e:ed:89:8b:50:0e
2020-01-25 00:34:45 DEBUG - Client identifier: 1
2020-01-25 00:34:45 INFO  - Controller '1' connected
2020-01-25 00:34:47 DEBUG - Battery type: 2, level: 3
2020-01-25 00:34:53 DEBUG - Battery type: 2, level: 2
```

I am sure that I've used the code from commit b58edb4ce50910fb7401c3d31919b2cc448aaaba.

I'm happy it works. I have no clue what exactly did fix it.

medusalix, 2020-01-25:
> I'm happy it works. I have no clue what exactly did fix it.

Neither do I 😄. Pairing can be a bit unreliable from time to time, but I'm happy that you've got it working!

closed 2020-01-25 by medusalix