medusalix/xow #26: Use of xow in combination with VirtualHere

andreashuetter, 2020-01-24:
I have a question:
xow runs perfectly well for me on my Linux machine if I plug to dongle directly into a USB-Port on that machine.
What does not work (yet) is the following: Plug the dongle into a USB-Port on my Steam Link and forward it via VirtualHere to my Linux machine.
This did work so far with every other USB device I ever tested, but not with the Xbox One Wireless Adapter.
When I forward the dongle from my Steam Link to my Linux machine where xow is running, xow crashes with the following output:

```
Jän 24 09:23:18 erazor systemd[1]: Started Xbox One Wireless Dongle Driver.
Jän 24 09:23:18 erazor xow[5528]: 2020-01-24 09:23:18 INFO  - xow v0.3 ©Severin v. W.
Jän 24 09:24:01 erazor xow[5528]: 2020-01-24 09:24:01 ERROR - Error resetting device: LIBUSB_ERROR_NOT_FOUND
Jän 24 09:24:01 erazor systemd[1]: xow.service: Main process exited, code=exited, status=1/FAILURE
Jän 24 09:24:01 erazor systemd[1]: xow.service: Failed with result 'exit-code'.
```

Is this a problem located inherently in VirtualHere, or is this something that can be fixed in xow?

Thanks in advance.

medusalix, 2020-01-24:
The error is thrown as a result of the call to `libusb_reset_device`:

https://github.com/medusalix/xow/blob/b58edb4ce50910fb7401c3d31919b2cc448aaaba/dongle/usb.cpp#L46

I'm not even sure if we *need* to reset the device, I included that statement just to be on the safe side (considering the case that some other driver misconfigures the device). You could try removing that line, it might work.

andreashuetter, 2020-01-24:
Wow, this is great!
I just commented out the above line as suggested by you, recompiled xow and now it works! I can plug the Xbox One Wireless Adapter into the USB-Port of my Steam Link now and forward it to my Linux machine. This is exactly what I wanted, thank you!

Do you think this line should be removed generally for the next release, or is it better to keep that line (but don't let the service crash if it returns said error)?

medusalix, 2020-01-25:
> Do you think this line should be removed generally for the next release, or is it better to keep that line (but don't let the service crash if it returns said error)?

Yeah, I'm going to remove that line in the next commit. It probably won't affect the operation of the driver in any negative way.

andreashuetter, 2020-01-25:
> EDIT: I just tried out the change in a virtual machine and xow crashed with `LIBUSB_ERROR_BUSY`. So I guess we'll have to keep that line. Did you try removing just the whole `if (error)` statement?

Well, you mean like so:
```
    error = libusb_reset_device(handle);
/*
    if (error)
    {
        libusb_close(handle);
        libusb_exit(nullptr);

        throw UsbException("Error resetting device: ", error);
    }
*/
```

In this case xow (when forwarding the USB-Dongle via VirtualHere) crashes with:
```
2020-01-25 20:27:44 INFO  - xow v0.3-2-gb6b635a ©Severin v. W.
2020-01-25 20:27:44 ERROR - Error opening device: LIBUSB_ERROR_ACCESS
```
But to be honest, this particular issue became less important to me personally as of today,, because the latest Steam Link Software Build now has xow integrated. So I don't need to use VirtualHere for this use case anymore, because the dongle now works directly on the Steam Link without the need to forward it to the other machine. Of course it would be nice to have xow not crashing when connecting a VirtualHere-forwarded dongle. But on the other hand, if such a change would mean to make xow potentially more unstable, then I'd say that's not worth it.

I won't feel offended if you decide to leave this as it is for now and concentrate on other things first (for example making audio work over the adapter, or what else).

I am so glad you made this thing, and now that it's also working on Steam Link I am even more happy.

medusalix, 2020-01-26:
> But to be honest, this particular issue became less important to me personally as of today, because the latest Steam Link Software Build now has xow integrated.

Oh wow, that's pretty cool 😃.

> But on the other hand, if such a change would mean to make xow potentially more unstable, then I'd say that's not worth it.

Yeah, let's leave it as it is.

closed 2020-01-26 by medusalix