medusalix/xow #28: [Discussion] Simple logo and web page for the project

cristianovitorino, 2020-01-31:
Hello,

First of all, thank you very much for the impressive work on the project! Working as a charm for me on Fedora 31 : )

I want to propose a simple logo and web page for the project. 

For the logo I thought about referencing both the physical dongle shape and imagery and also the command to run xow on ther terminal.

For the web page I thought about just going simple and effective.

Here are some ideas:

[image]

[image]

[image]

AntiGuide, 2020-01-31:
Nice I like it! A website would be cool. The last time i searched for the project and didn´t remember the name it was hard to find. This could greatly improve the visibility of the project.

NoXPhasma, 2020-01-31:
That logo looks really good, reminds me on `./xow`.

medusalix, 2020-02-01:
@cristianovitorino Very good idea. The minimalitic logos/designs you created look fantastic ❤️. We could maybe use [GitHub Pages](https://pages.github.com/) to host the web page.

cristianovitorino, 2020-02-01:
> @cristianovitorino Very good idea. The minimalitic logos/designs you created look fantastic . We could maybe use [GitHub Pages](https://pages.github.com/) to host the web page.

Thank you! GitHub Pages sounds good. How should I proceed? Should we vote on a particular design for the logo and web page? Or do you have a preferred design for the logo and webpage? Let me know and I'll try and put a web page together and post the files here ready to go and ready for you to work with.

medusalix, 2020-02-03:
I really can't decide between the two designs you posted. Just an idea: Fade from the black logo on white background to the white logo on black background when you hover over the banner. Maybe a mix of the two designs? I'm really busy right now so I apologize for my delayed responses.

NoXPhasma, 2020-02-03:
I personally prefer the first logo, it looks very clean.

cristianovitorino, 2020-02-04:
> I really can't decide between the two designs you posted. Just an idea: Fade from the black logo on white background to the white logo on black background when you hover over the banner. Maybe a mix of the two designs? I'm really busy right now so I apologize for my delayed responses.

No need to apologize! We all do Open Source on our free time : )

Sounds good to me, thanks for the idea. I'll see what I'm able to come up with.

cristianovitorino, 2020-02-05:
Done! [xow-website.zip](https://github.com/medusalix/xow/files/4158419/xow-website.zip)

I separated a logo for the README.md too. Suggestion:

```
<p align="center">
  <img src="logo.png" alt="Xow">
</p>
```

cristianovitorino, 2020-02-06:
Hey @medusalix I see that you've uploaded the website, nice! Can you give me access to do pull requests? I see that I need to update the website to reflect the repo latest changes and fix some sections, and also to help maintain it.

I have the changes ready for review on my local repo.

AntiGuide, 2020-02-07:
I ran the page trough some speed benchmarks and discovered that they all complain about the missing cache configuration. Is this something we can change or is this a server config thing?

cristianovitorino, 2020-02-07:
> I ran the page trough some speed benchmarks and discovered that they all complain about the missing cache configuration. Is this something we can change or is this a server config thing?

Seems to me like a GitHub Pages limitation. "CacheControl: max-age=600" (10m), which is impossible to change, see [this](https://webapps.stackexchange.com/questions/119286/caching-assets-in-website-served-from-github-pages) for more info.

AntiGuide, 2020-02-07:
Ok I see... Thanks for checking tough. Maybe with more supporters some time in the future we could rent a small server :blush:

AntiGuide, 2020-02-07:
I also found a small bug: When the install instructions overflow on a tall screen (mobile) the grey box doesn't get extended. I don't know if you already fixed this in the open PR but I thought I should put this here just in case :slightly_smiling:

cristianovitorino, 2020-02-08:
> I also found a small bug: When the install instructions overflow on a tall screen (mobile) the grey box doesn't get extended. I don't know if you already fixed this in the open PR but I thought I should put this here just in case :slightly_smiling:

Yeah I know about the issue, didn't have time to fix that. At first it seamed to be fixed, but I see it went throught into final version. There is no fix in the current PR. Feel free to do a PR later, or open a new issue on this bug, I'll look into it when I have some available time and also I need to wait for @medusalix to review and process the current PR.

Thanks for reporting!

medusalix, 2020-02-08:
> I also found a small bug: When the install instructions overflow on a tall screen (mobile) the grey box doesn't get extended.

Yeah, on Chrome it overflows horizontally whereas on Firefox the grey box is cut off vertically (Android):

[image]

cristianovitorino, 2020-02-08:
> > I also found a small bug: When the install instructions overflow on a tall screen (mobile) the grey box doesn't get extended.
> 
> Yeah, on Chrome it overflows horizontally whereas on Firefox the grey box is cut off vertically (Android):
> 

Fixed the issue.

Improved the aesthetics of that box along the way. Also improved the overall website performance by leaving all the files on their minified version.

PR is up and ready for review.

cristianovitorino, 2020-02-19:
The website is done, patched, fully functional and has good performance. Closing.

closed 2020-02-19 by cristianovitorino

medusalix, 2020-02-20:
Thank you very much for your awesome work on the website ❤️.

cristianovitorino, 2020-02-20:
> Thank you very much for your awesome work on the website .

No problem, was really fun! :+1: