medusalix/xow #29: Xbox Elite Controller 2

g00sed, 2020-02-03:
Looks like it does not work. 

If I run `# xow` I get the following output:
`2020-02-03 01:41:17 INFO  - xow  ©Severin v. W.
2020-02-03 01:41:18 INFO  - Dongle plugged in
2020-02-03 01:41:18 INFO  - Chip address: 6c:5d:3a:00:2c:75
2020-02-03 01:41:18 INFO  - Dongle initialized
2020-02-03 01:41:22 INFO  - Pairing initiated
2020-02-03 01:41:22 INFO  - Pairing initiated
2020-02-03 01:41:23 INFO  - Pairing initiated
2020-02-03 01:41:23 INFO  - Pairing initiated`

The light blinks, but when I attempt to pair the controller, nothing happens.

OS: Manjaro 5.4.16-1

medusalix, 2020-02-03:
Maybe @Hielyr can help you. It seems like the Elite Controller 2 should be working.

Hielyr, 2020-02-03:
I already have mine paired, but this is my output:

```
2020-02-03 06:52:46 INFO  - xow v0.3-3-gb6569fb ©Severin v. W.
2020-02-03 06:52:46 INFO  - Dongle plugged in
2020-02-03 06:52:46 INFO  - Chip address: 62:45:b5:05:20:7e
2020-02-03 06:52:46 INFO  - Dongle initialized
2020-02-03 06:53:02 INFO  - Controller '1' connected
2020-02-03 06:53:02 INFO  - Serial number: 0263XXXXXX4943
```
I could try unpairing and re-pairing the controller, but if memory serves, the output looked pretty much like that originally.

One thing I will note however, and it's just a peculiar thing that likely has no real relevance - when I connect my Elite v2 via the dongle using xow, Steam recognizes it as one of my other Xbox controllers - specifically one I had named, "Black Xbox One Controller" through Steam's interface.

g00sed, 2020-02-04:
Well I'm glad it's working for you at least!

How did you go about installing it and running it?

Hielyr, 2020-02-05:
I just followed the build instructions and set xow as a systemd service as recommended. 

I will say however, that I do have the latest firmware installed on the controller. I've read there are ways to do it via a Windows VM with USB passthrough using VirtualBox, but I just used an old Windows 10 laptop I happen to have and connected it via the USB cable and updated it using the Xbox Accessories application.

If your controller isn't fully updated, it may be worth a shot to see if updating it changes anything.

iroedius, 2020-02-07:
I have the same problem. I'm getting three 'Pairing initiated' messages and the dongle light starts to blink after pressing the dongle button. Then I press the bind button on the controller and its light starts to blink faster searching for the dongle but that's it, no pairing happens.

medusalix, 2020-03-09:
@g00sed @iroedius Are the pairing problems still present in the latest version of the driver?

iroedius, 2020-03-20:
@medusalix Yes, although in my case it may have something to do with the dongle itself. Its version is 1790 which is problematic as reported in #37.

medusalix, 2020-03-20:
@iroedius Yeah, I'll have to get my hands on one of these dongles to find out why it doesn't work. I'll close this issue for now, as the problems don't seem to be related to the elite controller.

closed 2020-03-20 by medusalix