medusalix/xow #30: terminate called after throwing an instance of 'InputException'

aguaviva, 2020-02-04:
Almost there! This is pretty exciting :) Here is the log in case it helps.

Some notes: The led in the dongle doesn't blink or lit at all.

```
$ sudo systemctl status xow -l
● xow.service - Xbox One Wireless Dongle Driver
   Loaded: loaded (/lib/systemd/system/xow.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Tue 2020-02-04 22:09:25 CET; 9s ago
  Process: 3900 ExecStart=/usr/local/bin/xow (code=killed, signal=ABRT)
 Main PID: 3900 (code=killed, signal=ABRT)

Feb 04 22:09:01 lenovo-thinkpad xow[3900]: 2020-02-04 22:09:01 DEBUG - Firmware already loaded
Feb 04 22:09:01 lenovo-thinkpad xow[3900]: 2020-02-04 22:09:01 DEBUG - Chip version: 7632
Feb 04 22:09:01 lenovo-thinkpad xow[3900]: 2020-02-04 22:09:01 INFO  - Chip address: 62:45:b4:e8:c6:60
Feb 04 22:09:01 lenovo-thinkpad xow[3900]: 2020-02-04 22:09:01 INFO  - Dongle initialized
Feb 04 22:09:25 lenovo-thinkpad xow[3900]: 2020-02-04 22:09:25 DEBUG - Client associating: 7e:ed:80:7e:71:3e
Feb 04 22:09:25 lenovo-thinkpad xow[3900]: 2020-02-04 22:09:25 DEBUG - Client identifier: 1
Feb 04 22:09:25 lenovo-thinkpad xow[3900]: terminate called after throwing an instance of 'InputException'
Feb 04 22:09:25 lenovo-thinkpad xow[3900]:   what():  Error opening device: Permission denied
Feb 04 22:09:25 lenovo-thinkpad systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Feb 04 22:09:25 lenovo-thinkpad systemd[1]: xow.service: Failed with result 'signal'.
```
Please let me know how can I help

aguaviva, 2020-02-04:
Not sure what happened but after a reboot, I was trying to capture a wireshark and it turned out that all was working fine! Thanks!!

medusalix, 2020-02-04:
That error is thrown when xow doesn't have the necessary permissions to open `/dev/uinput` (they're set by the udev rules). I'll add a corresponding notice to the README for future users.

closed 2020-02-04 by medusalix

aguaviva, 2020-02-04:
Awesome, I had a great time using the wireless controller, thanks for that :)

FrandSX, 2020-04-26:
This behavior is happening with  v0.4-11-gbea16b4 on latest retropie pi4 weekly image.
How would one add the necessary permissions?

medusalix, 2020-04-27:
@FrandSX They should be set automatically by the udev rules that are added during installation. You might want to check if there are other udev rules that change the permissions back (like in issue #54).

Rtful, 2021-02-04:
@medusalix How can I give xow permissions to open `/dev/uinput`?

Rtful, 2021-02-04:
I have a similar problem: 
```
$ sudo systemctl status xow -l
● xow.service - Xbox One Wireless Dongle Driver
     Loaded: loaded (/etc/systemd/system/xow.service; enabled; vendor preset: disabled)
     Active: failed (Result: core-dump) since Thu 2021-02-04 17:37:49 CET; 1min 13s ago
    Process: 1512 ExecStart=/usr/local/bin/xow (code=dumped, signal=ABRT)
   Main PID: 1512 (code=dumped, signal=ABRT)

Feb 04 17:37:48 Aaron-Manjaro systemd[1]: xow.service: Failed with result 'core-dump'.
Feb 04 17:37:48 Aaron-Manjaro systemd[1]: Started Xbox One Wireless Dongle Driver.
Feb 04 17:37:48 Aaron-Manjaro xow[1512]: 2021-02-04 17:37:48 INFO  - xow v0.5-24-g9d5a11b ©Severin v. W.
Feb 04 17:37:48 Aaron-Manjaro xow[1512]: 2021-02-04 17:37:48 INFO  - Waiting for device...
Feb 04 17:37:49 Aaron-Manjaro xow[1512]: 2021-02-04 17:37:49 ERROR - Firmware reset timed out
Feb 04 17:37:49 Aaron-Manjaro xow[1512]: terminate called after throwing an instance of 'Mt76Exception'
Feb 04 17:37:49 Aaron-Manjaro xow[1512]:   what():  Failed to load firmware
Feb 04 17:37:49 Aaron-Manjaro systemd[1]: xow.service: Main process exited, code=dumped, status=6/ABRT
Feb 04 17:37:49 Aaron-Manjaro systemd[1]: xow.service: Failed with result 'core-dump'.
```

kakra, 2021-02-04:
Run `sudo udevadm monitor -p` in a terminal before connecting the controller, it should then show all udev rules applied when the device is detected.

But there may also be a problem that the dongle is already loaded with a firmware by the mt76 kernel module for wifi (I've seen this behavior while experimenting with xow). Could you also show the output of `dmesg|grep mt` and `lsmod`?

Rtful, 2021-02-04:
@kakra `sudo udevadm monitor -p` just keeps scanning, when I manually start xow with `systemctl start xow`, I get this:
```
KERNEL[4350.154600] change   /devices/pci0000:00/0000:00:14.0/usb2/2-1 (usb)
ACTION=change
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-1
SUBSYSTEM=usb
DEVNAME=/dev/bus/usb/002/000
DEVTYPE=usb_device
DRIVER=usb
PRODUCT=45e/2fe/100
TYPE=0/0/0
BUSNUM=002
DEVNUM=000
SEQNUM=5033
MAJOR=189
MINOR=129

KERNEL[4350.298919] change   /devices/pci0000:00/0000:00:14.0/usb2/2-1 (usb)
ACTION=change
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-1
SUBSYSTEM=usb
DEVNAME=/dev/bus/usb/002/002
DEVTYPE=usb_device
DRIVER=usb
PRODUCT=45e/2fe/100
TYPE=0/0/0
BUSNUM=002
DEVNUM=002
SEQNUM=5034
MAJOR=189
MINOR=129

UDEV  [4350.303752] change   /devices/pci0000:00/0000:00:14.0/usb2/2-1 (usb)
ACTION=change
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-1
SUBSYSTEM=usb
DEVNAME=/dev/bus/usb/002/000
DEVTYPE=usb_device
DRIVER=usb
PRODUCT=45e/2fe/100
TYPE=0/0/0
BUSNUM=002
DEVNUM=000
SEQNUM=5033
USEC_INITIALIZED=5631894
ID_VENDOR=Microsoft_Inc.
ID_VENDOR_ENC=Microsoft\x20Inc.
ID_VENDOR_ID=045e
ID_MODEL=XBOX_ACC
ID_MODEL_ENC=XBOX\x20ACC
ID_MODEL_ID=02fe
ID_REVISION=0100
ID_SERIAL=Microsoft_Inc._XBOX_ACC_210793
ID_SERIAL_SHORT=210793
ID_BUS=usb
ID_USB_INTERFACES=:ffffff:
ID_VENDOR_FROM_DATABASE=Microsoft Corp.
ID_PATH=pci-0000:00:14.0-usb-0:1
ID_PATH_TAG=pci-0000_00_14_0-usb-0_1
ID_FOR_SEAT=usb-pci-0000_00_14_0-usb-0_1
MAJOR=189
MINOR=129
TAGS=seat
CURRENT_TAGS=seat

UDEV  [4350.307489] change   /devices/pci0000:00/0000:00:14.0/usb2/2-1 (usb)
ACTION=change
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-1
SUBSYSTEM=usb
DEVNAME=/dev/bus/usb/002/002
[... 115 lines cut ...]
fuse                  139264  5
crypto_user            16384  0
ip_tables              32768  0
x_tables               53248  1 ip_tables
ext4                  815104  2
crc32c_generic         16384  0
crc16                  16384  2 bluetooth,ext4
mbcache                16384  1 ext4
jbd2                  139264  1 ext4
sr_mod                 28672  0
cdrom                  77824  1 sr_mod
uas                    32768  0
usb_storage            81920  1 uas
hid_generic            16384  0
ata_generic            16384  0
usbhid                 65536  0
pata_acpi              16384  0
hid                   147456  2 usbhid,hid_generic
xhci_pci               20480  0
crc32c_intel           24576  4
xhci_hcd              286720  1 xhci_pci
ehci_pci               20480  0
ata_piix               36864  5
ehci_hcd               98304  1 ehci_pci
```