medusalix/xow #32: Dead Zones Joysticks

Mernst1, 2020-02-08:
Hello,

great job so far, i love it.
The only thing that bothers me is the dead zones on the joysticks.
Especially on the right side. If it is centered, it always gets low inputs down in games.
Is it possible to enlarge the dead zones or to implement them if they do not yet exist?

On bluetooth connection I have this behavior not.

Thank you

medusalix, 2020-02-08:
Deadzones should be handled by Linux' input system. I justed looked through [xpadneo](https://github.com/atar-axis/xpadneo)'s source and saw that they were [specifying](https://github.com/atar-axis/xpadneo/blob/4414f984997e38c40480790948f3a84bd75083fa/hid-xpadneo/src/hid-xpadneo.c#L1048) `fuzz` and `flat` values for the joysticks. That may lead to the behavior you are seeing. I'll make sure to get those changes into xow too, so you can test them.

medusalix, 2020-02-10:
@Mernst1 I've pushed some changes to the `deadzones` branch that should improve the input handling. Let me know if it works for you.

Mernst1, 2020-02-10:
Great! It's working.
Thank you.

closed 2020-02-10 by Mernst1