medusalix/xow #35: Controllers losing connection for a couple of seconds

zakaprov, 2020-02-11:
Operating system: Raspbian Buster
Kernel version: 4.19.75
Controller model: 1708
Dongle variant: slim
xow version: v0.3-10-gf06f669

With my last update of the driver (latest commit: https://github.com/medusalix/xow/commit/f06f669d38905c645c3ca1bcffdecff8a487bc67) all 4 of my controllers started randomly losing connection for a couple of seconds at a time.
After checking out the previous commit (https://github.com/medusalix/xow/commit/4edcee5e5bee2a467ef9126c0f85380d33345a42) and reinstalling xow the issue is no longer present and controllers maintain a stable connection.

I've attached xow debug logs from two sessions: one before downgrading (`xow-v0.3-10-gf06f669.log`) the driver and one after doing so (`xow-v0.3-9-g4edcee5.log`).

Let me know if there's anything I can do on my side to help with debugging this problem. Also, thank you for all the great work on this project!

[xow-v0.3-9-g4edcee5.log](https://github.com/medusalix/xow/files/4188836/xow-v0.3-9-g4edcee5.log)
[xow-v0.3-10-gf06f669.log](https://github.com/medusalix/xow/files/4188837/xow-v0.3-10-gf06f669.log)

Hedronmx, 2020-02-12:
Same here! Latest commit disconnects my controller for a second or two

NoXPhasma, 2020-02-15:
I can confirm this behaviour, going back to https://github.com/medusalix/xow/commit/f06f669d38905c645c3ca1bcffdecff8a487bc67 makes it work flawless again.

medusalix, 2020-02-15:
@NoXPhasma It's highly unlikely that 50b46c4fe407bd152e8eecf7bda0f458d950e904 caused any connection problems. I'll have to investigate how the changes in f06f669d38905c645c3ca1bcffdecff8a487bc67 lead to the random disconnects.

zakaprov, 2020-02-15:
@NoXPhasma I'm sure about that, I encountered the problem before https://github.com/medusalix/xow/commit/50b46c4fe407bd152e8eecf7bda0f458d950e904 was created.

NoXPhasma, 2020-02-15:
Alright, I was just curious as it seems not to happen for me with https://github.com/medusalix/xow/commit/f06f669d38905c645c3ca1bcffdecff8a487bc67. At least not inside a time period of around one hour.

medusalix, 2020-02-15:
I can't reproduce this issue with my 1708 controller and slim dongle. Left them connected for over 10 minutes and nothing happened. Commit f06f669d38905c645c3ca1bcffdecff8a487bc67 made some changes to the way disconnects are handled. Previous versions of the driver didn't detect when controllers got out of range (or somehow lost the connection). It just ignored the fact that these controllers were no longer sending any inputs. It must have something to do with that.

tmertens, 2020-02-16:
@medusalix I have been seeing this as well with my RetroPi. We've seen this with two controllers connected while actively playing games, so it doesn't seem related to the controllers not sending any inputs or going out of range.

I haven't tested to see if it happens when only one controller is connected, but can test that as well if it would be helpful.

andreashuetter, 2020-02-16:
Never had this issue until today.
Today this issue happened three times within only 1 hour to me after I had set the the steaming quality from "balanced" to "beautiful" (I am currently using the dongle on my Steam Link to stream the games from my Linux PC to my TV in the living room).
Possibly just a matter of coincidence, as I can think of no reason why the streaming quality could have an impact on this (I am using a wired Ethernet connection for streaming, no Wi-Fi). But after that I set the streaming quality back to "balanced", and played another 2 hours, without any issues.

djibux, 2020-02-16:
I confirm having this bug as well.  I'm happy to help debugging if you need.

fattredd, 2020-02-17:
>But after that I set the streaming quality back to "balanced", and played another 2 hours, without any issues.

I don't think this is related. I'm seeing the same disconnection issue without steam being a factor at all.

djibux, 2020-02-19:
@medusalix I don't know if that helps but leaving the issue doesn't seem to happen when the controller is idle / not being used.

Hedronmx, 2020-02-20:
> @medusalix I don't know if that helps but leaving the issue doesn't seem to happen when the controller is idle / not being used.
I can confirm this. only disconnects when using the controller

medusalix, 2020-02-20:
> I don't know if that helps but the issue doesn't seem to happen when the controller is idle / not being used.

That's definitely helpful, I'll look into it when I find the time.

djibux, 2020-02-21:
> After checking out the previous commit ([4edcee5](https://github.com/medusalix/xow/commit/4edcee5e5bee2a467ef9126c0f85380d33345a42)) and reinstalling xow the issue is no longer present and controllers maintain a stable connection.

What's the proper way to revert to a previous version?
I tried:
```
git checkout 4edcee5
make BUILD=RELEASE
sudo make install
sudo systemctl enable xow
sudo systemctl start xow
```
but I am still experiencing the issue.

pierrecollignon, 2020-02-22:
Same here, and going back to the working commit simply prevented me to connect the controller at all (but I might be doing it wrong).

Really good job nonetheless, please send me a message if you want some info about my setup.

medusalix, 2020-02-23:
Hey, I've got two things that might help me with this matter:

- What happens when you change the channel in the [Makefile](https://github.com/medusalix/xow/blob/a35977eaada7b6cf8191e47e9f6bf5473b9ee18a/Makefile#L3) to something like `165` (or one of the other available channels)?
- Do the disconnects still happen when you get really close to the dongle with your controller?

pierrecollignon, 2020-02-24:
- The disconnect still happen regardless of the distance it seems
- They're not too bad when playing a modern game with auto reconnect
- Here's a short session without changing the channel:
```2020-02-24 22:13:10 INFO  - xow v0.3-20-g2d00485 ©Severin v. W.
2020-02-24 22:13:10 INFO  - Dongle plugged in
2020-02-24 22:13:11 INFO  - Chip address: 62:45:b4:ea:a9:cf
2020-02-24 22:13:11 INFO  - Dongle initialized
2020-02-24 22:13:11 INFO  - Controller '1' connected
2020-02-24 22:19:34 INFO  - Controller '1' disconnected
2020-02-24 22:19:38 INFO  - Controller '1' connected
2020-02-24 22:25:47 INFO  - Controller '1' disconnected
2020-02-24 22:25:52 INFO  - Controller '1' connected
2020-02-24 22:28:47 INFO  - Controller '1' disconnected
2020-02-24 22:28:52 INFO  - Controller '1' connected
```

And after I switch to 165:
```
2020-02-24 22:32:15 INFO  - xow v0.3-20-g2d00485 ©Severin v. W.
2020-02-24 22:32:15 INFO  - Dongle plugged in
2020-02-24 22:32:15 INFO  - Chip address: 62:45:b4:ea:a9:cf
2020-02-24 22:32:15 INFO  - Dongle initialized
2020-02-24 22:32:22 INFO  - Controller '1' connected
2020-02-24 22:32:32 INFO  - Controller '1' disconnected
2020-02-24 22:32:36 INFO  - Controller '1' connected
2020-02-24 22:34:53 INFO  - Controller '1' disconnected
2020-02-24 22:34:58 INFO  - Controller '1' connected

```

pierrecollignon, 2020-02-24:
And I did a little change:
```
diff --git a/dongle/mt76.cpp b/dongle/mt76.cpp
index 7a65ea8..61cd882 100644
--- a/dongle/mt76.cpp
+++ b/dongle/mt76.cpp
@@ -230,7 +230,7 @@ void MT76::handleBulkPacket(const Bytes &packet)
 
         else if (info->eventType == EVT_CLIENT_LOST)
         {
-            handleClientLost(data);
+            //handleClientLost(data);
         }
 
         else if (info->eventType == EVT_BUTTON_PRESS)

```
And now it works perfectly, no deconnection:
```
2020-02-24 22:54:33 INFO  - xow v0.3-20-g2d00485 ©Severin v. W.
2020-02-24 22:54:34 INFO  - Dongle plugged in
2020-02-24 22:54:34 INFO  - Chip address: 62:45:b4:ea:a9:cf
2020-02-24 22:54:34 INFO  - Dongle initialized
2020-02-24 22:54:40 INFO  - Controller '1' connected
-> Ctrl +C
2020-02-24 23:06:53 ERROR - Packet for unconnected controller '1'
2020-02-24 23:06:53 INFO  - Dongle disconnected
```

- So it begs the question, I tried to find an m76 mcu doc, whatever that is, and couldn't. How did you figure 0x0e meant to deco the pad ? Wireshark seemed to say so ? Could it be a typo, feature of newer controllers, something the driver ignore on windows etc ?

l0ud, 2020-02-24:
I'm observing frequents disconnects too, they happen so often so driver is not usable in current form.
I noticed disconnect happen on CPU spikes, for example: turning on Steam - disconnect and reconnect, turning on game - the same, game compiling shaders=stutters -> disconnect and reconnect.
In my short gameplay, controller got somehow reconnected as second controller making it impossible to play any longer.
`lut 24 21:17:08 l0ud-pc xow[4209]: 2020-02-24 21:17:08 INFO  - Controller '1' disconnected
lut 24 21:17:12 l0ud-pc xow[4209]: 2020-02-24 21:17:12 INFO  - Controller '1' connected
lut 24 21:17:14 l0ud-pc xow[4209]: 2020-02-24 21:17:14 INFO  - Controller '1' disconnected
lut 24 21:17:17 l0ud-pc xow[4209]: 2020-02-24 21:17:17 INFO  - Controller '1' connected
lut 24 21:18:08 l0ud-pc xow[4209]: 2020-02-24 21:18:08 INFO  - Controller '1' disconnected
lut 24 21:18:13 l0ud-pc xow[4209]: 2020-02-24 21:18:13 INFO  - Controller '1' connected
lut 24 21:19:34 l0ud-pc xow[4209]: 2020-02-24 21:19:34 INFO  - Controller '1' disconnected
lut 24 21:19:39 l0ud-pc xow[4209]: 2020-02-24 21:19:39 INFO  - Controller '1' connected
lut 24 21:19:40 l0ud-pc xow[4209]: 2020-02-24 21:19:40 INFO  - Controller '2' connected
lut 24 21:19:46 l0ud-pc xow[4209]: 2020-02-24 21:19:46 INFO  - Controller '1' disconnected`

andreashuetter, 2020-02-24:
@l0ud: So you have disconnects on CPU spikes, and I have disconnects on my Steam Link only when the streaming quality is set to "beautiful", not when it's set to "balanced", that is only when the Steam Link is heavily utilized, but not when it's partially idle...

Looks like some kind of timeout-issue to me.

medusalix, 2020-02-24:
> Could you build a debug mode with more log if not already

You can run `make BUILD=DEBUG` to get a little bit of extra verbosity from xow. Logs are available via `journalctl -u xow -b`. Thanks for testing.

> Looks like some kind of timeout-issue to me, when the CPU is so heavily used that it takes longer than normal for the xow driver to respond...

Could be. You could ry giving the process a higher priority though.

I've made some general connection reliability/range improvements to xow (available in the [`reliability`](https://github.com/medusalix/xow/tree/reliability) branch). There's a *very* small chance that these changes will affect this issue.

l0ud, 2020-02-24:
@medusalix OK, so I performed some testing and I have good news.
Thread priority does not seem to be the cause. I tried boosting it and still got disconnects.
However, I tested reliability branch and played like 30 minutes without any problems.
I went back to master branch - instant disconnect on game launch.
Back to reliability - no issues at all. Tried changing locations in game to stress some CPU, still rock solid.
So, at least for my setup, you nailed it in reliability branch.

Mernst1, 2020-02-25:
Can confirm, it's working for me to now.

zakaprov, 2020-02-26:
@medusalix I tested the version from [`reliability`](https://github.com/medusalix/xow/tree/reliability) branch under very similar conditions as in my initial issue report (i.e. with two controllers used simultaneously to play). During the ~40 minute session I didn't encounter any problems, the connection on both controllers was solid.
Given the others' reports I'd say this resolves the issue and reliability can be merged into master. Thanks a lot for taking the time to fix this!

pierrecollignon, 2020-02-27:
Confirm, this is working perfectly now. I feel like windows. Is it because of the automated channel switching, you think ?

medusalix, 2020-02-28:
> Confirm, this is working perfectly now. I feel like windows. Is it because of the automated channel switching, you think ?

I'm really glad that it works 😄. I still need to do some testing before merging this into `master`. I'm not exactly sure what the code changes actually *do* in terms of the wireless connection and that's something I'd like to know. *If* these changes actually enabled automated channel switching, then it is probably implemented in the firmware.

zakaprov, 2020-02-28:
Got it, that makes sense. The remark about merging was overly-enthusiastic on my part I guess. :D In any case, I'll keep running the driver off `reliability` branch for now. Is there any reason for you to think this affects the channels being used?

djibux, 2020-02-28:
What’s the proper way to install a new version of the driver?  I am still experiencing issues after switching to `reliability` but I am not sure if I follow the proper process.  Do I need to uninstall the previous driver first?

zakaprov, 2020-02-28:
@djibux You can check your `xow` logs to see which version you're currently running.
Assuming you're running the driver as a `systemd` service, this will give you the `xow` logs since last machine boot: `sudo journalctl -u xow -b`.
The first log line should include the version, e.g.: `xow v0.3-9-g4edcee5`. The tail of the version string includes the first few characters of the last commit hash it's using (in my example that's `4edcee5`, exclude the `g` in the beginning). You can use it in GitHub's search to check which version of the program you're running.
As for installing the driver my usual routine goes something like this (assume running from `xow`'s project directory):
```
sudo systemctl stop xow
make clean
sudo make install
sudo systemctl start xow
```
Uninstalling the driver first (`sudo make uninstall`) won't do harm, so I guess you can try that before re-installing. Good luck!

djibux, 2020-02-29:
Thanks, I will try that.  I was not using `make clean`.

medusalix, 2020-02-29:
> Is there any reason for you to think this affects the channels being used?

Yes, definitely. The change that fixed this issue was most likely the introduction of [channel candidates](https://github.com/medusalix/xow/blob/a1fdf057c564601803bc864352b1f1a09030c987/dongle/mt76.cpp#L616). The Windows driver goes through a bunch of channels on startup, taking measurements that probably decide whether that specific channels ends up in the list of candidates (xow uses a hardcoded list for now).
I've captured the wireless traffic and noticed that the dongle is now operating on channel 165 despite being set to channel 1 in the Makefile. I'm not 100% percent sure how the switching between the channels works, as the dongle always chooses the first channel of the candidates.
I think earlier versions of xow never unleashed the full wireless power that the dongle could operate with, because the controllers always lost the signal being a couple of meters away.

Anyways, the changes have been pushed into `master`.

closed 2020-02-29 by medusalix