medusalix/xow #37: Can't pair Model 1708

Bur0k, 2020-02-15:
I am currently on Arch Linux and i tried both xow packages from the AUR and also directly compiled and installed from github.
Pairing works in a VM and in Windows directly but never saw a successful connection on Arch (Even tried another adapter + controller combo).

Slim Adapter + 1708.
How can i debug this problem?

`2020-02-15 19:06:45 INFO  - xow v0.3-15-g8ee82a4 ©Severin v. W.
2020-02-15 19:06:46 INFO  - Dongle plugged in
2020-02-15 19:06:46 DEBUG - Firmware already loaded
2020-02-15 19:06:46 DEBUG - Chip version: 7632
2020-02-15 19:06:46 INFO  - Chip address: 6c:5d:3a:00:d1:ca
2020-02-15 19:06:46 INFO  - Dongle initialized
2020-02-15 19:06:50 INFO  - Pairing initiated
2020-02-15 19:06:50 INFO  - Pairing initiated`

medusalix, 2020-02-16:
Pairing 1708 controllers should work fine. I've noticed that it sometimes needs about a *minute* to complete the pairing. Debugging the actual pairing process is unfortunately next to impossible without access to the source code of the dongle's firmware.

Bur0k, 2020-02-17:
Unfortunately i can't seem to pair the controller. I tried it several times until the pairing search process on the controller times out (rapid blinking to slower blinking) and never saw a single message on the console.

Is this easier to debug with a 5Ghz dongle so i can capture packets and what not? If yes, any cheap recommendations?

medusalix, 2020-02-17:
The first thing you could try is a simple Wireshark USB capture between the dongle and your system. Maybe the pairing packets get received by the dongle but aren't recognized by xow. If the capture doesn't show any signs of the pairing process you'll have to do a wireless capture (2.4 GHz will work, xow uses channel 1 by default).

Bur0k, 2020-02-17:
What i know until now:

USB packets get sent to dongle for initialization of the pairing, LED in dongle flashes.
I tried to capture packets on 2.4GHz Channel 1 but either there were none from the dongle or i couldn't find them.
Therefore the controller probably never finds the dongle and therefore the dongle never sends USB packets back to the PC.

medusalix, 2020-02-17:
You should at least see the beacon frames of the dongle when you capture on channel 1. You've put your network interface into monitor mode, right?

Bur0k, 2020-02-17:
I did and i also see other traffic on Channel 1.
I tried to identify the beacon packets with the help of https://gist.github.com/devkid/4b3bd50760504d1b93ea684cfd3ed895 and this is where i got the idea of the 5GHz dongle.
Is it correct that i am searching for a packet with that exact vendor specific tag?

medusalix, 2020-02-17:
Yes, that's correct. The beacon data is hardcoded to the following:

https://github.com/medusalix/xow/blob/e8a6d7513df30ce9f71e1e9edda16e5baf8f58c9/dongle/mt76.cpp#L725-L731

You could also try changing the wireless channel in the [Makefile](
https://github.com/medusalix/xow/blob/e8a6d7513df30ce9f71e1e9edda16e5baf8f58c9/Makefile#L3) (take a look at the list of valid channels in [mt76.h](https://github.com/medusalix/xow/blob/e8a6d7513df30ce9f71e1e9edda16e5baf8f58c9/dongle/mt76.h#L749)).
Another strange thing I've noticed is that the MAC address of your dongle starts with `6c:5d:3a` whereas most of the addresses I've seen from other people start with either `62:45:b4` or `62:45:b5`. Maybe you're using a different variant/type of dongle that's somehow incompatible with the way xow works?

Bur0k, 2020-02-17:
Tried changing channels and so on a little while ago.
Could be that there is a even newer version. Two of the brand new "Controller + Wireless Adapter for Windows 10" included dongles started with 6c, cant remember what exactly the other one was.
So i guess the dongle is the problem here and not the Controller....

When i press the pairing button on the dongle, is it expected to get two messages for "Pairing initiated"? So two messages for pressing it and two for releasing it

medusalix, 2020-02-18:
> When i press the pairing button on the dongle, is it expected to get two messages for "Pairing initiated"?

Yes, that happens in Microsoft's driver too. It doesn't cause any harm to initiate the pairing multiple times, so I didn't bother differentiating between button presses and releases.

> So i guess the dongle is the problem here and not the Controller....

Very likely. Have you checked whether the firmware in `C:\Windows\System32\DriverStore\FileRepository\mt7612us_xxxxx` is the same as the binary file included with xow? I've noticed that Microsoft distributes different packages of their driver with distinct firmware images.

Bur0k, 2020-02-19:
> Very likely. Have you checked whether the firmware in `C:\Windows\System32\DriverStore\FileRepository\mt7612us_xxxxx` is the same as the binary file included with xow? I've noticed that Microsoft [...]
Just checked and diff'ed it. They are the same

medusalix, 2020-02-20:
> Just checked and diff'ed it. They are the same

Then I'm unfortunately out of ideas of what we could try to make this work. Unless I'll be able to get my hands on one of these dongles, I don't think there's much I can do to help you. I'd like to keep this issue open in case someone else has something important to share.

x4mers, 2020-02-22:
Maybe the dongles should be referred to by model number, instead of slim and bulky, since there seems to be more than 2 now. On the edge of the dongle opposite the sync button, there is black writing. At the top right corner of that edge, mine says Microsoft and just below it says Model 1713. I'm pretty sure mine is the old bulky one. Anyone with the slim one can provide theirs? Bur0k, what does yours say?

medusalix, 2020-02-23:
I think we're dealing with different revisions/versions of the two models here. Take a look at [Microsoft's FCC applications](https://fccid.io/C3K). Their application ids always refer to the model number of a specific product. In case of the old dongle, it's [C3K1713](https://fccid.io/C3K1713). I couldn't spot more than two models of the dongle.

Bur0k, 2020-02-23:
> Maybe the dongles should be referred to by model number, instead of slim and bulky, since there seems to be more than 2 now. On the edge of the dongle opposite the sync button, there is black [...]
Mine is 1790

> > Just checked and diff'ed it. They are the same
> 
> Then I'm unfortunately out of ideas of what we could try to make this work confused. Unless I'll be able to get my hands on one of these dongles, I don't think there's much I can do to help you. I'd like to keep this issue open in case someone else has something important to share.
If you want i can mail it to you and when you are done i'll provide the label to send it back

iroedius, 2020-02-26:
Just to confirm, I also have the 1790 version of the dongle and my Elite 2 also won't pair.

Hielyr, 2020-03-21:
My dongle model is a 1790 and I'm able to pair just fine. 

Are there some logs or anything that I can provide that could possibly help narrow down why I can pair my Elite v2 while using the 1790 dongle and others can't?

Bur0k, 2020-03-21:
```
2020-03-21 23:20:32 DEBUG - ASIC version: 7632
2020-03-21 23:20:32 DEBUG - Chip id: 7613
2020-03-21 23:20:32 INFO  - Wireless address: 6c:5d:3a:00:d1:ca
```
Maybe these three informations from the log could be usefull

medusalix, 2020-03-24:
@Bur0k Can you provide me with a packet capture of the USB communication (both on Windows and on Linux with xow)? I might be able to spot some differences that explain why it's not working as intended. It might take a while for me to look through all the packets though.

Bur0k, 2020-03-26:
> @Bur0k Can you provide me with a packet capture of the USB communication (both on Windows and on Linux with xow)? I might be able to spot some differences that explain why it's not working as intended. It might take a while for me to look through all the packets though. May I ask where and when you bought the dongle? That might help to narrow down the problems.

The two capture files in the attached zip archive.
[pcapng.zip](https://github.com/medusalix/xow/files/4387534/pcapng.zip)

I bought the Controller+Dongle bundle from otto(dot)de

medusalix, 2020-04-07:
@Bur0k While looking through the packet captures of people that had similar issues, I've noticed a **very** strange behavior of the Windows driver: The source address of the beacon frames is (in your case) *not* the actual MAC address read from the dongle. It's `62:45:bd:00:d1:ca` despite the reading from the EEPROM being `6c:5d:3a:00:d1:ca`. That might explain why the controllers didn't connect to your dongle.

medusalix, 2020-04-08:
@WernerAUT I've implemented a fix in the latest commit, you might want to check it out. I also added a check make sure only one instance of xow is running at a time.

WernerAUT, 2020-04-08:
tested it already right now and it works perfect, controller connected after seconds! Well done

iroedius, 2020-04-08:
Elite 2 controller also paired right away.

sharkwouter, 2020-04-08:
I was having the same issue, but that commit fixed it for me as well.

Bur0k, 2020-04-08:
Updated fixed it, thanks for all your investigations and work

closed 2020-04-08 by Bur0k

BrianPHChen, 2020-04-09:
Raspbian tested fine, thanks a lot