medusalix/xow #39: xow stops working after suspending and waking up the computer

webodan, 2020-02-26:
Hi,

Basically title is what's happening. I've noticed that after waking up the computer and trying to sync the controller, xow would not work, so I rebooted the service using `systemctl` and it's working fine again. This might not be appropriate for a retropie style build like the one I'm trying to make, so it would be fine if the program was able to be aware that it resumed from sleep and kept working. 

I'm using the reliability branch btw.
Also a big thank you for the author of this software as I'm finally able to use my xbox controller wirelessly on Linux! <3

medusalix, 2020-02-29:
I've done some testing and it seems like `libusb` is returning `LIBUSB_TRANSFER_NO_DEVICE` after waking up the computer. This usually only happens when you physically remove the dongle. I'll probably have to make some adjustments to xow to handle this case.

medusalix, 2020-03-08:
@webodan Please try out the [`suspend`](https://github.com/medusalix/xow/tree/suspend) branch (make sure that the updated service definition gets loaded). While I couldn't find a way for xow to recover after a suspend, I made sure that the service restarts when the system wakes up.

medusalix, 2020-03-14:
@NoXPhasma and @andreashuetter, could you try this out for me? I made some significant changes to the USB backend (making it simpler) which will definitely need testing. As an inevitable result of fixing the suspend issue, xow does now terminate when you wake from sleep or unplug the dongle and is restarted automatically by `systemd`.

NoXPhasma, 2020-03-15:
The service stops when the dongle gets removed, but does not restart when I plug in the dongle again.

```
Mär 15 04:30:31 NoXP-PC systemd[1]: Started Xbox One Wireless Dongle Driver.
Mär 15 04:30:31 NoXP-PC xow[533968]: 2020-03-15 04:30:31 INFO  - xow v0.3-27-g28cbc4d ©Severin v. W.
Mär 15 04:30:31 NoXP-PC xow[533968]: 2020-03-15 04:30:31 INFO  - Dongle plugged in
Mär 15 04:30:31 NoXP-PC xow[533968]: 2020-03-15 04:30:31 INFO  - Wireless address: 62:45:b4:fe:81:16
Mär 15 04:30:31 NoXP-PC xow[533968]: 2020-03-15 04:30:31 INFO  - Dongle initialized
Mär 15 04:30:35 NoXP-PC xow[533968]: 2020-03-15 04:30:35 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
Mär 15 04:30:35 NoXP-PC xow[533968]: 2020-03-15 04:30:35 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
Mär 15 04:30:35 NoXP-PC systemd[1]: xow.service: Succeeded.
```

medusalix, 2020-03-15:
Make sure that you run `systemctl enable xow` and `systemctl restart xow` after installing the branch to reload the service config.

NoXPhasma, 2020-03-15:
Yes, I've done that. It just doesn't restart itself after I remove the dongle and reconnect it.

NoXPhasma, 2020-03-15:
Okay I found the issue, `sudo make install` won't install the new service file which includes `Restart=always`. Once I update the service file manually it works.

medusalix, 2020-03-15:
It should override them. When I checkout `master`, do `install` and then checkout `suspend` followed by `install` it replaces the contents of the service.

NoXPhasma, 2020-03-15:
I expected the same here as well, but for some reason it didn't. I even tried to install it two times to get sure.

medusalix, 2020-03-15:
Oh well, I'm just going to add a notice to run `sudo make uninstall` to the README to make sure everyone's covered.

medusalix, 2020-03-18:
Merged into `master` (196371dc40683e52af572347a94bd8193fa2a28a).

closed 2020-03-18 by medusalix