medusalix/xow #41: xow generates too long Rumble in some games

NoXPhasma, 2020-03-04:
In some games, in my case I can reproduce it in [SteamWorld Dig 2](https://store.steampowered.com/app/571310/SteamWorld_Dig_2/), the rumble is **way** too long. Instead of a short light rumble, when hitting a rock, with the controller connected by wire, it rumbles for several seconds. And even overlaps and makes the rumble more intense. 

Debug log output:
```
Mär 04 02:38:34 NoXP-PC systemd[1]: Started Xbox One Wireless Dongle Driver.
Mär 04 02:38:34 NoXP-PC xow[31807]: 2020-03-04 02:38:34 INFO  - xow v0.3-21-ga35977e ©Severin v. W.
Mär 04 02:38:34 NoXP-PC xow[31807]: 2020-03-04 02:38:34 INFO  - Dongle plugged in
Mär 04 02:38:34 NoXP-PC xow[31807]: 2020-03-04 02:38:34 DEBUG - Firmware already loaded, resetting...
Mär 04 02:38:34 NoXP-PC xow[31807]: 2020-03-04 02:38:34 DEBUG - Firmware loaded
Mär 04 02:38:34 NoXP-PC xow[31807]: 2020-03-04 02:38:34 DEBUG - Chip version: 7632
Mär 04 02:38:34 NoXP-PC xow[31807]: 2020-03-04 02:38:34 INFO  - Chip address: 62:45:b4:fe:81:16
Mär 04 02:38:34 NoXP-PC xow[31807]: 2020-03-04 02:38:34 INFO  - Dongle initialized
Mär 04 02:38:38 NoXP-PC xow[31807]: 2020-03-04 02:38:38 DEBUG - Client associating: 7e:ed:81:4f:f7:3c
Mär 04 02:38:38 NoXP-PC xow[31807]: 2020-03-04 02:38:38 INFO  - Controller '1' connected
Mär 04 02:38:39 NoXP-PC xow[31807]: 2020-03-04 02:38:39 INFO  - Serial number: 02600108383810
Mär 04 02:38:39 NoXP-PC xow[31807]: 2020-03-04 02:38:39 DEBUG - Battery type: 1, level: 2
Mär 04 02:38:40 NoXP-PC xow[31807]: 2020-03-04 02:38:40 DEBUG - Battery type: 1, level: 2
Mär 04 02:38:41 NoXP-PC xow[31807]: 2020-03-04 02:38:41 DEBUG - Battery type: 1, level: 2
Mär 04 02:38:42 NoXP-PC xow[31807]: 2020-03-04 02:38:42 DEBUG - Feedback length: 100, delay: 0, direction: 0, weak: 62, strong: 62
Mär 04 02:38:43 NoXP-PC xow[31807]: 2020-03-04 02:38:43 DEBUG - Battery type: 1, level: 2
Mär 04 02:38:44 NoXP-PC xow[31807]: 2020-03-04 02:38:44 DEBUG - Battery type: 1, level: 2
Mär 04 02:38:45 NoXP-PC xow[31807]: 2020-03-04 02:38:45 DEBUG - Feedback length: 100, delay: 0, direction: 0, weak: 61, strong: 61
Mär 04 02:38:47 NoXP-PC xow[31807]: 2020-03-04 02:38:47 DEBUG - Feedback length: 100, delay: 0, direction: 0, weak: 63, strong: 63
Mär 04 02:38:47 NoXP-PC xow[31807]: 2020-03-04 02:38:47 DEBUG - Feedback length: 100, delay: 0, direction: 0, weak: 57, strong: 57
Mär 04 02:38:48 NoXP-PC xow[31807]: 2020-03-04 02:38:48 DEBUG - Feedback length: 200, delay: 0, direction: 0, weak: 25, strong: 25
Mär 04 02:38:48 NoXP-PC xow[31807]: 2020-03-04 02:38:48 DEBUG - Feedback length: 100, delay: 0, direction: 0, weak: 58, strong: 58
Mär 04 02:38:48 NoXP-PC xow[31807]: 2020-03-04 02:38:48 DEBUG - Battery type: 1, level: 2
Mär 04 02:38:48 NoXP-PC xow[31807]: 2020-03-04 02:38:48 DEBUG - Feedback length: 150, delay: 0, direction: 0, weak: 9, strong: 9
Mär 04 02:38:48 NoXP-PC xow[31807]: 2020-03-04 02:38:48 DEBUG - Feedback length: 200, delay: 0, direction: 0, weak: 25, strong: 25
Mär 04 02:38:48 NoXP-PC xow[31807]: 2020-03-04 02:38:48 DEBUG - Feedback length: 100, delay: 0, direction: 0, weak: 68, strong: 68
Mär 04 02:38:49 NoXP-PC xow[31807]: 2020-03-04 02:38:49 DEBUG - Feedback length: 500, delay: 0, direction: 0, weak: 134, strong: 134
Mär 04 02:38:49 NoXP-PC xow[31807]: 2020-03-04 02:38:49 DEBUG - Feedback length: 200, delay: 0, direction: 0, weak: 105, strong: 105
Mär 04 02:38:49 NoXP-PC xow[31807]: 2020-03-04 02:38:49 DEBUG - Feedback length: 200, delay: 0, direction: 0, weak: 51, strong: 51
Mär 04 02:38:49 NoXP-PC xow[31807]: 2020-03-04 02:38:49 DEBUG - Feedback length: 200, delay: 0, direction: 0, weak: 27, strong: 27
Mär 04 02:38:50 NoXP-PC xow[31807]: 2020-03-04 02:38:50 DEBUG - Feedback length: 200, delay: 0, direction: 0, weak: 14, strong: 14
Mär 04 02:39:03 NoXP-PC xow[31807]: 2020-03-04 02:39:03 DEBUG - Feedback length: 100, delay: 0, direction: 0, weak: 69, strong: 69
Mär 04 02:39:04 NoXP-PC xow[31807]: 2020-03-04 02:39:04 DEBUG - Feedback length: 100, delay: 0, direction: 0, weak: 69, strong: 69
Mär 04 02:39:08 NoXP-PC xow[31807]: 2020-03-04 02:39:08 DEBUG - Battery type: 1, level: 2
Mär 04 02:39:13 NoXP-PC xow[31807]: 2020-03-04 02:39:13 DEBUG - Terminating device...
Mär 04 02:39:13 NoXP-PC systemd[1]: Stopping Xbox One Wireless Dongle Driver...
Mär 04 02:39:13 NoXP-PC xow[31807]: 2020-03-04 02:39:13 ERROR - Packet for unconnected controller '1'
Mär 04 02:39:13 NoXP-PC xow[31807]: 2020-03-04 02:39:13 INFO  - Dongle disconnected
Mär 04 02:39:13 NoXP-PC systemd[1]: xow.service: Succeeded.
Mär 04 02:39:13 NoXP-PC systemd[1]: Stopped Xbox One Wireless Dongle Driver.
```

Tell me if you need any more informations.

medusalix, 2020-03-06:
Hmm, I'm not sure how I can fix this in xow. The game should be sending a force feedback command with a magnitude of zero (weak: 0, strong: 0) to stop the rumbling (which it doesn't do). It probably expects the rumbling to stop after the specified `length` parameter (ignored by xow). Ideally, we'd need something like [`ff-memless`](https://github.com/torvalds/linux/blob/master/drivers/input/ff-memless.c) for user space applications to handle the timing. I don't want to implement this from scratch.

We could try mapping the feedback's `length` parameter (probably in milliseconds) to the rumble's [`duration`](https://github.com/medusalix/xow/blob/c8a1554a1de6ad3cf77bc7f50104f9bc5d147a43/controller/controller.cpp#L103) field. They don't match up 1:1, so we'd have to introduce some kind of scaling factor.

medusalix, 2020-05-28:
Please give the `rumble` branch a try.
I've decided to pass the rumble's `length` and `delay` parameters directly to the controller which should work better for games that support it. Most games don't specify the `length` value at all, so we fall back to the maximum duration in that case.
While testing the force feedback in various games, I've noticed that the controller sometimes refuses to stop the rumble until it is requested to rumble again with a different strength. I don't know if I can fix this in the driver.

NoXPhasma, 2020-05-28:
In SteamWorld Dig 2 it still rumbles pretty long.

```
Mai 28 14:40:36 NoXP-PC xow[3125790]: 2020-05-28 14:40:36 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:36 NoXP-PC xow[3125790]: 2020-05-28 14:40:36 DEBUG - Rumble strong: 7946, weak: 7946, direction: 0
Mai 28 14:40:36 NoXP-PC xow[3125790]: 2020-05-28 14:40:36 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:36 NoXP-PC xow[3125790]: 2020-05-28 14:40:36 DEBUG - Rumble strong: 19242, weak: 19242, direction: 0
Mai 28 14:40:41 NoXP-PC xow[3125790]: 2020-05-28 14:40:41 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:41 NoXP-PC xow[3125790]: 2020-05-28 14:40:41 DEBUG - Rumble strong: 0, weak: 0, direction: 0
Mai 28 14:40:46 NoXP-PC xow[3125790]: 2020-05-28 14:40:46 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:46 NoXP-PC xow[3125790]: 2020-05-28 14:40:46 DEBUG - Rumble strong: 8000, weak: 8000, direction: 0
Mai 28 14:40:46 NoXP-PC xow[3125790]: 2020-05-28 14:40:46 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:46 NoXP-PC xow[3125790]: 2020-05-28 14:40:46 DEBUG - Rumble strong: 18818, weak: 18818, direction: 0
Mai 28 14:40:49 NoXP-PC xow[3125790]: 2020-05-28 14:40:49 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:49 NoXP-PC xow[3125790]: 2020-05-28 14:40:49 DEBUG - Rumble strong: 18824, weak: 18824, direction: 0
Mai 28 14:40:50 NoXP-PC xow[3125790]: 2020-05-28 14:40:50 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:50 NoXP-PC xow[3125790]: 2020-05-28 14:40:50 DEBUG - Rumble strong: 19044, weak: 19044, direction: 0
Mai 28 14:40:52 NoXP-PC xow[3125790]: 2020-05-28 14:40:52 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:52 NoXP-PC xow[3125790]: 2020-05-28 14:40:52 DEBUG - Rumble strong: 17818, weak: 17818, direction: 0
Mai 28 14:40:53 NoXP-PC xow[3125790]: 2020-05-28 14:40:53 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:53 NoXP-PC xow[3125790]: 2020-05-28 14:40:53 DEBUG - Rumble strong: 16284, weak: 16284, direction: 0
Mai 28 14:40:53 NoXP-PC xow[3125790]: 2020-05-28 14:40:53 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:53 NoXP-PC xow[3125790]: 2020-05-28 14:40:53 DEBUG - Rumble strong: 16164, weak: 16164, direction: 0
Mai 28 14:40:58 NoXP-PC xow[3125790]: 2020-05-28 14:40:58 DEBUG - Rumble count: 1, duration: 65535, delay: 0
Mai 28 14:40:58 NoXP-PC xow[3125790]: 2020-05-28 14:40:58 DEBUG - Rumble strong: 0, weak: 0, direction: 0
Mai 28 14:41:03 NoXP-PC xow[3125790]: 2020-05-28 14:41:03 DEBUG - Rumble count: 0, duration: 65535, delay: 0
```

medusalix, 2020-05-28:
It doesn't look like you're playing the same game based on your logs. I'm seeing a constant duration of 65535 whereas your original post had much smaller length values in the range of 100 to 200. There must be something else going on.

NoXPhasma, 2020-05-28:
I've noticed that as well, but it's the same game. And there was [no update to the game](https://steamdb.info/app/571310/history/) in the meantime. I'll try if different versions of xow are the cause for the different duration.

NoXPhasma, 2020-05-28:
Turns out it was a combination of native runtime and Steam Controller API enabled which caused that high duration time. Enabling Steam runtime and disabling Steam Controller API brought the 200 duration back and the `rumble` branch works then.

medusalix, 2020-05-28:
Alright, thanks for testing. Merge into `master`.

closed 2020-05-28 by medusalix