medusalix/xow #44: Chatpad support

Kitrinx, 2020-03-20:
The XBONE Wireless dongle supports official chatpad functionality on windows PCs. Is chatpad support in the scope of this driver? It would be quite useful for multi-purpose HTPC type setups and small gaming/computing platforms like MiSTer FPGA.

medusalix, 2020-03-20:
Yes, definitely. I'm going to need a packet capture (using Wireshark) where you plug in the chatpad and press some buttons on it. That will help me to understand the protocol it's using.

medusalix, 2020-04-03:
@TyanColte I found your comment regarding packet captures of the chatpad on `360Controller`'s issue tracker. I'd like to implement chatpad support for this driver, would you be so kind as to do a USBPcap capture for me? I can't get my hands on a chatpad as they seem to have been discontinued.

medusalix, 2020-05-08:
Closing for lack of packet captures and hardware availability.

HeyPabloW, 2020-06-29:
Hi.  Attached contains three Wireshark USBPcap1 packet captures for the Xbox Elite Series 2 with official chatpad, connected to a PC via official 2.4g Windows 10 wireless adapter.  (Also includes another three capture for the standard Xbox One controller.  Line 1 and 2 device request and response events happen immediately, so the chatpad presses may be in the middle of the log.  Let me know if you need any other captures or if these didn't turn out quite right.

[Xbox Chatpad capture.zip](https://github.com/medusalix/xow/files/4846602/Xbox.Chatpad.capture.zip)

medusalix, 2020-07-01:
@HeyPabloW Would you mind doing another capture where you plug in the chatpad after you've started the capturing process? The authentication process is really hard for me to figure out so this would be a tremendous help. Ideally, I'd also need a list of the keys you've pressed so I can associate them to the respective chatpad packet.
I'm a bit busy right now, so it might take some time for me to analyze the captures.

HeyPabloW, 2020-07-01:
No problem at all.  In these captures I used the below steps.  Let me know if I can provide any other captures.

- Connected controller to 2.4g external adapter
- Launched Wireshark and USBPcap1 capture
- Attached Chatpad
- Typed QWERTY

[Xbox chaptpad v2.zip](https://github.com/medusalix/xow/files/4857352/Xbox.chaptpad.v2.zip)

medusalix, 2020-07-04:
OK, so here's a short summary of the things I've found:

- The input packet for the chatpad is 8 bytes long.
- The third byte contains the key you've pressed on the chatpad or zero if the key is released. All other bytes were zero for the packets you've captured.
- `Q` is encoded as 0x14 (20), `W` as 0x1a (26), `E` as 0x08 (8) and so forth. That means `A` would be 0x03 (3), I wonder what keys 1 and 2 would refer to.
- There are still 7 unknown bytes in the input packet. They might be used for modifiers like shift or special keys.

HeyPabloW, 2020-07-05:
Great. Glad those steps produce a useful capture. Is there any benefit to a more detailed capture to progress support for the driver? Pressing every key on the pad, for example.

medusalix, 2020-07-11:
I'd like to add full support for the chatpad to xow eventually, but I guess for a first implementation the current captures will suffice. I'm still busy right now, but I'll let you know once I've got the implementation ready.

joaodforce, 2020-08-09:
I have a Xbox One S ChatPad too, if someone wants me to test something out I make myself available, It would be great to use the volume keys on the Chatpad to control system volume, even windows doesn't support that.

OrpheeGT, 2020-12-18:
Hello,

I have an Xbox Elite 2 series controller with chatpad on my Windows 10.
I'm waiting the Xbox one Adapter (currently using bluetooth for controller only, and USB when chatpad connected as it does not work with bluetooth)

I would be happy to help or test once I receive the dongle.

Thanks

sofakng, 2021-03-15:
Also looking forward to seeing this supported ... it would be extremely useful!

mikegreiling, 2021-03-18:
I would also be interested in this. I own a chat pad myself and would be happy to contribute to this effort @medusalix

medusalix, 2021-03-18:
I'm currently working on a new kernel driver that will eventually replace xow. Chatpad support is on my todo list.

medusalix, 2021-06-04:
I've added preliminary chatpad support to the new driver and I really need some testers to make sure it works correctly. I've invited @Kitrinx, @joaodforce and @sofakng to the project, it would be great if you could share your thoughts over on [Discord](https://discord.gg/FDQxwWk).
For all the others: Feel free to message me if you want to get access to the repository.

medusalix, 2022-03-20:
I've released a new driver for the Xbox wireless dongle that supports the chatpad. You can check it out [here](https://github.com/medusalix/xone).