medusalix/xow #45: Model 1537 not connecting

Yomas, 2020-03-21:
I'm using Ubuntu LTS 18.04 and Xbox one controller 1537 and compiled xov from github. The controller was paired on a Windows 10 machine successfully to the same dongle. 
When I try to connect my controller in Linux it doesn't work. The controller is associated multiple times with a new ID.

Is there a way to debug this or is there a simple fix?

2020-03-21 16:35:00 INFO  - xow v0.4 ©Severin v. W.
2020-03-21 16:35:00 DEBUG - Opening device...
2020-03-21 16:35:00 INFO  - Dongle plugged in
2020-03-21 16:35:00 DEBUG - Firmware already loaded, resetting...
2020-03-21 16:35:00 DEBUG - Firmware loaded
2020-03-21 16:35:00 DEBUG - ASIC version: 7632
2020-03-21 16:35:00 DEBUG - Chip id: 7613
2020-03-21 16:35:00 INFO  - Wireless address: 6c:5d:3a:01:dc:dd
2020-03-21 16:35:00 INFO  - Dongle initialized
2020-03-21 16:35:10 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:10 INFO  - Controller '1' connected
2020-03-21 16:35:15 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:15 INFO  - Controller '2' connected
2020-03-21 16:35:20 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:20 INFO  - Controller '3' connected
2020-03-21 16:35:25 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:25 INFO  - Controller '4' connected
2020-03-21 16:35:30 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:30 INFO  - Controller '5' connected
2020-03-21 16:35:34 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:34 INFO  - Controller '6' connected
2020-03-21 16:35:39 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:39 INFO  - Controller '7' connected
2020-03-21 16:35:44 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:44 INFO  - Controller '8' connected
2020-03-21 16:35:49 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:49 INFO  - Controller '9' connected
2020-03-21 16:35:54 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:54 INFO  - Controller '10' connected
2020-03-21 16:35:59 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:35:59 INFO  - Controller '11' connected
2020-03-21 16:36:04 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:36:04 INFO  - Controller '12' connected
2020-03-21 16:36:09 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:36:09 INFO  - Controller '13' connected
2020-03-21 16:36:14 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:36:14 INFO  - Controller '14' connected
2020-03-21 16:36:19 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:36:19 INFO  - Controller '15' connected
2020-03-21 16:36:23 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:36:23 INFO  - Controller '16' connected
2020-03-21 16:36:28 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-21 16:36:28 ERROR - All WCIDs are taken
2020-03-21 16:36:28 ERROR - Failed to associate client

medusalix, 2020-03-22:
Is your controller running the latest firmware version? Please check with the Xbox Accessories app.

Yomas, 2020-03-23:
I've updated the controller firmware to 2.3.2381.0 as suggested but the controller will still not connect. But now I get a pairing message.
I've tried the following:
- Pair the controller on a Windows machine, swap dongle to Linux machine (controller on and off during swap): Nothing happens
- Pair the controller on Linux: I now get a pairing message, but after that the behavior is the same as before 

2020-03-23 15:21:45 INFO  - xow v0.4 ©Severin v. W.
2020-03-23 15:21:45 DEBUG - Opening device...
2020-03-23 15:21:46 INFO  - Dongle plugged in
2020-03-23 15:21:46 DEBUG - Firmware loaded
2020-03-23 15:21:46 DEBUG - ASIC version: 7632
2020-03-23 15:21:46 DEBUG - Chip id: 7613
2020-03-23 15:21:46 INFO  - Wireless address: 6c:5d:3a:01:dc:dd
2020-03-23 15:21:46 INFO  - Dongle initialized
2020-03-23 15:22:37 INFO  - Pairing initiated
2020-03-23 15:22:37 INFO  - Pairing initiated
2020-03-23 15:22:37 INFO  - Pairing initiated
2020-03-23 15:22:37 INFO  - Pairing initiated
2020-03-23 15:23:46 DEBUG - Client pairing: 7e:ed:85:de:de:c0
2020-03-23 15:23:49 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-23 15:23:49 INFO  - Controller '1' connected
2020-03-23 15:23:49 DEBUG - Client disassociating: 7e:ed:85:de:de:c0
2020-03-23 15:23:49 INFO  - Controller '1' disconnected
2020-03-23 15:23:49 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-23 15:23:49 INFO  - Controller '1' connected
2020-03-23 15:23:54 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-23 15:23:54 INFO  - Controller '2' connected
2020-03-23 15:23:58 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-03-23 15:23:58 INFO  - Controller '3' connected
2020-03-23 15:24:03 DEBUG - Client associating: 7e:ed:85:de:de:c0
[...]

medusalix, 2020-03-24:
Alright, it looks like I'm going to need a packet capture from you (on Windows). I've summarized the instructions down below:

- Install [Wireshark](https://www.wireshark.org/). Make sure to tick the option to install **USBPcap** in the installer.
- At the bottom of the Wireshark window you should see multiple USBPcap entries.
- You have to identify which one is the correct one by clicking on one of them to start the capture.
- Now plug in the dongle. If you see lots of messages appearing, then it's probably the correct one. Otherwise you have to repeat the procedure with one of the other USBPcap entries.
- Turn on the controller and wait for it to connect.
- Stop the capture and send me a link to your `pcap` file.

Yomas, 2020-03-26:
Will do as soon as I get my hands on the windows machine again. Sadly I currently have no access to a Windows 10 machine.

medusalix, 2020-03-31:
Could you please check if the latest changes to xow made any difference regarding your problem?

Yomas, 2020-04-01:
Sadly the changes did not fix the issue but I noticed a small difference. After pairing the xbox button stays lit for ~1 second before starting to blink again. Trying to connect a paired controller still shows no visible actions.

I've created the logs you requested:
https://filebin.net/ku6exhnd3z7ic2vm
There are two logs, one from pairing the controller and one from connecting the (paired) controller

Log from connecting with latest changes:
2020-04-01 18:38:44 INFO  - xow v0.4-4-g141b10c ©Severin v. W.
2020-04-01 18:38:51 DEBUG - Opening device...
2020-04-01 18:38:51 INFO  - Dongle plugged in
2020-04-01 18:38:51 DEBUG - Firmware loaded
2020-04-01 18:38:51 DEBUG - ASIC version: 7632
2020-04-01 18:38:51 DEBUG - Chip id: 7613
2020-04-01 18:38:51 INFO  - Wireless address: 6c:5d:3a:01:dc:dd
2020-04-01 18:38:51 INFO  - Dongle initialized
2020-04-01 18:39:31 INFO  - Pairing initiated
2020-04-01 18:39:31 INFO  - Pairing initiated
2020-04-01 18:39:32 INFO  - Pairing initiated
2020-04-01 18:39:32 INFO  - Pairing initiated
2020-04-01 18:39:37 DEBUG - Client pairing: 7e:ed:85:de:de:c0
2020-04-01 18:39:39 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-01 18:39:39 INFO  - Controller '1' connected
2020-04-01 18:39:39 DEBUG - Client disassociating: 7e:ed:85:de:de:c0
2020-04-01 18:39:39 INFO  - Controller '1' disconnected
2020-04-01 18:39:39 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-01 18:39:39 INFO  - Controller '1' connected
2020-04-01 18:39:44 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-01 18:39:44 INFO  - Controller '2' connected
2020-04-01 18:39:49 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-01 18:39:49 INFO  - Controller '3' connected

SerafinTech, 2020-04-03:
I'm having the same issue.

medusalix, 2020-04-04:
@Yomas I've looked through the captures and noticed something that *could* be the cause of the problems. Would you mind doing a packet capture on Linux too? That might help me to narrow down my findings.

Yomas, 2020-04-04:
Sure no problem (hoping I captured the right device):
https://filebin.net/spxgqp2w1m6tqvfx

medusalix, 2020-04-05:
Thanks. I've pushed a fix to the [`probe`](https://github.com/medusalix/xow/tree/probe) branch, let me know if it works.

Yomas, 2020-04-05:
Sadly no.

Pairing:
2020-04-05 14:04:15 INFO  - xow v0.4-7-g8da08ae ©Severin v. W.
2020-04-05 14:04:15 DEBUG - Opening device...
2020-04-05 14:04:16 INFO  - Dongle plugged in
2020-04-05 14:04:16 DEBUG - Firmware already loaded, resetting...
2020-04-05 14:04:16 DEBUG - Firmware loaded
2020-04-05 14:04:16 DEBUG - ASIC version: 7632
2020-04-05 14:04:16 DEBUG - Chip id: 7613
2020-04-05 14:04:16 INFO  - Wireless address: 6c:5d:3a:01:dc:dd
2020-04-05 14:04:16 INFO  - Dongle initialized
2020-04-05 14:04:19 INFO  - Pairing initiated
2020-04-05 14:04:19 INFO  - Pairing initiated
2020-04-05 14:04:19 INFO  - Pairing initiated
2020-04-05 14:04:19 INFO  - Pairing initiated
2020-04-05 14:04:24 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-05 14:04:24 INFO  - Controller '1' connected
2020-04-05 14:04:24 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:04:24 DEBUG - Client disassociating: 7e:ed:85:de:de:c0
2020-04-05 14:04:24 INFO  - Controller '1' disconnected
2020-04-05 14:04:27 DEBUG - Client pairing: 7e:ed:85:de:de:c0
2020-04-05 14:04:30 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-05 14:04:30 INFO  - Controller '1' connected
2020-04-05 14:04:30 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:04:30 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:04:33 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:04:34 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-05 14:04:34 INFO  - Controller '2' connected
2020-04-05 14:04:34 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:04:35 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:04:38 DEBUG - Client probe request: 7e:ed:85:de:de:c0

Connecting paired controller:
2020-04-05 14:03:00 INFO  - xow v0.4-7-g8da08ae ©Severin v. W.
2020-04-05 14:03:00 DEBUG - Opening device...
2020-04-05 14:03:00 INFO  - Dongle plugged in
2020-04-05 14:03:00 DEBUG - Firmware already loaded, resetting...
2020-04-05 14:03:00 DEBUG - Firmware loaded
2020-04-05 14:03:00 DEBUG - ASIC version: 7632
2020-04-05 14:03:00 DEBUG - Chip id: 7613
2020-04-05 14:03:00 INFO  - Wireless address: 6c:5d:3a:01:dc:dd
2020-04-05 14:03:00 INFO  - Dongle initialized
2020-04-05 14:03:06 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-05 14:03:06 INFO  - Controller '1' connected
2020-04-05 14:03:06 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:07 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:10 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:11 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-05 14:03:11 INFO  - Controller '2' connected
2020-04-05 14:03:11 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:11 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:14 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:15 DEBUG - Client associating: 7e:ed:85:de:de:c0
2020-04-05 14:03:15 INFO  - Controller '3' connected
2020-04-05 14:03:15 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:16 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:19 DEBUG - Client probe request: 7e:ed:85:de:de:c0
2020-04-05 14:03:20 DEBUG - Client associating: 7e:ed:85:de:de:c0

medusalix, 2020-04-08:
@Yomas I've just fixed an issue that caused problems for users with a wireless address starting with `6c:5d:3a`. Could you try out the latest changes from master?

SerafinTech, 2020-04-08:
Your latest fix worked for me.  I did need to pair it 4 times before it started working.

Yomas, 2020-04-09:
That did it for me as well. Paired on the first try.

Thanks for the support.

medusalix, 2020-04-09:
I'm glad that it's working :smiley:.

closed 2020-04-09 by medusalix