medusalix/xow #47: Dongle 1790 issue? LIBUSB_ERROR_NO_DEVICE

WernerAUT, 2020-04-02:
i am not sure if this is a dongle issue or my setup issue? LIBUSB_ERROR_NO_DEVICE could that be a different issue?
I tried to connect with 

Xbox One Wireless Controller 1708
Xbox One Wireless Controller Elite 2 1797

```
2020-04-02 14:43:05 INFO  - xow v0.4-5-g222063c ©Severin v. W.
2020-04-02 14:43:05 DEBUG - Opening device...
2020-04-02 14:43:05 INFO  - Dongle plugged in
2020-04-02 14:43:05 DEBUG - Firmware already loaded, resetting...
2020-04-02 14:43:05 DEBUG - Firmware loaded
2020-04-02 14:43:05 DEBUG - ASIC version: 7632
2020-04-02 14:43:05 DEBUG - Chip id: 7613
2020-04-02 14:43:05 INFO  - Wireless address: 6c:5d:3a:01:e7:81
2020-04-02 14:43:05 INFO  - Dongle initialized
2020-04-02 14:43:08 INFO  - Pairing initiated
2020-04-02 14:43:08 INFO  - Pairing initiated
2020-04-02 14:43:08 INFO  - Pairing initiated
2020-04-02 14:43:08 INFO  - Pairing initiated
2020-04-02 14:44:05 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
2020-04-02 14:44:05 DEBUG - Device error, terminating...
2020-04-02 14:44:05 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
2020-04-02 14:44:05 DEBUG - Device error, terminating...
2020-04-02 14:44:05 ERROR - Error in bulk write: LIBUSB_ERROR_IO
2020-04-02 14:44:05 DEBUG - Device error, terminating...
2020-04-02 14:44:05 ERROR - Failed to write command
2020-04-02 14:44:05 ERROR - Failed to burst write register
2020-04-02 14:44:05 ERROR - Failed to write beacon
2020-04-02 14:44:05 ERROR - Failed to write pairing beacon
```

medusalix, 2020-04-03:
It's probably an issue with your Linux system or `libusb`. The `LIBUSB_ERROR_NO_DEVICE` error code is usually thrown when you disconnect the dongle. You might want to try to plug the dongle into another USB port to see if it works.

WernerAUT, 2020-04-03:
i tried already different ports but its the same, i will install on a different pc a different Linux Distribution to test, but i think maybe it is same as others have also issue with 1790 Dongle

WernerAUT, 2020-04-03:
tested now on 

Linux Mint 19.3
Ubuntu 20.04 beta
Manjaro 19.0.2

on all the same issue no connect

FYI: Adapter was bought from Microsoft Shop

BrianPHChen, 2020-04-07:
I met the same problem

WernerAUT, 2020-04-07:
@BrianPHChen do you have Dongle Version? would be also good wehn you post also terminal output like i did, so first uninstall xow and ten build xow with DEBUG

medusalix, 2020-04-07:
@WernerAUT Could you do a USB packet capture of the dongle using Wireshark for me (on Linux where it's failing)? I can't promise to fix this problem but I'd like to take a closer look.

BrianPHChen, 2020-04-07:
OK, I build a debug version one and post the information later.
BTW, I used latest raspbian

WernerAUT, 2020-04-07:
@medusalix i hope that is ok
[linux.zip](https://github.com/medusalix/xow/files/4443498/linux.zip)

BrianPHChen, 2020-04-07:
Raspbian version: Feb 2020
Getting the same message to @WernerAUT 
[image]

WernerAUT, 2020-04-07:
so yours is also the 1790 dongle, maybe medusalix can find out the issue, would be great to use this dongle :) else i need to look around and find a older version

BrianPHChen, 2020-04-07:
Yeah, I'm trying to make raspberry pi as a portable game station. So, really want to make the dongle work for multiple players.

medusalix, 2020-04-07:
Are you running `./xow` while still having the system service enabled?

WernerAUT, 2020-04-07:
yes, you are right that is the issue with LIBUSB_ERROR_NO_DEVICE i made now

sudo systemctl stop xow
sudo systemctl disable xow

and started then from terminal xow and now no LIBUSB_ERROR_NO_DEVICE anymore, but controller still doesn't connect, when i start pairing the controller blinks fast and then after sometime it starts blinking slow again, dongle also still blinks with same speed.

I made another USB packet capture with Wireshark now.
[linux1.pcapng.zip](https://github.com/medusalix/xow/files/4445755/linux1.pcapng.zip)

WernerAUT, 2020-04-07:
I made now also for you a USB packet capture with Wireshark in Windows, I installed Windows today on one of my PC :)
[Windows.pcapng.zip](https://github.com/medusalix/xow/files/4445787/Windows.pcapng.zip)

there it is pretty fast and connect without issues

medusalix, 2020-04-07:
Thank you for the captures. It seems like you're having the exact same problem as described in issue #37, so I suggest we close this issue in favor of the existing one.

WernerAUT, 2020-04-07:
so also the Mac Adress issue, ok i am fine to close it.

closed 2020-04-07 by WernerAUT