medusalix/xow #48: Question: what is the use of firmware.bin? Will it still work with windows after using this ?

Rathna-K, 2020-04-04:
Question: what is the use of firmware.bin? Will it still work with windows after using this or will it break the windows functionality of the dongle? Nice work btw. Thank you!

medusalix, 2020-04-04:
The `firmware.bin` file is loaded onto the wireless chip inside the dongle (MediaTek MT76xx) every time you plug the dongle into your computer. No matter if you use the original Windows driver or the Linux driver, the chip needs the firmware to operate. The firmware included with xow is in fact the exact same file you would get if you extracted the builtin Windows driver, so the use of xow doesn't alter the functionality of the dongle in any way.

Rathna-K, 2020-04-04:
Got it, so like its on volatile memory so the OS loads the firmware every time I guess? Thanks for clarifying.

medusalix, 2020-04-04:
Yes, exactly. It even allows you to use the dongle as a standard WiFi dongle by running a different [driver](https://github.com/openwrt/mt76) and firmware.

closed 2020-04-04 by medusalix