medusalix/xow #49: OSMC LIBUSB_ERROR_NO_DEVICE

HPRIOR, 2020-04-06:
The driver is successful while running on Raspbian, but not on kodi/OSMC (on the same machine - raspberry pi 3 b+). 

osmc version:
4.19.55-6-osmc

dmesg:
[2.452045] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[2.583039] usb 1-1.2: New USB device found, idVendor=045e, idProduct=02e6, bcdDevice= 1.00
[2.583047] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[2.583051] usb 1-1.2: Product: XBOX ACC
[2.583056] usb 1-1.2: Manufacturer: Microsoft Inc.
[2.583060] usb 1-1.2: SerialNumber: 461454

When I issue /usr/local/bin/xow:
2020-04-05 13:15:19 INFO - xow v0.4-5-g222063c ©Severin v. W.
2020-04-05 13:15:19 INFO - Dongle plugged in
2020-04-05 13:15:19 INFO - Wireless address: 62:45:b4:f9:3d:fb 
2020-04-05 13:15:19 INFO - Dongle initialized
2020-04-05 13:15:20 INFO - Controller '1' connected
2020-04-05 13:15:21 INFO - Product ID: 02dd
2020-04-05 13:16:20 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE
2020-04-05 13:16:20 ERROR - Error in bulk read: LIBUSB_ERROR_NO_DEVICE

I had a back and forth with someone on a forum with limited success. The controller would go straight to flashing the Xbox button and the dongle would light up when the controller was turned on, but no connection would be made. I was told to issue sed -i 's@/lib/modules-load.d@/etc/modules-load.d@' Makefile and rebuild and reinstall. After doing this the controller connected  briefly upon the next restart (possibly just a fluke, as I don't really know what this does), but then continued with the same behaviour afterwards  (being disconnected).

medusalix, 2020-04-07:
You should *never* run xow from `/usr/local/bin` manually, as it interferes with the system service. A simple reboot after running `sudo systemctl start xow` (as described in the README) is usually enough to get it running. To my knowledge, this can lead to the `LIBUSB_ERROR_NO_DEVICE` you are getting.

WernerAUT, 2020-04-07:
thats correct i can confirm that LIBUSB_ERROR_NO_DEVICE won't show up if you don't run xow in terminal with 

sudo systemctl stop xow
sudo systemctl disable xow

HPRIOR, 2020-04-07:
> You should _never_ run xow from `/usr/local/bin` manually, as it interferes with the system service. A simple reboot after running `sudo systemctl start xow` (as described in the README) is usually enough to get it running. To my knowledge, running multiple instances of xow can lead to the `LIBUSB_ERROR_NO_DEVICE` error you are getting.

Sorry I should have said in my initial comment that the steps taken after the discussion on a forum were because it did not work initially (on OSMC) after following the steps in the README

HPRIOR, 2020-04-07:
> thats correct i can confirm that LIBUSB_ERROR_NO_DEVICE won't show up if you run xow in terminal with
> 
> sudo systemctl stop xow
> sudo systemctl disable xow

Have you tried on OSMC?

WernerAUT, 2020-04-07:
no sorry for hijacking your report

BrianPHChen, 2020-04-08:
@HPRIOR I still can't pair my controller to the dongle on the raspbian. How do you get this done?
I follow all the steps on README and reboot the machine.
 
My dongle number is 1790 and controller number is 1708

WernerAUT, 2020-04-08:
@BrianPHChen i think you have the same issue (MAC different to EPROM) as most with the new 1790 dongle, you should follow https://github.com/medusalix/xow/issues/37 medusalix is working on a fix

HPRIOR, 2020-04-08:
> @HPRIOR I still can't pair my controller to the dongle on the raspbian. How do you get this done?
> I follow all the steps on README and reboot the machine.
> 
> My dongle number is 1790 and controller number is 1708

My dongle is 1713 and my controller model is 1697. So maybe it's an issue with with your hardware. Otherwise, are there any steps in the installation process that you were not confident with?

HPRIOR, 2020-04-09:
Would it help to make an add-on for kodi?

medusalix, 2020-04-09:
I don't think it needs any addons to function properly. The issue could be that OSMC is based on Debian *Stretch*, while Raspbian is actually Debian *Buster* (newer version). There might be a bug inside `libusb` that was fixed in newer Debian releases. They are apparently working on [upgrading to Buster](https://discourse.osmc.tv/t/osmc-update-to-buster/81364).

HPRIOR, 2020-04-09:
> I don't think it needs any addons to function properly. The issue could be that OSMC is based on Debian _Stretch_, while Raspbian is actually Debian _Buster_ (newer version). There might be a bug [...]

Ok. I guess I'll have to wait then!

medusalix, 2020-04-26:
I'll close this for now, but please feel free to reopen this issue if you still encounter this bug after upgrading to Debian Buster.

closed 2020-04-26 by medusalix