medusalix/xow #50: make BUILD=RELEASE fatal error

rqtruong, 2020-04-07:
Hi there, thanks for making this driver! I was able to install the package but make BUILD=RELEASE gave this fatal error:
`make BUILD=RELEASE
g++  -Wall -Wpedantic -std=c++11 -MMD -O3 -DCHANNEL=1 -DVERSION=\"v0.4-6-ga3dcb38\" -c -o xow.o xow.cpp
In file included from xow.cpp:20:0:
dongle/usb.h:28:10: **fatal error**: libusb-1.0/libusb.h: No such file or directory
 #include **<libusb-1.0/libusb.h>**
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:30: recipe for target 'xow.o' failed
make: *** [xow.o] Error 1
`
This is on Linux Mint 19.3.

WernerAUT, 2020-04-07:
follow this please i wrote what is needed under Mint 19.3
https://github.com/medusalix/xow/issues/46

rqtruong, 2020-04-07:
Thanks! It was libusb-1.0-0-dev that I was missing.

closed 2020-04-07 by rqtruong