medusalix/xow #51: Cant pair Model 1797 (newest Firmware)

AlterDaddler, 2020-04-07:
Hi,

first of all thnx for the great work! 
My XBOX Elite Controller (2019) cant be paired. It seems that the dongle cant find any device:

2020-04-08 00:46:06 INFO  - xow v0.4-6-ga3dcb38 ©Severin v. W.
2020-04-08 00:46:06 DEBUG - Opening device...
2020-04-08 00:46:06 INFO  - Dongle plugged in
2020-04-08 00:46:06 DEBUG - Firmware already loaded, resetting...
2020-04-08 00:46:06 DEBUG - Firmware loaded
2020-04-08 00:46:06 DEBUG - ASIC version: 7632
2020-04-08 00:46:06 DEBUG - Chip id: 7613
2020-04-08 00:46:06 INFO  - Wireless address: 6c:5d:3a:00:1b:f2
2020-04-08 00:46:06 INFO  - Dongle initialized
2020-04-08 00:46:16 INFO  - Pairing initiated
2020-04-08 00:46:16 INFO  - Pairing initiated
2020-04-08 00:46:16 INFO  - Pairing initiated
2020-04-08 00:46:16 INFO  - Pairing initiated

Daniel

WernerAUT, 2020-04-08:
you have the same issue as most with the new 1790 dongle have, as it seems dongles starting with MAC Address 6c: have this issue, you should follow #37 medusalix is working on a fix

AlterDaddler, 2020-04-08:
Oh didnt see that... sry.. i close this one. Thnx

closed 2020-04-08 by AlterDaddler