medusalix/xow #52: Segmentation fault when exiting xow (ctrl+c)

ddragic, 2020-04-11:
I noticed every time I close xow with CTRL+C (when running from cli) there is a segfault:
```
ddragic:pts/1:xow $ ./xow 
2020-04-11 14:03:28 INFO  - xow v0.4-9-g7675112 ©Severin v. W.
2020-04-11 14:03:30 DEBUG - Opening device...
2020-04-11 14:03:30 INFO  - Dongle plugged in
2020-04-11 14:03:30 DEBUG - Firmware loaded
2020-04-11 14:03:31 DEBUG - ASIC version: 7632
2020-04-11 14:03:31 DEBUG - MAC version: 7662
2020-04-11 14:03:31 DEBUG - Chip id: 7613
2020-04-11 14:03:31 INFO  - Wireless address: 62:45:b5:01:9a:09
2020-04-11 14:03:31 INFO  - Dongle initialized
2020-04-11 14:03:34 DEBUG - Client associating: 7e:ed:8c:1c:8f:74
2020-04-11 14:03:34 INFO  - Controller '1' connected
2020-04-11 14:03:34 INFO  - Product ID: 02ea
2020-04-11 14:03:34 DEBUG - Firmware version: 3.1.1221.0
2020-04-11 14:03:34 DEBUG - Hardware version: 1284.1.1.1
2020-04-11 14:03:35 INFO  - Serial number: 03600079292806
2020-04-11 14:03:38 DEBUG - Battery type: 1, level: 2
2020-04-11 14:03:58 DEBUG - Battery type: 1, level: 2
^C2020-04-11 14:03:59 DEBUG - Stop signal received
2020-04-11 14:03:59 DEBUG - Closing device...
2020-04-11 14:03:59 INFO  - Dongle power-off
xow: io.c:2116: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
Segmentation fault (core dumped)
```
Sometimes the last line is `Aborted (core dumped)` instead of Segmentation fault.
Everything seems to power down correctly, the controller turns off and so does the adapter (light goes out), but it always segfaults.

This is on Fedora 31 x64 with libusbx-1.0.22-4.fc31.

gdb backtrace:
```
#0  discard_urbs (itransfer=itransfer@entry=0x7f9194002fb0, first=first@entry=0, last_plus_one=1) at os/linux_usbfs.c:1830
#1  0x00007f91a1a8b7ed in op_cancel_transfer (itransfer=0x7f9194002fb0) at os/linux_usbfs.c:2256
#2  0x00007f91a1a890d9 in libusb_cancel_transfer (transfer=<optimized out>) at io.c:1572
#3  0x00007f91a1a8b03d in sync_transfer_wait_for_completion (transfer=transfer@entry=0x7f9194003020) at sync.c:57
#4  0x00007f91a1a8b10b in do_sync_bulk_transfer (dev_handle=0x7f919c000de0, endpoint=<optimized out>, buffer=buffer@entry=0x7f919bffeca0 "\006", length=length@entry=512, 
    transferred=transferred@entry=0x7f919bffec5c, timeout=timeout@entry=0, type=2 '\002') at sync.c:191
#5  0x00007f91a1a8b423 in libusb_bulk_transfer (dev_handle=<optimized out>, endpoint=<optimized out>, data=data@entry=0x7f919bffeca0 "\006", length=length@entry=512, 
    transferred=transferred@entry=0x7f919bffec5c, timeout=timeout@entry=0) at sync.c:273
#6  0x0000000000410059 in UsbDevice::bulkRead (this=this@entry=0x7ffe9fe69980, endpoint=endpoint@entry=5 '\005', buffer=...) at /usr/include/c++/9/array:234
#7  0x000000000040c76d in Mt76::readBulkPackets (this=0x7ffe9fe69980, endpoint=5 '\005') at dongle/mt76.cpp:950
#8  0x000000000040c806 in std::__invoke_impl<void, void (Mt76::*)(unsigned char), Mt76*, int> (__t=<optimized out>, __f=<optimized out>)
    at /usr/include/c++/9/bits/invoke.h:89
#9  std::__invoke<void (Mt76::*)(unsigned char), Mt76*, int> (__fn=<optimized out>) at /usr/include/c++/9/bits/invoke.h:95
#10 std::thread::_Invoker<std::tuple<void (Mt76::*)(unsigned char), Mt76*, int> >::_M_invoke<0ul, 1ul, 2ul> (this=<optimized out>) at /usr/include/c++/9/thread:244
#11 std::thread::_Invoker<std::tuple<void (Mt76::*)(unsigned char), Mt76*, int> >::operator() (this=<optimized out>) at /usr/include/c++/9/thread:251
#12 std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (Mt76::*)(unsigned char), Mt76*, int> > >::_M_run (this=<optimized out>)
    at /usr/include/c++/9/thread:195
#13 0x00007f91a1cb83d4 in execute_native_thread_routine () from /lib64/libstdc++.so.6
#14 0x00007f91a1a664e2 in start_thread () from /lib64/libpthread.so.0
#15 0x00007f91a197b6a3 in clone () from /lib64/libc.so.6
```
gdb backtrace full:
```
#0  discard_urbs (itransfer=itransfer@entry=0x7f9194002fb0, first=first@entry=0, last_plus_one=1) at os/linux_usbfs.c:1830
        transfer = 0x7f9194003020
        tpriv = 0x7f9194003060
        dpriv = 0x50
        i = 0
        ret = 0
        urb = 0x7f9194002770
        __FUNCTION__ = "discard_urbs"
#1  0x00007f91a1a8b7ed in op_cancel_transfer (itransfer=0x7f9194002fb0) at os/linux_usbfs.c:2256
        tpriv = 0x7f9194003060
        transfer = 0x7f9194003020
        r = <optimized out>
#2  0x00007f91a1a890d9 in libusb_cancel_transfer (transfer=<optimized out>) at io.c:1572
        itransfer = <optimized out>
[... 117 lines cut ...]
            0 '\000', 0 '\000', 0 '\000', 0 '\000', 0 '\000', 0 '\000', 0 '\000', 0 '\000', 0 '\000', 0 '\000'...}}
        buffer = {data = {
            _M_elems = "d\000\b\000\222`\000\000\001HA\000\020\000\a@\301ǣ\002@\255\225\225\000R\301\307\000\000\000\000\000\000\000\000\210\021<\000bE\265\001\232\t~\355\214\034\217t`E\265\001\232\t\020\000\000\000\000\000\003 \001#\206\003\000\000\002\003\000\000\000\000\000\000\000\000\000\001\000\002\001\v", '\000' <repeats 16 times>, "d\000\b", '\000' <repeats 404 times>}}
        transferred = <optimized out>
#12 0x000000000040c806 in std::__invoke_impl<void, void (Mt76::*)(unsigned char), Mt76*, int> (__t=<optimized out>, __f=<optimized out>)
    at /usr/include/c++/9/bits/invoke.h:89
No locals.
#13 std::__invoke<void (Mt76::*)(unsigned char), Mt76*, int> (__fn=<optimized out>) at /usr/include/c++/9/bits/invoke.h:95
No locals.
#14 std::thread::_Invoker<std::tuple<void (Mt76::*)(unsigned char), Mt76*, int> >::_M_invoke<0ul, 1ul, 2ul> (this=<optimized out>) at /usr/include/c++/9/thread:244
No locals.
#15 std::thread::_Invoker<std::tuple<void (Mt76::*)(unsigned char), Mt76*, int> >::operator() (this=<optimized out>) at /usr/include/c++/9/thread:251
No locals.
#16 std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (Mt76::*)(unsigned char), Mt76*, int> > >::_M_run (this=<optimized out>)
    at /usr/include/c++/9/thread:195
No locals.
#17 0x00007f301f4d43d4 in execute_native_thread_routine () from /lib64/libstdc++.so.6
No symbol table info available.
#18 0x00007f301f2824e2 in start_thread () from /lib64/libpthread.so.0
No symbol table info available.
#19 0x00007f301f1976a3 in clone () from /lib64/libc.so.6
No symbol table info available.
```

I hope some of this is helpful.

medusalix, 2020-04-14:
Thanks for the report, I've seen this issue a couple of times but couldn't reproduce it reliably. I suppose it happens because I run two threads that read from the USB device using `libusb_bulk_transfer` which get interrupted when `libusb_exit` is called during Ctrl+C handling. There's no way to stop those threads before exiting because `libusb_bulk_transfer` itself is blocking, so I'm not sure how to prevent the segementation fault that apparently occurs in `libusb` code.

ddragic, 2020-04-14:
Well, if you figure something out just let me know, I can test it since it's reproducible pretty reliably on this system.

medusalix, 2020-04-29:
I just rewrote most of the USB code of xow (it was really unstable). Could you please check if you still get segmentation faults in the [`usb`](https://github.com/medusalix/xow/tree/usb) branch?

ddragic, 2020-04-29:
Just tried it, works great, no segfaults when closing with ctrl+c or in any other case (unplugging the dongle etc.) . Although it's spamming the controller serial number about 4 times per second :smile: 
```
2020-04-30 00:59:29 INFO  - xow v0.4-15-g62d3743 ©Severin v. W.
2020-04-30 00:59:29 INFO  - Waiting for device...
2020-04-30 00:59:29 DEBUG - Opening device...
2020-04-30 00:59:29 DEBUG - Firmware already loaded, resetting...
2020-04-30 00:59:29 DEBUG - Firmware loaded
2020-04-30 00:59:29 DEBUG - ASIC version: 7632
2020-04-30 00:59:29 DEBUG - MAC version: 7662
2020-04-30 00:59:29 DEBUG - Chip id: 7613
2020-04-30 00:59:29 INFO  - Wireless address: 62:45:b5:01:9a:09
2020-04-30 01:01:58 INFO  - Controller '1' connected
2020-04-30 01:01:59 INFO  - Device announced, product id: 02ea
2020-04-30 01:01:59 DEBUG - Firmware version: 3.1.1221.0
2020-04-30 01:01:59 DEBUG - Hardware version: 1284.1.1.1
2020-04-30 01:01:59 INFO  - Serial number: 02600309943829
2020-04-30 01:01:59 INFO  - Serial number: 02600309943829
2020-04-30 01:02:00 INFO  - Serial number: 02600309943829
2020-04-30 01:02:00 INFO  - Serial number: 02600309943829
2020-04-30 01:02:00 INFO  - Serial number: 02600309943829
2020-04-30 01:02:00 INFO  - Serial number: 02600309943829
2020-04-30 01:02:01 INFO  - Serial number: 02600309943829
2020-04-30 01:02:01 INFO  - Serial number: 02600309943829
2020-04-30 01:02:01 INFO  - Serial number: 02600309943829
2020-04-30 01:02:01 INFO  - Serial number: 02600309943829
2020-04-30 01:02:02 INFO  - Serial number: 02600309943829
2020-04-30 01:02:02 INFO  - Serial number: 02600309943829
2020-04-30 01:02:02 INFO  - Serial number: 02600309943829
2020-04-30 01:02:02 DEBUG - Battery type: 1, level: 2
2020-04-30 01:02:02 INFO  - Serial number: 02600309943829
2020-04-30 01:02:03 INFO  - Serial number: 02600309943829
2020-04-30 01:02:03 INFO  - Serial number: 02600309943829
...
^C2020-04-30 01:03:08 INFO  - Shutting down...
2020-04-30 01:03:08 DEBUG - Closing device...
```

As a side note, I also noticed the dongle power usage now drops when xow exits. Previously current draw only slightly decreased and remained at about 160mA, but now it seems to go to power save/sleep mode and drops to ~18mA.

Would be possible to put the dongle in low power mode when there are no controllers connected, like windows does? I'll probably open a new feature request since it's not directly related to this issue.

medusalix, 2020-04-30:
> Although it's spamming the controller serial number about 4 times per second :smile:

Hmm, this might be caused by a change in commit c4ec92d9f5cdc7d0d13eab8c2b9d1f5e5a963c0e, not the USB rewrite.

> As a side note, I also noticed the dongle power usage now drops when xow exits.

Yep, I'm setting the radio's power mode to `RADIO_OFF` when xow shuts down. I've noticed that the dongle got quite warm even when xow was not running, so this should be fixed by now.

> Would be possible to put the dongle in low power mode when there are no controllers connected, like windows does?

Yes, I'll probably have to do a packet capture to figure out how to do that. The radio's datasheet mentions a `Wake on WLAN` feature that probably needs to be enabled.

ddragic, 2020-04-30:
> Hmm, this might be caused by a change in commit c4ec92d, not the USB rewrite.

You're right. I commented out the serial packet handling for now, it was annoying.

> Yep, I'm setting the radio's power mode to RADIO_OFF when xow shuts down. I've noticed that the dongle got quite warm even when xow was not running, so this should be fixed now.

Awesome, thanks. I noticed it too, that's why I measured the current to see the difference between xow and the windows driver. 

Should we keep this issue open until this work is merged in master?

medusalix, 2020-04-30:
Can you check if commit b148d9b3c32f36f8e8ba3eb3673d8cd07628c749 fixes the serial number issue? I'll close this issue once that's solved and I've merged my work.

ddragic, 2020-04-30:
No, b148d9b doesn't help, it's the same.

medusalix, 2020-04-30:
Please try changing this line to `out.pad(4)`:
https://github.com/medusalix/xow/blob/c4ec92d9f5cdc7d0d13eab8c2b9d1f5e5a963c0e/controller/gip.cpp#L217

I honestly don't know what specific part of that commit caused this behavior.

ddragic, 2020-04-30:
Changing the padding did it, now it's ok.

---
Just my 2c, but you probably shouldn't mess with git history in the public repo too much (if at all), it can create problems for others when they sync their local copies of the repo.
It took me a second to figure out that master HEAD now points to the previous commit and not b148d9b and the change on the usb branch earlier created a merge conflict.

medusalix, 2020-04-30:
Yeah, sorry about that. I didn't feel like cluttering the commit history by reverting that last commit, so I force pushed master (as quickly as possible to keep the number of clones small). I'll avoid force pushing feature branches in the future.

closed 2020-05-01 by medusalix