medusalix/xow #53: coredump after connecting

PetrusZ, 2020-04-14:
➜ sudo xow
Alias tip: _ xow
2020-04-14 14:03:25 INFO  - xow  ©Severin v. W.
2020-04-14 14:03:25 INFO  - Dongle plugged in
2020-04-14 14:03:25 INFO  - Wireless address: 62:45:b4:e7:09:7f
2020-04-14 14:03:25 INFO  - Dongle initialized
terminate called after throwing an instance of 'InputException'
  what():  Error opening device: No such file or directory
[1]    385131 abort      sudo xow

[core.xow.0.2cebbca1920c48598f01d3ba74272143.385132.1586844208000000000000.lz4.zip](https://github.com/medusalix/xow/files/4473458/core.xow.0.2cebbca1920c48598f01d3ba74272143.385132.1586844208000000000000.lz4.zip)

PetrusZ, 2020-04-14:
BTW, xow version is 0.4

medusalix, 2020-04-14:
You have to make sure that `/dev/uinput` is available (that's the reason for the `InputException`). The `uinput` kernel module should be loaded automatically when you boot your system, if it's not, you could try running `modprobe uinput` to see if it can be loaded manually.

PetrusZ, 2020-04-14:
Thank you, problem fixed.

closed 2020-04-14 by PetrusZ