medusalix/xow #54: xow crashing when started with systemctl

avdz, 2020-04-17:
Hi,

I've an issue with xow:
I can't seem to start xow with systemctl. It exits after receiving a permission denied error.
I can start it without issue manually.

`pi@SteamLink:~ $ sudo journalctl -u xow -b -f
-- Logs begin at Fri 2020-04-17 18:42:59 CEST. --
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 DEBUG - Opening device...
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 INFO  - Dongle plugged in
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 DEBUG - Firmware already loaded, resetting...
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 DEBUG - Firmware loaded
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 DEBUG - Invalid MAC address, correcting...
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 DEBUG - ASIC version: 7632
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 DEBUG - MAC version: 7662
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 DEBUG - Chip id: 7613
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 INFO  - Wireless address: 62:45:bd:02:25:4e
Apr 17 18:56:21 SteamLink xow[1395]: 2020-04-17 18:56:21 INFO  - Dongle initialized
Apr 17 18:56:34 SteamLink xow[1395]: 2020-04-17 18:56:34 DEBUG - Client associating: 7e:ed:82:4d:8e:67
Apr 17 18:56:34 SteamLink xow[1395]: terminate called after throwing an instance of 'InputException'
Apr 17 18:56:34 SteamLink xow[1395]:   what():  Error opening device: Permission denied
Apr 17 18:56:34 SteamLink systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Apr 17 18:56:34 SteamLink systemd[1]: xow.service: Failed with result 'signal'.`

It seems that it is similar to #23 
I've tried following commands to trigger the udev rules again but it does not change the permission of the /dev/uinput file 
'sudo udevadm control --reload-rules
sudo udevadm trigger'

Why could this be?

medusalix, 2020-04-17:
Have you done a restart after installing xow? As far as I know the udev rules only trigger when `/dev/uinput` is created when the system boots.

avdz, 2020-04-17:
Yes I did. 
Is it possible that this steamlink rule is alse changing the /dev/uinput permission?
`sudo cat /lib/udev/rules.d/55-steamlink.rules
# USB devices
SUBSYSTEM=="usb", GROUP="plugdev"

# HID devices
KERNEL=="hidraw*", GROUP="input", MODE:="0660"`

medusalix, 2020-04-17:
> Is it possible that this steamlink rule is alse changing the /dev/uinput permission?

Probably not, as these rules only affect USB and HID devices.
Please try replacing the last line in `/lib/udev/rules.d/99-xow.rules` with the following (and reboot):

```
KERNEL=="uinput", SUBSYSTEM=="misc", MODE="0666", OPTIONS+="static_node=uinput"
```

avdz, 2020-04-20:
Nope still 
pi@SteamLink:~ $ ls -la /dev/uinput
crw-rw----+ 1 root input 10, 223 Apr 20 17:16 /dev/uinput

closed 2020-04-20 by avdz

musse, 2020-04-25:
I had the same issue and changing the MODE to `0666` on the `lib/udev/rules.d/56-steamlink.rules` also did the trick. I'm using xow and steamlink on a Raspberry Pi 4.

Relevant thread in the Steam Link RPi forums: https://steamcommunity.com/app/353380/discussions/6/2260187517782948889/