medusalix/xow #55: cow immediately crashes after connecting the headset to the gamepad

NTMan, 2020-04-19:
I decided to check how works xow with audio.
And builded xow from the audio branch (fe2b41ae632120c6a334520c5ffd79d531c35240).
Testing ended immediately after connecting the headset to the gamepad as xow immediately crashed.

Here is log + backtrace [xow-backtrace.txt](https://github.com/medusalix/xow/files/4500284/xow-backtrace.txt)

medusalix, 2020-04-20:
Should be fixed in 070fbb7889cf7abe6f24b0fd4cd0e2f0aae1c4cb.

NTMan, 2020-04-20:
Unfortunately cow continue crashed even on 070fbb7889cf7abe6f24b0fd4cd0e2f0aae1c4cb
[xow-backtrace2.txt](https://github.com/medusalix/xow/files/4503986/xow-backtrace2.txt)

medusalix, 2020-04-20:
That's related to PulseAudio refusing xow's connection. Could you check if PulseAudio is running in system-mode (with `ps ax | grep pulse`)?

NTMan, 2020-04-20:
> That's related to PulseAudio refusing xow's connection. Could you check if PulseAudio is running in system-mode (with `ps ax | grep pulse`)?
```
$ ps ax | grep pulse
  99208 ?        S<sl  47:57 /usr/bin/pulseaudio --daemonize=no
 481866 pts/16   S+     0:00 grep --color=auto pulse
```

medusalix, 2020-04-20:
> 99208 ?        S<sl  47:57 /usr/bin/pulseaudio --daemonize=no

It looks like PulseAudio is running with user privileges which should be fine. Do commands like `pactl info` work? If not, you might have to add your user to the `audio` group. I'm not sure how PulseAudio is configured on your system, so I can only guess what's going wrong. You basically have to allow applications to access the PulseAudio daemon somehow.

NTMan, 2020-04-20:
> It looks like PulseAudio is running with user privileges which should be fine. Do commands like `pactl info` work? If not, you might have to add your user to the `audio` group. I'm not sure how [...]
```
$ pactl info
Server String: /run/user/1000/pulse/native
Library Protocol Version: 33
Server Protocol Version: 33
Is Local: yes
Client Index: 288
Tile Size: 65472
User Name: mikhail
Host Name: localhost.localdomain
Server Name: pulseaudio
Server Version: 13.99.1-rebootstrapped
Default Sample Specification: s16le 2ch 44100Hz
Default Channel Map: front-left,front-right
Default Sink: alsa_output.usb-HyperX_Cloud_Orbit_S_2Ch_HyperX_Cloud_Orbit_S_2Ch_HyperX_Cloud_Orbit_S_2Ch-00.analog-stereo
Default Source: alsa_input.usb-HyperX_Cloud_Orbit_S_2Ch_HyperX_Cloud_Orbit_S_2Ch_HyperX_Cloud_Orbit_S_2Ch-00.analog-stereo
Cookie: 51f5:9365
```

medusalix, 2020-04-20:
Did you configure `xow` as a `systemd` *user* service?

NTMan, 2020-04-20:
> Did you configure `xow` as a `systemd` _user_ service (with `systemctl --user`)?

No, I launch `xow` from terminal `$ sudo ~/packaging-work/git/xow/xow`

medusalix, 2020-04-20:
That's most likely the issue :smiley:. PulseAudio doesn't allow connections from `root`, so you have to launch xow without `sudo`. Make sure that you installed the `udev` rules, so xow has access to the USB and uinput devices.

closed 2020-04-26 by medusalix