medusalix/xow #56: Random controller disconnects

nkoester, 2020-04-21:
Similarly to issue #35 I still experience random controller disconnects, however not too often. Logs of such occasions are:

```Apr 18 15:17:25 systemd[1]: Started Xbox One Wireless Dongle Driver.
Apr 18 15:17:26 xow[462]: 2020-04-18 15:17:26 INFO  - xow v0.4-10-g53f489f ©Severin v. W.
Apr 18 15:17:26 xow[462]: 2020-04-18 15:17:26 INFO  - Dongle plugged in
Apr 18 15:17:26 xow[462]: 2020-04-18 15:17:26 INFO  - Wireless address: 62:45:bd:00:c9:e3
Apr 18 15:17:26 xow[462]: 2020-04-18 15:17:26 INFO  - Dongle initialized
Apr 18 15:17:27 xow[462]: 2020-04-18 15:17:27 INFO  - Controller '1' connected
Apr 18 15:17:28 xow[462]: 2020-04-18 15:17:28 ERROR - Error stopping event loop: Bad file descriptor
Apr 18 15:17:28 xow[462]: 2020-04-18 15:17:28 INFO  - Controller '1' disconnected
Apr 18 15:18:14 xow[462]: 2020-04-18 15:18:14 INFO  - Controller '1' connected
Apr 18 15:18:14 xow[462]: 2020-04-18 15:18:14 INFO  - Product ID: 02ea
Apr 18 15:18:14 xow[462]: 2020-04-18 15:18:14 INFO  - Serial number: 02600224763946
Apr 18 15:18:30 xow[462]: 2020-04-18 15:18:30 INFO  - Controller '2' connected
```
```
Apr 20 20:57:44 systemd[1]: Started Xbox One Wireless Dongle Driver.
Apr 20 20:57:44 xow[759588]: 2020-04-20 20:57:44 INFO  - xow v0.4-10-g53f489f ©Severin v. W.
Apr 20 20:57:44 xow[759588]: 2020-04-20 20:57:44 INFO  - Dongle plugged in
Apr 20 20:57:44 xow[759588]: 2020-04-20 20:57:44 INFO  - Wireless address: 62:45:bd:00:c9:e3
Apr 20 20:57:44 xow[759588]: 2020-04-20 20:57:44 INFO  - Dongle initialized
Apr 20 20:57:46 xow[759588]: 2020-04-20 20:57:46 INFO  - Controller '1' connected
Apr 20 20:57:46 xow[759588]: 2020-04-20 20:57:46 INFO  - Product ID: 02ea
Apr 20 20:57:47 xow[759588]: 2020-04-20 20:57:47 INFO  - Serial number: 02600224753946
Apr 20 21:39:53 xow[759588]: 2020-04-20 21:39:53 INFO  - Controller '2' connected
Apr 20 21:39:53 xow[759588]: 2020-04-20 21:39:53 INFO  - Product ID: 02ea
Apr 20 21:39:54 xow[759588]: 2020-04-20 21:39:54 INFO  - Serial number: 02600224763946
Apr 20 21:46:32 xow[759588]: 2020-04-20 21:46:32 INFO  - Controller '2' disconnected
Apr 20 21:46:36 xow[759588]: 2020-04-20 21:46:36 INFO  - Controller '2' connected
Apr 20 21:46:37 xow[759588]: 2020-04-20 21:46:37 INFO  - Product ID: 02ea
Apr 20 21:46:37 xow[759588]: 2020-04-20 21:46:37 INFO  - Serial number: 02600224763946
Apr 20 23:06:00 xow[759588]: 2020-04-20 23:06:00 INFO  - Controller '2' disconnected
Apr 20 23:06:05 xow[759588]: 2020-04-20 23:06:05 INFO  - Controller '2' connected
Apr 20 23:06:06 xow[759588]: 2020-04-20 23:06:06 INFO  - Product ID: 02ea
Apr 20 23:06:07 xow[759588]: 2020-04-20 23:06:07 INFO  - Serial number: 02600224763946
Apr 20 23:49:36 xow[759588]: 2020-04-20 23:49:36 INFO  - Controller '2' disconnected
Apr 20 23:49:41 xow[759588]: 2020-04-20 23:49:41 INFO  - Controller '2' connected
Apr 20 23:49:41 xow[759588]: 2020-04-20 23:49:41 INFO  - Product ID: 02ea
Apr 20 23:49:42 xow[759588]: 2020-04-20 23:49:42 INFO  - Serial number: 02600224763946
```

Using arch and installed the xow-git version. xow runs as a system service as suggested in the package in the [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=xow-git) Am I doing anything wrong here?

I now build the debug version to gather further logs - is there any other data I can collect to help debug this issue?

mooock, 2020-04-23:
I also see a lot of disconnects and connects, and sometimes it ends with no controller are able to connect without a reboot of the PI4, maybe some services could be restart to make it work again, but for me it seems like the issue is only when having two controllers connected and playing at the same time. i couldn't reproduce it with both controller turned on and only using player 1

medusalix, 2020-04-24:
@nkoester I'm not sure if I'll be able to help you, as these edge cases (and especially connectivity issues) are always hard for me to diagnose. The debug logs will be a good start, but a full USB packet capture might provide additional insights.

mooock, 2020-04-26:
yesterday i tried playing for 4 hours, and when only using one controller there is  no issue. I started playing with two controllers, and after 10 min the first controller disconnected/reconnected, after that it worked fine for an hour. I also tried two different dongles. i will next try out some different controllers.

also i might look into doing a debug version, but have no experience how to create such

mooock, 2020-04-26:
I have tested with the newest build. And could see it had a newer version then the previous. It seems good now. For many hours no disconnect. Will keep you posted

medusalix, 2020-04-26:
> I have tested with the newest build. And could see it had a newer version then the previous. It seems good now. For many hours no disconnect. Will keep you posted

Awesome! I made some changes to the wireless channels in the latest build to make it behave like the original Microsoft driver.

medusalix, 2020-05-08:
@mooock Any news?

mooock, 2020-05-08:
i have not had any disconnects after last time, it seems the thing you did with wifi channels fixed the problem.

medusalix, 2020-05-15:
@mooock I've just pushed a change that should read the power values for the wireless channels from the dongle's internal memory (they are actually different for every dongle out there). I'd like to make sure that everything still works correctly without any disconnects, so I'd really appreciate if you could give this another try :smiley:.

nkoester, 2020-05-17:
I also experimented with the latest version and can happily report that all scenarios I tested (1 - 4 controllers on a single stick) work flawlessly. No disconnects, no problems. Thanks so much!

PS. in some tests the controllers were simply out of battery, which I notice only after way too long. Maybe some form of indication could be implemented showing that the battery is empty? A LED flashing pattern every 5 minutes or so?

mooock, 2020-05-17:
I am happy that you had the time to look into this issue, most other forums has not been as helpful as you, and mostly you only see people ask questions then help out or do their own findings for troubleshooting.

I will say this version is on par with the windows experience.

medusalix, 2020-05-18:
Thanks for the kind words, I'm very glad that it's working fine :smiley:.

> Maybe some form of indication could be implemented showing that the battery is empty?

Yeah, I really want to implement a standardized solution for battery reporting rather than LED patterns, but the progress on that issue has unfortunately come to a halt.

closed 2020-05-18 by medusalix

mooock, 2020-05-18:
For me i have never cared about battery level, i have a good idea of when to change battery, and if they die on me, in the middle of action. i get a quick workout of running after new batteries.

nkoester, 2020-05-19:
Unfortunately, my controllers do not seem to run out of battery at once but rather (for several days) miss inputs or do not hold a button when pressed continuously before finally turning off completely. I feel like missing inputs is really bad and any feedback to the users could avoid the issue of "did i miss to press the button? is the controller broken?"

I saw your efforts for a standardized solution and also the upstream problems.

One argument for an additional (infrequent) LED pattern would be a system setup not using the standard tools. A minimal setup without centralized power management would really benefit from any notification beyond OSD hints. I would not mind enabling this feature via a command line argument  ;)

medusalix, 2020-05-20:
I'm currently trying to make xow behave as similar to the Windows driver as possible, while keeping the amount of configuration options low. Ideally, I'd like to expose the LEDs, the battery status and various other stuff via a D-Bus interface. That would allow you to write simple Python scripts to control xow's behavior.