medusalix/xow #57: Range issues between controller and dongle

thewoza, 2020-04-23:
OS - Elementary OS
Controller - 1708
Dongle - 1790

Hi. Moving the controller more than about 1.5m away from the dongle results in disconnects with the controller reconnecting almost immediately when moved back into range. This doesn't happen on Windows so I'm assuming the hardware is good and I've tried various different 2.4GHz and 5GHz channels (although I'm not sure how to check if the channel has actually changed) with no change in behaviour.

If you need any more info just let me know.
Thanks

```
INFO  - xow v0.4-10-g53f489f ©Severin v. W.
DEBUG - Opening device...
INFO  - Dongle plugged in
DEBUG - Firmware loaded
DEBUG - ASIC version: 7632
DEBUG - MAC version: 7662
DEBUG - Chip id: 7613
INFO  - Wireless address: 62:45:b5:10:e4:fd
INFO  - Dongle initialized
DEBUG - Client associating: 7e:ed:82:1d:40:34
INFO  - Controller '1' connected
INFO  - Product ID: 02ea
DEBUG - Firmware version: 4.8.1923.0
DEBUG - Hardware version: 1284.1.1.1
INFO  - Serial number: 02600167125929
DEBUG - Battery type: 1, level: 2
...
DEBUG - Client lost: 1
INFO  - Controller '1' disconnected
DEBUG - Client associating: 7e:ed:82:1d:40:34
INFO  - Controller '1' connected
INFO  - Product ID: 02ea
DEBUG - Firmware version: 4.8.1923.0
DEBUG - Hardware version: 1284.1.1.1
INFO  - Serial number: xxx
DEBUG - Battery type: 1, level: 2
```

medusalix, 2020-04-24:
> I've tried various different 2.4GHz and 5GHz channels (although I'm not sure how to check if the channel has actually changed)

Well, I'm not even sure if that setting has any consequences anymore since I've introduced channel switching (76fac9f7fd052c33752aa91bf8d5a3ed0bcf8904, don't ask me how that actually works, I don't even know). I can only assure you that I'm getting **5+ meters** of wireless range with xow *through a wall* into the next room, although there are certain spots where the signal drops for xow but not for the original Windows driver. Nevertheless, I'm still looking into ways to increase the wireless range.

thewoza, 2020-04-27:
I've tried the latest build and it seems to have helped - a little bit, sometimes! Range seems to have improved slightly, sometimes. other times it behaves as before.

Not very helpful I know but thats all I know. I think it's a range / signal strength issue rather than random disconnects because it will disconnect if i move it behind my laptop where it would work OK without the laptop screen between the dongle and controller line of sight. This occurs on the limit of it's range (about 1.5 -2m). leaning back in my sofa can sometimes cause it to disconnect too, reconnecting when I lean forward again.

Thanks for your continued work on this project too!

medusalix, 2020-05-30:
Can you do some A/B tests with Linux (latest version of xow) and Windows where you use the *exact* same setup (USB port, dongle position, etc...) to compare the range of both systems?

thewoza, 2020-05-31:
Sure, I'll get back to you when I've had a chance but I'd say the range under Linux has definitely improved.

Although now I'm experiencing random dropouts - this may have also been happening at the same time as the range issues. The controller sometimes just drops connection when in same place as it's been working fine for 5 to 10 minutes.

Thanks