medusalix/xow #58: Boot of PI4 sometimes does not connect controllers

mooock, 2020-04-24:
I run on a fully updated retro bliss image from Vman. and have installed XOW, all other controller drivers has been uninstalled and have a xbox elite and xbox one s controller connected. sometimes after a full reboot with power off, the controller won't connect, then i can reboot once again and suddenly it works fine.

medusalix, 2020-05-11:
Hi, sorry for my late reply, I was pretty busy with all the other issues that suddenly appeared. Can you post the logs of xow's service after one of the occasions where they don't connect? Please make sure you're running the latest version too.

mooock, 2020-05-11:
After the latest update with the wifi channel fix, i have not seen issues when booting as well. everything has really worked great out

medusalix, 2020-05-11:
That's great! I'm trying my best to make xow as reliable as possible :smiley:.

closed 2020-05-11 by medusalix

mooock, 2020-05-11:
I have also tested it with a new and old dongle, and elite and non elite