medusalix/xow #59: Send pair command dirrect to wireless dongle

refeved, 2020-04-26:
I'm wondering is there a way to send pairing command to USB dongle? So that we don't have to press the pair button on dongle manually. I know this is kind of silly, but I need this feature for specific project. Can you please provide some guidance, and I will try to make it work.

medusalix, 2020-04-26:
Sure, great idea. The pairing mode is actually very simple: The dongle's wireless beacon frame contains a single byte that has to be set to `1` to enable the pairing. All you need to do is call `Mt76::writeBeacon(bool pairing)` with `pairing` set to `true`. Alternatively, you could hardcode the value of the `pairing` parameter to `true` in the `writeBeacon` to keep the dongle in pairing mode at all times.

andreashuetter, 2020-04-26:
That's truly a great idea! With one of my dongles I have the problem that the pairing button doesn't seem to work. So whenever I need to pair a controller to that dongle I need to do the following workaround to be able to actually perform the pairing:
I need to to plug the dongle into the USB port of a Windows machine and also connect the controller via a USB cable to that Windows machine. Thus the controller can get paired to this dongle without needing to press the pairing button on the dongle, the driver on Windows seems to take care about this. After that I can disconnect both devices from the Windows machine and can use it wirelessly on my Linux machine (the pairing persists).

I believe that the reason might be an actual hardware defect on the dongle's pairing button. However, having the opportunity to achieve this without needing access to a Windows machine would be great!

ddragic, 2020-04-26:
> I believe that the reason might be an actual hardware defect on the dongle's pairing button. However, having the opportunity to achieve pairing without needing access to a Windows machine would be great!

/slightly offtopic
Yea, the button probably died. Last year I repaired a dongle with the same defect, but it's a pain to open, glued shut, the only way in I could find was to cut it open along one side.

This is the replacement button part number if anybody needs it :grin: :
Panasonic EVPAVAA1A - https://www3.panasonic.biz/ac/e/search_num/index.jsp?c=detail&part_no=EVPAVAA1A

refeved, 2020-04-28:
> 
> 
> Sure, great idea. The pairing mode is actually very simple: The dongle's wireless beacon frame contains a single byte that has to be set to `1` to enable the pairing. All you need to do is call `Mt76::writeBeacon(bool pairing)` with `pairing` set to `true`. Alternatively, you could hardcode the value of the `pairing` parameter to `true` in `writeBeacon` to keep the dongle in pairing mode at all times.

Thanks very much for the quick reply. I managed to get it work by starting pairing in the end of Mt76::afterOpen(). Pairing will start automatically after dongle plugin. I did some tests using my dongle(model 1790) and controller(model 1708), it works as expected.

bool Mt76::afterOpen()
{
    if (!initChip())
    {
        Log::error("Failed to initialize chip");

        return false;
    }

    std::thread(&Mt76::readBulkPackets, this, MT_EP_READ).detach();
    std::thread(&Mt76::readBulkPackets, this, MT_EP_READ_PACKET).detach();

    // Start sending the 'pairing' beacon
    if (!writeBeacon(true))
    {
        Log::error("Failed to write pairing beacon");

        return false;
    }

    if (!setLedMode(MT_LED_BLINK))
    {
        Log::error("Failed to set LED mode");

        return false;
    }

    Log::info("Pairing initiated");

    return true;
}

Dragon20C, 2020-04-29:
I actually cant get my dongle to pair, the button doesnt do anything, And I know it works because I tested it on windows so I actually would love a command option to enter pair mode!

medusalix, 2020-04-30:
I think a handler for the [`SIGUSR1`](https://en.wikipedia.org/wiki/Signal_(IPC)#SIGUSR1) signal would be the easiest way to implement this. We could then connect that to the activation of the pairing.

medusalix, 2020-05-02:
Enabling the pairing mode via `SIGUSR1` is now implemented in the `signal` branch. You can use *any* of the following commands to trigger the pairing:

```
sudo kill -SIGUSR1 `pidof xow`
sudo pkill -SIGUSR1 xow
sudo systemctl kill -s SIGUSR1 xow
```

Please note that the signal handler is only installed **after** you plug in the dongle. The default behavior for `SIGUSR1` (if not handled) is to terminate the process.

andreashuetter, 2020-05-02:
Brilliant!

I just tried the `signal` branch and typing `sudo systemctl kill -s SIGUSR1 xow` turned the dongle into pairing mode, just like expected! Paired one controller with my dongle that has a defect pairing button, and it worked! 

Great, thanks!

medusalix, 2020-05-03:
@Dragon20C If you want you can do a Wireshark USB packet capture (where you press the dongle's button) and send that to me so I can figure out why the button doesn't work in your case.

Dragon20C, 2020-05-04:
@medusalix Hello! I have got wireshark setup on linux, could you please send me a quick guide on how to capture the usb dongle or walk me through it please!

Dragon20C, 2020-05-04:
@medusalix I also tried the xow signal branch but receive this error

`  sudo ./xow
[sudo] password for carlos: 
2020-05-04 10:29:13 INFO  - xow  ©Severin v. W.
2020-05-04 10:29:13 INFO  - Waiting for device...
2020-05-04 10:29:13 DEBUG - Opening device...
2020-05-04 10:29:14 DEBUG - Firmware already loaded, resetting...
2020-05-04 10:29:14 DEBUG - Firmware loaded
2020-05-04 10:29:14 ERROR - Error in control transfer: LIBUSB_ERROR_NO_DEVICE
2020-05-04 10:29:14 DEBUG - ASIC version: 7632
2020-05-04 10:29:14 DEBUG - MAC version: 7662
2020-05-04 10:29:14 DEBUG - Chip id: 7612
2020-05-04 10:29:14 INFO  - Wireless address: 62:45:b4:e7:a4:ef
2020-05-04 10:29:14 ERROR - Error in bulk write: LIBUSB_ERROR_IO
2020-05-04 10:29:14 ERROR - Failed to write command
2020-05-04 10:29:14 ERROR - Failed to init gain
terminate called after throwing an instance of 'Mt76Exception'
  what():  Failed to init gain
Aborted
`

medusalix, 2020-05-04:
@Dragon20C The `LIBUSB_ERROR_NO_DEVICE` error *could* indicate a hardware issue (e.g. unstable Linux drivers for your USB controller). Please run `dmesg` and upload the output. Did you try running xow on some other Linux computer (with a different hardware configuration)?

Dragon20C, 2020-05-04:
@medusalix the funny thing is, if I install xow from source not the signal branch it doesn't error out, I am not at the computer atm will do as you say when I can!

Dragon20C, 2020-05-04:
Hello again! @medusalix I have the output here.
[Output for dmesg.txt](https://github.com/medusalix/xow/files/4575855/Output.for.dmesg.txt)

medusalix, 2020-05-04:
@Dragon20C Are you running multiple instances of xow? Make sure that the systemd service is stopped before executing the binary from the `signal` branch.

Dragon20C, 2020-05-06:
@medusalix I have made sure I have uninstalled the xow source version (Looks like I forgot to run the last command to delete the files) But now I am receiving a different error!

Here it is:
`2020-05-06 09:09:39 INFO  - xow  ©Severin v. W.
2020-05-06 09:09:39 INFO  - Waiting for device...
2020-05-06 09:09:39 ERROR - Error in control transfer: LIBUSB_ERROR_NO_DEVICE
2020-05-06 09:09:40 ERROR - Error in bulk write: LIBUSB_ERROR_IO
2020-05-06 09:09:40 ERROR - Failed to write command
2020-05-06 09:09:40 ERROR - Failed to burst write register
2020-05-06 09:09:40 ERROR - Failed to write MAC address
terminate called after throwing an instance of 'Mt76Exception'
  what():  Failed to init registers
Aborted
`

medusalix, 2020-05-06:
@Dragon20C I've merged the `signal` branch into `master` now. Please open a new issue regarding the error you are getting (it's apparently not really related to this issue).

closed 2020-05-06 by medusalix