medusalix/xow #60: Segmentation fault after a couple of minutes

benjamimgois, 2020-04-28:
I'm using the latest xow-git and i'm having a segmentation error since the last update.

> 2020-04-27 20:52:47 INFO  - xow v0.4-11-gbea16b4 ©Severin v. W.
> 2020-04-27 20:52:47 INFO  - Dongle plugged in
> 2020-04-27 20:52:47 INFO  - Wireless address: 62:45:b4:ed:7a:0c
> 2020-04-27 20:52:47 INFO  - Dongle initialized
> 2020-04-27 20:52:49 INFO  - Controller '1' connected
> 2020-04-27 20:52:49 INFO  - Product ID: 02ea
> 2020-04-27 20:52:49 INFO  - Serial number: 09600692813926
> 2020-04-27 20:53:35 INFO  - Pairing initiated
> 2020-04-27 20:54:43 INFO  - Pairing initiated
> 2020-04-27 20:56:33 INFO  - Pairing initiated
> 2020-04-27 20:59:03 INFO  - Pairing initiated
> 2020-04-27 21:00:34 INFO  - Pairing initiated
> 2020-04-27 21:01:43 INFO  - Pairing initiated
> 2020-04-27 21:02:01 INFO  - Pairing initiated
> 2020-04-27 21:02:36 INFO  - Pairing initiated
> 2020-04-27 21:03:34 INFO  - Pairing initiated
> 2020-04-27 21:04:12 INFO  - Pairing initiated
> 2020-04-27 21:04:34 INFO  - Pairing initiated
> 2020-04-27 21:04:36 INFO  - Pairing initiated
> Falha de segmentação (imagem do núcleo gravada)
>

medusalix, 2020-04-29:
Please check if the error is present in the [`usb`](https://github.com/medusalix/xow/tree/usb) as well. If it is, then please run xow with `valgrind` or `gdb` to get a full stacktrace.

benjamimgois, 2020-04-30:
I got the same problem with the "usb" branch. Another thing i observed is that it keeps printing the controller serial number on the console the whole time it's connected. Here is log i got with `xow gdb`

https://pastebin.com/ke6MiuGY

medusalix, 2020-05-01:
Sorry for being a bit unclear in my previous answer. What I actually meant was to run xow with the [`valgrind`](https://valgrind.org/) tool (e.g. `valgrind ./xow`). This should tell me exactly where the segmentation fault happens. Please also do a packet capture with Wireshark (on Linux). You can follow [this guide](https://wiki.wireshark.org/CaptureSetup/USB#Linux) for further instructions on that. That will allow me to find out why xow is constantly printing `Pairing initiated`.

benjamimgois, 2020-05-03:
Hey @medusalix , sorry for my late. Here is the output of xow-git with valgrind, hope it helps. And just to remember, the latest stable release works fine.

https://pastebin.com/G0x3qMmy

I'll try to make a wireshark capture tomorrow

medusalix, 2020-05-03:
> Here is the output of xow-git with valgrind, hope it helps.

Please run the *debug* build of xow with valgrind (`make BUILD=DEBUG`). The release version doesn't print any code references, so I can't see where it is failing.

benjamimgois, 2020-05-05:
@medusalix i think now i got it right. Take a look

https://pastebin.com/xPWWNyWM

medusalix, 2020-05-05:
Thank you, that was very helpful. I've pushed a fix to master. I'm a little bit concerned about those `Pairing initiated` messages in your initial post. Did you actually press the pairing button or is that another bug :sweat_smile:?

benjamimgois, 2020-05-05:
@medusalix i recentely wiped my old manjaro installation and made a new one. After that the paring messages disappeared. I'll try the new build asap.

benjamimgois, 2020-05-06:
It's a lot better now, but i still got segmentation fault after some time. Here is another log using the latest build.

https://pastebin.com/te2Fj44x

medusalix, 2020-05-06:
Looks like another out-of-bounds case for the packets. I added a bunch of additional checks to the packet handling logic, please check if these fix your issue.

benjamimgois, 2020-05-07:
@medusalix I made several testes today and it seems that the problem is solved. Thanks so much for the help ! Your work is awesome !

closed 2020-05-07 by benjamimgois