medusalix/xow #62: what():  Error opening device: Permission denied

Thib1404, 2020-05-01:
I am using a fresh install of Raspbian GNU/Linux 10 (buster) and a Xbox One controller 1537 that I am planning to use with Steam link. I have a large wireless dongle. It works fine on Windows 10. I have installed xov successfully and the service is running. Yet, when I try to pair the controller I have the following error displayed with "sudo systemctl status xow" :

● xow.service - Xbox One Wireless Dongle Driver
   Loaded: loaded (/lib/systemd/system/xow.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Fri 2020-05-01 15:45:08 CEST; 757ms ago
  Process: 1840 ExecStart=/usr/local/bin/xow (code=killed, signal=ABRT)
 Main PID: 1840 (code=killed, signal=ABRT)

mai 01 15:44:56 raspberrypi xow[1840]: 2020-05-01 15:44:56 INFO  - xow v0.4-16-ge54c8fd ©Severin v. W.
mai 01 15:44:56 raspberrypi xow[1840]: 2020-05-01 15:44:56 INFO  - Waiting for device...
mai 01 15:44:56 raspberrypi xow[1840]: 2020-05-01 15:44:56 INFO  - Wireless address: 62:45:b4:ec:e0:00
mai 01 15:45:04 raspberrypi xow[1840]: 2020-05-01 15:45:04 INFO  - Pairing initiated
mai 01 15:45:04 raspberrypi xow[1840]: 2020-05-01 15:45:04 INFO  - Pairing initiated
mai 01 15:45:04 raspberrypi xow[1840]: 2020-05-01 15:45:04 INFO  - Pairing initiated
mai 01 15:45:08 raspberrypi xow[1840]: terminate called after throwing an instance of 'InputException'
mai 01 15:45:08 raspberrypi xow[1840]:   what():  Error opening device: Permission denied
mai 01 15:45:08 raspberrypi systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
mai 01 15:45:08 raspberrypi systemd[1]: xow.service: Failed with result 'signal'.

medusalix, 2020-05-01:
We've had this issue a couple of times already: Most of the time users didn't reboot after installation or had conflicts with existing `udev` rules. Please take a look at issues #54 and #23.

FrandSX, 2020-05-01:
The conflict with Steam Link udev rules is also noted on the [Steam Link Discussions](https://steamcommunity.com/app/353380/discussions/6/2260187517782948889/)

The solution in #54 fixed the same problem for me.

Thib1404, 2020-05-01:
Indeed, thank you for your quick support ! I haven't looked deep enough in the past issues.. The solution #54 worked for me !
In this file /lib/udev/rules.d/56-steamlink.rules

```
# USB devices
SUBSYSTEM=="usb", GROUP="plugdev"

# HID devices
KERNEL=="hidraw*", GROUP="input", MODE:="0660"

# Creating virtual devices
KERNEL=="uinput", GROUP="input", MODE:="0660"
```

I changed the MODE to 0666 and it works as expected !

closed 2020-05-01 by Thib1404