medusalix/xow #63: Add $XOW_NAME  or some other ENV to control stringname of controller

x4mers, 2020-05-06:


andreashuetter, 2020-05-07:
I think that's a great idea!
So do you think the default string should be with a hyphen, or without one?
Which string is reported by the driver on Windows, I guess xow should report the same by default?
But being able to override the default by setting an environment variable sounds like a good idea.

medusalix, 2020-05-07:
> Some badly programmed games do string searches to determine if a supported controller is connected.

Oh, I remember that from when I added dongle support to [360Controller](https://github.com/360Controller/360Controller) (macOS driver). The controller worked fine, but I literally couldn't find any game that accepted inputs from it because they **all** performed string comparison with the product name.

> I know you said game controls fixes were beyond the scope of your project, but by allowing a controller name env variable. it would allow people to apply their own fixes without having to compile custom builds.

Yeah, I'm mostly trying to avoid issue like "dpad-up is mapped to dpad-down in game XY". I don't want to introduce fixes for specific games, but I think introducing an environment variable that helps to make xow compatible with a bunch of games is a perfectly valid suggestion :+1:.

> Which string is reported by the driver on Windows, I guess xow should report the same by default?

The Windows driver names the controller "Xbox One Wireless Controller", just like xow. I wonder why projects like [xpad-neo](https://github.com/atar-axis/xpadneo) never ran into this problem though (I couldn't find any issue that mentioned this, maybe you should create one @x4mers).

x4mers, 2020-05-07:
> Yeah, I'm mostly trying to avoid issue like "dpad-up is mapped to dpad-down in game XY". I don't want to introduce fixes for specific games, but I think introducing an environment variable that helps to make xow compatible with a bunch of games is a perfectly valid suggestion .

 That's the kind of issue Flamebreak was having until I changed the name. It would accept the controller, but the axes/buttons were wrong with some buttons sending the same signal as another, and the split axes triggers made it impossible to remap in game. Nice to hear you're open to it.

> The Windows driver names the controller "Xbox One Wireless Controller", just like xow. The Linux driver for wired controllers calls it "Microsoft X-Box One pad". I wonder why projects like [xpad-neo](https://github.com/atar-axis/xpadneo) never ran into this problem though (I couldn't find any issue that mentioned this, maybe you should create one @x4mers).

 I keep games with problems set in special categories (like broken controls on Linux) in Steam. Xpad neo does have the problem. I hadn't played Savant Ascent since I bought it, since xpad via BT and xpadneo both saw no controller connected. So many people raved about how xpad neo was better than the kernel driver, but I had problems with it every time I installed it. It would usually be OK for short (30 mins to an hour) gaming sessions, but it would be rare to make it through a long session without the controller bugging out with rumble stuck on in the middle of a turn in a driving game, or dropping connection and refusing to reconnect without a complete power down to reset the BT receiver.

medusalix, 2020-05-07:
> I ended up going with "Microsoft X-Box 360 pad" for the modified string since I figured that pad had been around a long time and had good support on Steam.

Do you think a simple environment flag to turn "Xbox One Wireless Controller" into "Microsoft X-Box 360 pad" would be enough to get xow working with all games that do string comparison (like a *360 compatibility mode*)?

x4mers, 2020-05-07:
> Do you think a simple environment flag to turn "Xbox One Wireless Controller" into "Microsoft X-Box 360 pad" would be enough to get xow working with all games that do string comparison (like a _360 compatibility mode_)?

It would seem so at this point, from the games I've tested so far. With the support that pad has on Steam, I think it makes a perfect candidate for a compatibility mode.

medusalix, 2020-05-11:
I've merged the `compatibility` branch into `master`. Huge thanks to @x4mers for all the testing work!

closed 2020-05-11 by medusalix