medusalix/xow #64: xow doesn't work with some native games

benjamimgois, 2020-05-07:
Hey @medusalix, firstly thanks for the amazing project ! 

i don't know if this is a game bug or a xow bug, but i noticed 2 native linux games (**Brutal Legend** and **Borderlands 2**) with controller support where the joystick isn't recognized by the game while using "xow" and a wireless dongle, but works perfectly fine on USB. Another observation is that i can use the controller with xow if i run the proton version of the same game.

andreashuetter, 2020-05-07:
Could this be the same issue described here: https://github.com/medusalix/xow/issues/63

benjamimgois, 2020-05-07:
Sure, it's the same issue. Should i close this one ?

x4mers, 2020-05-07:
> Sure, it's the same issue. Should i close this one ?

I was going to install them to try with my build, but does this mean you already compiled with changed string for the controller and they now work?

benjamimgois, 2020-05-07:
no, i didn't... i though the issue could be managed in #63

x4mers, 2020-05-07:
> no, i didn't... i though the issue could be managed in #63

I'll check them out. It may be unrelated to #63. I have two other games found so far, that don't recognize a controller and are not fixed by #63. The 2 games are Geometry Wars 3, and Spec Ops the Line. They both wouldn't work regardless of what I set the string to. They both use SDL for input though and are fixed by using an ENV variable to tell the game the specifics of your controller, a list of which can be found here https://github.com/gabomdq/SDL_GameControllerDB/blob/bb589c6b8e3be7f7041d8ca292082d8c8dd2c36b/gamecontrollerdb.txt. Spec Ops stores it a folder in the game directory, while GW3 needs it sent as a launch option in Steam.

benjamimgois, 2020-05-07:
Ok, i'll check if i find this problem on other games.

x4mers, 2020-05-07:
Both Brutal Legends, and Borderlands 2 were ported to Linux by Aspyr, the same company that ported the GW3 game I mentioned above. They must be using the same code framework for all(?) their ports, and it does not involve string matching, so will not be fixable with #63.

 They are all using SDL mapping entries to add controller support. I guess they don't release patches to add in newer controller GUIDs to the game, and since X-Box One wireless controllers weren't working on Linux when those games came out.....

 To get them working with the current xow builds, right click the game in your library, select properties and then launch options. Paste the following all on one line

SDL_GAMECONTROLLERCONFIG="0300000058626f78204f6e6520576900,Xbox One Wireless Controller,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux," %command%

That gets both games working with controller here, using standard xow build. My controllers are 1708 One S versions of the pad, so my GUID may be different than yours if you're using non 1708 versions. If that's the case, then reference the github link I posted earlier, and check the Linux section for different GUIDs that are XBox One controllers to try to find your match. You'll only have to change the long GUID number at the beginning of the quote to find the one that matches your pad.

benjamimgois, 2020-05-08:
@x4mers i still couldn't test your solution, but will try it asap.

medusalix, 2020-05-08:
@x4mers I just found [this](https://github.com/atar-axis/xpadneo/blob/7574862faeb60cfba03d804a70a5f1b007cc9b84/docs/SDL.md) article on xpadneo's page. Could you try adding `setup.id.version = 0xabcd;` to `InputDevice::create` instead of the SDL fix you mentioned above? That's how xpadneo fixed this issue.

x4mers, 2020-05-08:
That article deals with multiple entities (SDL,Wine,Steam client) all adding workarounds for mapping issues. When multiple of them are active at any time, they work to undo the fix of the other. The pad still works, but the mappings are all wrong.

 All four games mentioned in #64 (Geometry Wars 3, Spec Ops the Line, Borderlands 2, and Brutal Legend), do not detect that any pad is connected at all. The sections for gamepad in the menus is missing or greyed out, until the fix above is applied to add the pad to the games' list of known devices.

 Maybe I am not understanding something. Is the addition in your post above intended to make unidentified devices become identified? I thought the fix by Xpad-neo was just to spoof so that the other entities didn't apply their mapping patches, thereby removing the conflicts among them, and leave xpad neo to apply it's own mapping fix undeterred.

x4mers, 2020-05-08:
Well, I went and tried it anyway. I added your line to input.cpp, recompiled and installed. I went and removed the SDL launch option from all 4 games discussed here, and tried them all. 3 out of the 4 now work fine, without any user intervention. Spec Ops the Line is the only one that still doesn't detect any controllers connected.

Before this change, running the controllermap utility resulted in the reported GUID as being 03.....VID....PID.... with all the spaces being all zeros. Trying to use this GUID in an SDL map like I did in the fix above would fail. Luckily Spec Ops the line logs it's own startup every time, and I could see in there that it was saying "new controller detected - calculated GUID of ......" if I used that GUID for the SDL command it worked, not only for that game but all the others as well. All the games were seeing that GUID as something different than what controllermap saw it as. Another problem was that the GUID seen by all the games would change if the name of the pad was changed, so the SDL config line wouldn't work anymore unless the GUID in it was ammended.

After your patch, the GUID now has cdab after the PID, instead of all zeros, according to controllermap. For the first time, the logs for Spec Ops the Line show that it sees the GUID as the same as what controllermap sees. On top of that, the GUID does not change when the controller name is changed, so the SDL config lines for games like Spec Ops stay valid regardless of controller name.

benjamimgois, 2020-05-08:
@x4mers i just found that "Dirt Rally" also have this problem, but the SDL_GAMECONTROLLERCONFIG="..." doesn't work for it.

x4mers, 2020-05-08:
> @x4mers i just found that "Dirt Rally" also have this problem, but the SDL_GAMECONTROLLERCONFIG="..." doesn't work for it.

 Just tested it here, with a new build that includes the patch mentioned above by medusalix, and it works fine without the need of any SDL command line. All 3 games you've mentioned in this thread work fine without my SDL instructions once the patch mentioned above is pushed to the repo and you build an up to date release.

medusalix, 2020-05-09:
@x4mers I've pushed the changes to the `compatibility` branch. There's now a `XOW_COMPATIBILITY` environment variable that you can set to any value to activate the compatibility changes.

Another idea that came to my mind is that we could try setting the PID and VID to the actual values of a Xbox 360 controller (0x045e, 0x028e). Could that fix the problem with Spec Ops?

x4mers, 2020-05-09:
> @x4mers I've pushed the changes to the `compatibility` branch. There's now a >`XOW_COMPATIBILITY` environment variable that you can set to any value to activate the >compatibility changes.

 I take it this ENV is checked at the time the service is started, so a reboot or systemctl stop/start xow is required if it's been changed?

> Another idea that came to my mind is that we could try setting the PID and VID to the actual >values of an Xbox 360 controller (`0x045e`, `0x028e`). Could that fix the problem with Spec Ops?

In Spec Ops logs, it shows that it goes thru a long list of GUID that are built into the game code. It then continues with any that are in a db file that is in the game directory. It seems the game's controller code is solely dependent on SDL map entries. Even if VID/PID were spoofed to that of a 360 controller, the rest of the GUID wouldn't match, especially with the 0xabcd being added. I'll check the logs again for those VID/PID, and maybe try changing the code though.

medusalix, 2020-05-09:
> I take it this ENV is checked at the time the service is started, so a reboot or systemctl stop/start xow is required if it's been changed?

Yes exactly. That's simply the way environment variables work, they are passed to the process on execution. Adding `Environment="XOW_COMPATIBILITY=1"` to the `[Service]` section in the systemd unit file should work (I'll document this in the README).

> In Spec Ops logs, it shows that it goes thru a long list of GUID that are built into the game code. It then continues with any that are in a db file that is in the game directory.

Ah, got it. So there's basically nothing we can do about this from xow's perspective. (BTW, I changed 0xabcd to 0xdead)

x4mers, 2020-05-09:
@medusalix  I've set up the compatibility branch build, but it seems to be ignoring the ENV. I added XOW_COMPATIBILITY=1 to /etc/environment to make it global, and rebooted. Echo $XOW_COMPATIBILITY returns a value of 1 whether logged in to my user account or with sudo su. Jstest and controllermap show that the controller name and GUID have not changed.  I tried sysytemctl stop/start xow as well, and no change. Since it's global, shouldn't it work, or it has to be in the systemd section you mentioned?

 Also, I see that the code says if COMPAT then name=compatname and id.version=spoof else name=stdname. Since it provides the last section of the GUID, shouldn't id.version have a value from somewhere when it's not spoofing, instead of being all zeros? Is it supposed to come from the devices firmware or something?

When my One S pad is connected via USB the last 8 digits of the GUID are 08040000, instead of all zeros when connected via xow. Looking at the controllerdb I mentioned near the top of this thread, the GUID in the Linux section have content in the last 8 digits, except for one 8BitDo pad.

x4mers, 2020-05-09:
Just booted in to Windows, to run the X-Box accessories app, and according to it the firmware version is 4.8....., so xpad kernel driver is using that for the 08040000?

medusalix, 2020-05-10:
> Since it provides the last section of the GUID, shouldn't id.version have a value from somewhere when it's not spoofing, instead of being all zeros? Is it supposed to come from the devices firmware or something?

Yeah, makes sense. I didn't set the `version` field at all until now.

medusalix, 2020-05-10:
OK, so the value of the `version` field is now derived from the controller's major and minor firmware version if `XOW_COMPATIBILITY` is not set.

> Even if VID/PID were spoofed to that of a 360 controller, the rest of the GUID wouldn't match [...]

I added complete spoofing for the 360 controller in the latest commit. The GUIDs now look like this:

USB:						`030000005e040000ea02000008040000`
Wireless (normal):			`030000005e040000ea02000008040000`
Wireless (compatibility):   `030000005e0400008e02000004010000`

According to the ControllerDB, the compatibility GUID should match the original 360 controller GUID.

x4mers, 2020-05-10:
> Yeah, makes sense. I didn't set the `version` field at all until now.
> OK, so the value of the `version` field is now derived from the controller's major and minor >firmware version if `XOW_COMPATIBILITY` is not set.

 Now that it's set, Dirt Rally, GW3, Brutal Legend and Borderlands2 all work fine without having to enable compatible mode.

 When launching Steam from the terminal, the initialization logging now shows a version number for the pad. It still shows the serial number as -0 and complains the pad has an invalid or missing serial number, but it seems to have no ill effect, and systemctl status xow shows the serial number fine, so something odd going on with Steam I guess.

> I added complete spoofing for the 360 controller in the latest commit. The GUIDs now look like 
> Wireless (compatibility): `030000005e0400008e02000004010000`

 Compatible continues to fix Savant Ascent, Candle, and Flamebreak without breaking any of the ones mentioned earlier. Since you did a full spoof of 360 pad now, and that is an older pad that Spec Ops knows about, Spec Ops the Line now picks up the controller in compatibility mode without having to resort to any SDL GUID launch options.

 I added //Environment="XOW_COMPATIBILITY=1" to the service.in file, and it works fine in normal mode like that. Just had to uncomment and reboot to get to compatible mode.

Great work. Thanks for your time and effort.

medusalix, 2020-05-10:
> It still shows the serial number as -0 and complains the pad has an invalid or missing serial number, but it seems to have no ill effect, and systemctl status xow shows the serial number fine, so something odd going on with Steam I guess.

There's unfortunately no way for me to pass the serial number to `uinput` (really only VID, PID and version), so it's only logged to the console for now. Nothing wrong with Steam :smiley:.

> Since you did a full spoof of 360 pad now, and that is an older pad that Spec Ops knows about, Spec Ops the Line now picks up the controller in compatibility mode without having to resort to any SDL GUID launch options.

Awesome! I'm just going to add some documentation to the README and service file and call it a day!

benjamimgois, 2020-05-11:
Hey guys , just tested the "compatibility" branch and every single problem i related here were solved. You guys rock ! Thanks !

closed 2020-05-11 by benjamimgois