medusalix/xow #65: Debian package link showing error.

Neverest, 2020-05-08:
Since I am going to upgrade to 20.04 soon, I decided to check the debian package from the readme info, but it seems to not exist:

[image]

Anything I did wrong or is missing?

medusalix, 2020-05-08:
It was apparently removed because the license of the `firmware.bin` file is still unclear (see [this](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=954876) issue on Debian's mailing list). I've requested a comment from MediaTek, but they didn't bother to answer (see #15).

samueloph, 2020-05-09:
Yes, I had to ask for the removal because of the licensing issue, after that is solved it won't take much for the package to go back to the repositories, but for now we can't distribute it because it could cause issues for both Debian and upstream.

medusalix, 2020-06-19:
I've removed the link to the Debian package from the README for now. Still no response from MediaTek, I'll update issue #15 when they get back to me.

closed 2020-06-19 by medusalix