medusalix/xow #66: Non-standard service file location

viggy96, 2020-05-10:
/usr/etc/systemd/system/xow.service does not appear to be a location that systemd looks for service files. So when you try to run the service, systemd does not find the service file that the user is referring to. I believe that this issue occurs due to 3ce97b6.

x4mers, 2020-05-10:
Looking at the Makefile, the install script says xow.service should be installed to /lib/systemd/system.

That's where the script installed it to on my system.

medusalix, 2020-05-10:
I've read through `systemd`'s [documentation](https://www.freedesktop.org/software/systemd/man/systemd.unit.html#Unit%20File%20Load%20Path) and noticed that `/lib` (or `/usr/lib` for that matter) is actually reserved for *System units installed by the distribution package manager*, so I decided to change all paths to use `/etc` instead (which in my opinion makes more sense).
If I run `sudo make install` it installs the service unit to `/etc/systemd/system` as expected, so I don't know why you're getting `/usr/etc/systemd/system` instead. Did you set the `DESTDIR` variable to `/usr`?

viggy96, 2020-05-10:
@medusalix It was the PKGBUILD in the AUR that was the issue. Thanks for your help, and great work.

medusalix, 2020-05-10:
@viggy96 Ah, that explains why it worked fine for me. Sorry for the trouble caused.

closed 2020-05-10 by medusalix