medusalix/xow #68: Getting a "Error in control transfer: LIBUSB_ERROR_TIMEOUT" after pairing in Windows

Neverest, 2020-05-14:
I installed the latest version of xow and paired fine. However when I boot to Windows it doesn't pair unless I remove and insert the dongle and start pairing again. 

If I then go back to my linux hard drive I can't connect my xbox one controller. The dongle doesn't light up and removing it and inserting it does nothing. Debug logs show one `Error in control transfer:LIBUSB_ERROR_IO ` and a lot of `Error in control transfer: LIBUSB_ERROR_TIMEOUT`.
If I restart and boot to Linux again it starts fine.

log of xow if I don't pair it on my windows hard drive:

```
neverest@neverest-pc:~$ journalctl -b -u xow
-- Logs begin at Thu 2018-07-26 15:24:31 AST, end at Thu 2020-05-14 07:34:00 AST. --
May 14 07:33:33 neverest-pc systemd[1]: Started Xbox One Wireless Dongle Driver.
May 14 07:33:38 neverest-pc xow[1321]: 2020-05-14 07:33:38 INFO  - xow v0.4-38-g240d111 ©S
May 14 07:33:38 neverest-pc xow[1321]: 2020-05-14 07:33:38 INFO  - Waiting for device...
May 14 07:33:38 neverest-pc xow[1321]: 2020-05-14 07:33:38 DEBUG - Opening device...
May 14 07:33:38 neverest-pc xow[1321]: 2020-05-14 07:33:38 DEBUG - Firmware loaded
May 14 07:33:38 neverest-pc xow[1321]: 2020-05-14 07:33:38 DEBUG - ASIC version: 7632
May 14 07:33:38 neverest-pc xow[1321]: 2020-05-14 07:33:38 DEBUG - MAC version: 7662
May 14 07:33:38 neverest-pc xow[1321]: 2020-05-14 07:33:38 DEBUG - Chip id: 7612
May 14 07:33:38 neverest-pc xow[1321]: 2020-05-14 07:33:38 INFO  - Wireless address: 62:45
May 14 07:33:55 neverest-pc xow[1321]: 2020-05-14 07:33:55 INFO  - Controller '1' connecte
May 14 07:33:56 neverest-pc xow[1321]: 2020-05-14 07:33:56 INFO  - Device announced, produ
May 14 07:33:56 neverest-pc xow[1321]: 2020-05-14 07:33:56 DEBUG - Firmware version: 3.1.1
May 14 07:33:56 neverest-pc xow[1321]: 2020-05-14 07:33:56 DEBUG - Hardware version: 1284.
May 14 07:33:56 neverest-pc xow[1321]: 2020-05-14 07:33:56 INFO  - Serial number: 02640261
May 14 07:33:58 neverest-pc xow[1321]: 2020-05-14 07:33:58 DEBUG - Battery type: 1, level: 2
```

log of xow after I pair it on my windows hard drive:

```
neverest@neverest-pc:~$ journalctl -b -u xow
-- Logs begin at Thu 2018-07-26 15:24:31 AST, end at Thu 2020-05-14 07:46:11 AST. --
May 14 07:41:31 neverest-pc systemd[1]: Started Xbox One Wireless Dongle Driver.
May 14 07:41:36 neverest-pc xow[1254]: 2020-05-14 07:41:36 INFO  - xow v0.4-38-g240d111 ©Severin v. W.
May 14 07:41:36 neverest-pc xow[1254]: 2020-05-14 07:41:36 INFO  - Waiting for device...
May 14 07:41:36 neverest-pc xow[1254]: 2020-05-14 07:41:36 DEBUG - Opening device...
May 14 07:41:36 neverest-pc xow[1254]: 2020-05-14 07:41:36 DEBUG - Firmware already loaded, resetting...
May 14 07:41:37 neverest-pc xow[1254]: 2020-05-14 07:41:37 ERROR - Error in control transfer: LIBUSB_ERROR_IO
May 14 07:41:38 neverest-pc xow[1254]: 2020-05-14 07:41:38 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:39 neverest-pc xow[1254]: 2020-05-14 07:41:39 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:40 neverest-pc xow[1254]: 2020-05-14 07:41:40 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:41 neverest-pc xow[1254]: 2020-05-14 07:41:41 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:42 neverest-pc xow[1254]: 2020-05-14 07:41:42 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:43 neverest-pc xow[1254]: 2020-05-14 07:41:43 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:44 neverest-pc xow[1254]: 2020-05-14 07:41:44 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:45 neverest-pc xow[1254]: 2020-05-14 07:41:45 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:46 neverest-pc xow[1254]: 2020-05-14 07:41:46 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:47 neverest-pc xow[1254]: 2020-05-14 07:41:47 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:48 neverest-pc xow[1254]: 2020-05-14 07:41:48 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:49 neverest-pc xow[1254]: 2020-05-14 07:41:49 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:50 neverest-pc xow[1254]: 2020-05-14 07:41:50 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:51 neverest-pc xow[1254]: 2020-05-14 07:41:51 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:52 neverest-pc xow[1254]: 2020-05-14 07:41:52 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
May 14 07:41:53 neverest-pc xow[1254]: 2020-05-14 07:41:53 ERROR - Error in control transfer: LIBUSB_ERROR_TIMEOUT
```

medusalix, 2020-05-14:
I don't think there's anything I can do in this case. The problem probably lies in the dongle's firmware, as both the Windows driver and xow are malfunctioning.

Neverest, 2020-05-16:
Hmm, I think I faintly remember using previous versions and not getting this error. How can I download versions 0.40,  0.30 and maybe lower?

medusalix, 2020-05-16:
If you downloaded the project with `git clone` you can do `git checkout v0.4` to get to version 0.4 (and `v0.3` for 0.3, etc...).
I remember adding a firmware reset in case the firmware was already loaded (which works in exactly the same way as the Windows driver) to prevent issues when rebooting from Windows to Linux (see commit a35977eaada7b6cf8191e47e9f6bf5473b9ee18a).

medusalix, 2020-05-16:
I just tried out versions 0.3 and 0.4 and they both show the same issue. I fired up Wireshark and it looks like the dongle doesn't accept any commands at all after going from Windows to Linux or vice versa.

Neverest, 2020-05-16:
So I gave it a try and 0.3 connected, but that was cause the dongle and the controller kept being on even when restarting. 

With 0.4 they connected fine without me having to unplug the dongle for it to work for about 2 linux to windows restart laps. Then it started having problems connecting. 

Debug log showed as if the controller kept connecting and creating new controller entries:

```
-- Logs begin at Tue 2018-09-25 13:04:44 AST, end at Sat 2020-05-16 13:15:49 AST. --
May 16 13:12:07 neverest-pc systemd[1]: Started Xbox One Wireless Dongle Driver.
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 INFO  - xow v0.4 ©Severin v. W.
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 DEBUG - Opening device...
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 INFO  - Dongle plugged in
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 DEBUG - Firmware already loaded, resetting...
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 DEBUG - Firmware loaded
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 DEBUG - ASIC version: 7632
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 DEBUG - Chip id: 7612
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 INFO  - Wireless address: 62:45:b4:f1:17:b5
May 16 13:12:12 neverest-pc xow[1275]: 2020-05-16 13:12:12 INFO  - Dongle initialized
May 16 13:12:18 neverest-pc xow[1275]: 2020-05-16 13:12:18 DEBUG - Client associating: 7e:ed:81:0c:08:0a
May 16 13:12:18 neverest-pc xow[1275]: 2020-05-16 13:12:18 INFO  - Controller '1' connected
May 16 13:12:23 neverest-pc xow[1275]: 2020-05-16 13:12:23 DEBUG - Client associating: 7e:ed:81:0c:08:0a
May 16 13:12:23 neverest-pc xow[1275]: 2020-05-16 13:12:23 INFO  - Controller '2' connected
May 16 13:12:29 neverest-pc xow[1275]: 2020-05-16 13:12:29 DEBUG - Client associating: 7e:ed:81:0c:08:0a
May 16 13:12:29 neverest-pc xow[1275]: 2020-05-16 13:12:29 INFO  - Controller '3' connected
May 16 13:12:34 neverest-pc xow[1275]: 2020-05-16 13:12:34 DEBUG - Client associating: 7e:ed:81:0c:08:0a
May 16 13:12:34 neverest-pc xow[1275]: 2020-05-16 13:12:34 INFO  - Controller '4' connected
May 16 13:13:01 neverest-pc xow[1275]: 2020-05-16 13:13:01 DEBUG - Client associating: 7e:ed:81:0c:08:0a
May 16 13:13:01 neverest-pc xow[1275]: 2020-05-16 13:13:01 INFO  - Controller '5' connected
May 16 13:13:07 neverest-pc xow[1275]: 2020-05-16 13:13:07 DEBUG - Client associating: 7e:ed:81:0c:08:0a
May 16 13:13:07 neverest-pc xow[1275]: 2020-05-16 13:13:07 INFO  - Controller '6' connected
May 16 13:13:12 neverest-pc xow[1275]: 2020-05-16 13:13:12 DEBUG - Client associating: 7e:ed:81:0c:08:0a
```
`
a35977e` didn't connect after first windows boot

medusalix, 2020-05-16:
It's definitely related to the dongle and not the driver. See this USB capture of the Windows driver after using the dongle on Linux:

[image]

The two rather large `URB_BULK out` transfers (14408 bytes) are part of the firmware upload process. The Windows driver resets the dongle and tries to upload the firmware. If you take a look at the second column, you see that the time between those transfers (48.727 and 78.125 seconds) is about 30 seconds. Normally, the whole firmware loading process would complete in a matter of *milliseconds*, which means that something unusual happened and the driver reached its timeout. The data is sent to the dongle and the dongle just doesn't respond *at all*. This is happening on the USB level and there's unfortunately nothing I'm aware of that can be done to fix this in xow.

medusalix, 2020-05-20:
I've implemented timeouts for the firmware loading, so that should at least allow you to restart the service in case it times out. Apart from that, there's unfortunately not much I can do.

closed 2020-05-20 by medusalix