medusalix/xow #71: xow driver crashed mid-game

Tourman36, 2020-05-21:
I'm running Manjaro 5.4.40-1 release and while playing Shadow of Mordor my controller suddenly became unpaired. The only way to restore connectivity is to `systemctl stop xow`, unplug usb dongle and then plug back in dongle.

I'm using a 2019 Xbox One Elite controller.

`May 21 16:50:01 dcass-lnx-desktop systemd-coredump[13145]: Process 13138 (xow) of user 64499 dumped core.
                                                           
                                                           Stack trace of thread 13138:
                                                           #0  0x00007f20d42e5ce5 raise (libc.so.6 + 0x3bce5)
                                                           #1  0x00007f20d42cf857 abort (libc.so.6 + 0x25857)
                                                           #2  0x00007f20d46a481d _ZN9__gnu_cxx27__verbose_terminate_handlerEv (libstdc++.so.6 + 0x9681d)
                                                           #3  0x00007f20d46b12ca _ZN10__cxxabiv111__terminateEPFvvE (libstdc++.so.6 + 0xa32ca)
                                                           #4  0x00007f20d46b1337 _ZSt9terminatev (libstdc++.so.6 + 0xa3337)
                                                           #5  0x00007f20d46b159e __cxa_throw (libstdc++.so.6 + 0xa359e)
                                                           #6  0x0000564f2e9d87ef _ZN4Mt76C2ESt10unique_ptrI9UsbDeviceSt14default_deleteIS1_EE.cold (xow + 0x67ef)
                                                           #7  0x0000564f2e9e1483 _ZN6DongleC2ESt10unique_ptrI9UsbDeviceSt14default_deleteIS1_EE (xow + 0xf483)
                                                           #8  0x0000564f2e9d9acd _Z3runv (xow + 0x7acd)
                                                           #9  0x0000564f2e9d931b main (xow + 0x731b)
                                                           #10 0x00007f20d42d1023 __libc_start_main (libc.so.6 + 0x27023)
                                                           #11 0x0000564f2e9d937e _start (xow + 0x737e)
                                                           
                                                           Stack trace of thread 13142:
                                                           #0  0x00007f20d439eabf __poll (libc.so.6 + 0xf4abf)
                                                           #1  0x00007f20d449b803 n/a (libusb-1.0.so.0 + 0x11803)
                                                           #2  0x00007f20d44af46f start_thread (libpthread.so.0 + 0x946f)
                                                           #3  0x00007f20d43a93d3 __clone (libc.so.6 + 0xff3d3)
-- Subject: Process 13138 (xow) dumped core
`

medusalix, 2020-05-23:
Are you running the latest version of xow? If you are, I'm going to need a full stacktrace with line numbers: Disable the systemd service and run xow manually with `valgrind ./xow` until it crashes.

medusalix, 2020-05-28:
@Tourman36 Any news regarding this issue?

Tourman36, 2020-05-28:
I haven't had a crash since. No idea what caused it.

closed 2020-05-28 by Tourman36