medusalix/xow #72: Unit file xow.service does not exist.

dave-atron, 2020-05-22:
On the newest build when I try to start the service I get the error in the title. I noticed the service file does not have 0755 permissions by default, so even after manually adding those the service still does not start. It is created correctly in /etc/systemd/system

The above is true even after rebooting the computer.

I am on Fedora 32. I was really enjoying this project before I upgraded it so I hope you can help, just let me know what you need from me.

medusalix, 2020-05-23:
The command `systemd-analyze unit-paths` should give you a list of paths where systemd searches for service files. Is `/etc/systemd/system` included in that list?

dave-atron, 2020-05-23:
Yes, `/etc/systemd/system` is included in the list.

medusalix, 2020-05-23:
Have you tried running `sudo make uninstall` followed by `sudo make install` and `sudo systemctl enable xow` to completely reinstall the service? What is the output of `sudo systemctl status xow`?

dave-atron, 2020-05-23:
Here's my output of that sequence:

`sudo make uninstall
[sudo] password for dave: 
rm -f /usr/local/bin/xow
rm -f /etc/udev/rules.d/99-xow.rules
rm -f /etc/modules-load.d/xow-uinput.conf
rm -f /etc/modprobe.d/xow-blacklist.conf
rm -f /etc/systemd/system/xow.service
[dave@sevastopol xow]$ sudo make install
sed 's|#BINDIR#|/usr/local/bin|' install/service.in > xow.service
install -D -m 755 xow /usr/local/bin/xow
install -D -m 644 install/udev.rules /etc/udev/rules.d/99-xow.rules
install -D -m 644 install/modules.conf /etc/modules-load.d/xow-uinput.conf
install -D -m 644 install/modprobe.conf /etc/modprobe.d/xow-blacklist.conf
install -D -m 644 xow.service /etc/systemd/system/xow.service
rm -f xow.service
[dave@sevastopol xow]$ systemctl enable xow
Failed to enable unit: Unit file xow.service does not exist.
[dave@sevastopol xow]$ sudo systemctl status xow
Unit xow.service could not be found.`

When I first bumped into this issue the other day (I've since uninstalled and reinstalled multiple times) it was saying that it did not have permission to read xow.service, which was what led me to change the permissions in the first place. Let me try creating a custom service and see what happens.

medusalix, 2020-05-23:
> I created a service and systemctl can see and run it just fine, so I don't think there's anything wonky going on with systemd on my machine.

What happens when you name that service `xow.service`? Does that still work fine?

dave-atron, 2020-05-23:
Good call. Although previous attempts to check the status of xow showed that the unit file was not found, renaming my test service revealed that systemd is getting permission denied when it tries to read the file for some reason. Don't mind the different path, I was trying to see if using a soft link instead of it being in the physical location would make a difference.

`● xow.service - Runs /usr/local/bin/test.sh`

     `Loaded: loaded (/usr/local/lib/systemd/system/xow.service; linked; vendor preset: disabled)`

     `Active: inactive (dead)`

`May 22 07:47:38 sevastopol systemd[1]: xow.service: Failed to open /etc/systemd/system/xow.service: Permission denied`

...

`May 23 13:18:13 sevastopol systemd[1]: xow.service: Failed to open /usr/local/lib/systemd/system/xow.service: Permission denied`

`May 23 13:27:39 sevastopol systemd[1]: Started Runs /usr/local/bin/test.sh.`
`
May 23 13:27:39 sevastopol systemd[1]: xow.service: Succeeded.`

medusalix, 2020-05-23:
Are you running `systemctl` without `sudo`? You definitely need `sudo` for things like `systemctl start xow`.

dave-atron, 2020-05-23:
Oh I am, my terminal prompts me for a password if I forget to type sudo.

medusalix, 2020-05-23:
What happens when you try to run the test script (as `xow.service`) from `/lib/systemd/system`? Does that work?

dave-atron, 2020-05-23:
My test service worked from /`lib/systemd/system/xow.service`, but moving xow's `xow.service` to `/lib` made no difference. Just now I was able to create a unit file and copy the contents of xow's `xow.service` into it, and that worked fine. I've also just renamed my latest test unit file to `xow.service` and that seems to be working.

I can also confirm on a different computer running Fedora 32 that I have the same issue.

dave-atron, 2020-05-23:
I figured out the problem (I think). For some reason, when `xow.service` gets created it does not get the correct SELinux type. It should be receiving `"system_u:object_r:systemd_unit_file_t:s0"` for its context but instead is getting `"system_u:object_r:default_t:s0"`. As soon as I corrected that the service worked. 

Adding the following lines to the makefile seems to take care of it ahead of time: 

`semanage fcontext -a -t systemd_unit_file_t /usr/lib/systemd/system/xow.service`
`restorecon -v /etc/systemd/system/xow.service`

I don't know enough about this to know if that's a portable solution, but it worked for me!

medusalix, 2020-05-24:
I just tried to reproduce this by installing xow on a live image of Fedora 32 with SELinux enabled. This is the output of `ls -Z /etc/systemd/system/*.service` after running `sudo make install`:

```
system_u:object_r:systemd_unit_file_t:s0 /etc/systemd/system/dbus-org.bluez.service
system_u:object_r:systemd_unit_file_t:s0 /etc/systemd/system/dbus-org.fedoraproject.FirewallD1.service
system_u:object_r:systemd_unit_file_t:s0 /etc/systemd/system/dbus-org.freedesktop.Avahi.service
system_u:object_r:systemd_unit_file_t:s0 /etc/systemd/system/dbus-org.freedesktop.ModemManager1.service
system_u:object_r:systemd_unit_file_t:s0 /etc/systemd/system/dbus-org.freedesktop.nm-dispatcher.service
system_u:object_r:systemd_unit_file_t:s0 /etc/systemd/system/dbus.service
system_u:object_r:systemd_unit_file_t:s0 /etc/systemd/system/display-manager.service
system_u:object_r:systemd_unit_file_t:s0 /etc/systemd/system/xow.service
```

`sudo systemctl enable xow` leads to this:

```
Created symlink /etc/systemd/system/multi-user.target.wants/xow.service → /etc/systemd/system/xow.service.
```

I'm able to install xow just fine.

dave-atron, 2020-05-24:
That works perfectly! 

It's really weird that I had the issue on two computers and you didn't, and SELinux applying incorrectly doesn't seem to be a super common issue. Not really sure how to troubleshoot my issue at this point. However, I appreciate your help, and I've learned a ton from this experience.

medusalix, 2020-05-24:
> Not really sure how to troubleshoot my issue at this point.

Yeah, me neither :smile:.

closed 2020-05-24 by medusalix